espressif/esp-idf #7960: 2 separate ESP32 S3 beta 3 modules are giving MAC ID as MAC: 00:00:00:00:00:00

naveenpsgit, 2021-11-24:
I am facing a problem with getting output for a sample code(helloworld) that we were able to upload to ESP32 S3.

I am using the following configurations

Windows 10
Visual Studio Code Version 1.62.3 with ESPIDF Extension
Python version 3.8
ESP-IDF Version (4.3.1 latest) - supports ESPS3Beta3 (Which is ours)
Board Version: ESP32 S3 WROOM 1 E2 N8R2

We put the ESP32S3 in boot mode by pulling down GPIO0 and GPIO46 to GND, and was able to successfully flash the code. But the MAC address for this is MAC: 00:00:00:00:00:00. 
We have tested yet another ESP32 S3 on a new board and the same No MAC Address issue is there. But it is flashing the code. The following is the log of code upload:

PS C:\Users\acer\Desktop\esp-ex\hello_world\hello_world> C:\Users\acer\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\Users\acer\esp\esp-idf\components\esptool_py\esptool\esptool.py --no-stub -p COM5 -b 115200 --before default_reset --after hard_reset --chip esp32s3beta3 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 build/hello-world.bin 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin
esptool.py v3.1-dev
Serial port COM5
Connecting....
Chip is ESP32-S3(beta3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00010000 to 0x00037fff...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Took 1.20s to erase flash block
Wrote 162816 bytes at 0x00010000 in 16.5 seconds (78.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.20s to erase flash block
Wrote 19456 bytes at 0x00000000 in 2.0 seconds (78.5 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.05s to erase flash block
Wrote 3072 bytes at 0x00008000 in 0.3 seconds (81.0 kbit/s)...
Hash of data verified.
 
Leaving...
Hard resetting via RTS pin.

As you can see Even Though the MAC Address issue is there for 2 separate boards, we were able to flash the code completely, 

Now coming to the second issue. When we monitor the uploaded code on both the ESP32 S3s, the following reset error is coming as following:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
Invalid chip id. Expected 9 read 4. Bootloader for wrong chip?
ets_main.c 329
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
Invalid chip id. Expected 9 read 4. Bootloader for wrong chip?
ets_main.c 329
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
Invalid chip id. Expected 9 read 4. Bootloader for wrong chip?
ets_main.c 329

But we have double checked the target device and it is set to the correct chip that we have. Because unless it is not set to the correct chip we won't be even able to upload the code in the first place. We have checked eFuse issues and that also didnt help us to solve the issue.

Any help would be appreciated. 

Thank You.

igrr, 2021-11-25:
Hi @naveenpsgit,
The error you are seeing indicates that you have an earlier version of the chip (engineering sample), ESP32-S3-beta2:
```
Invalid chip id. Expected 9 read 4. Bootloader for wrong chip?
```
Initial support for ESP32-S3-beta2 was added in commit https://github.com/espressif/esp-idf/commit/2db04d6e, with subsequent commits enabling some of the IDF features on this chip.
It was replaced with support for the next pre-release version of the chip, ESP32-S3-beta3, in https://github.com/espressif/esp-idf/commit/fa2946d6.
Finally, it was replaced with support for the mass production version of the chip, ESP32-S3, in https://github.com/espressif/esp-idf/commit/e005ec08.

You can still use your development board in the range of ESP-IDF commits where ESP32-S3-beta2 support is present, but unfortunately there is no active support or bug fixing for this chip version anymore.

naveenpsgit, 2021-11-25:
Hi @igrr 

First and foremost, thank you for your time to answer my query. The problem is that I am having the beta 3 version only. If you see my flashing log you can see "Chip is ESP32-S3(beta3)"  in it. Whenever I do s3beta2 flash its saying that invalid board type and i should select beta3 as that is the detected board. Also what is your comment on the MAC ID being invalid.

Thanks Again.
Naveen PS

igrr, 2021-11-25:
Oh, I'm sorry, I read the message the other way around. I'm going to blame early morning hours...

The chip (ROM) bootloader expected chip ID 9, which is actually the production version of the chip (not beta). However in the 2nd stage bootloader header, it saw chip ID 4, which is for the beta2.

So you actually have the production version of ESP32-S3 (not ESP32-S3-beta3). Please use `release/v4.4` or `master` branch for development. The branch `release/v4.3` only contains support for ESP32-S3-beta2.

The reason why esptool is identifying the chip as ESP32-S3-beta3 is that in release/v4.3 branch, an older version of esptool is used. That version of esptool didn't know of the existence of ESP32-S3 (final version), so it identifies the chip as beta3. It also isn't aware of the eFuse layout of the final version of the chip, so it cannot find the correct MAC address.

naveenpsgit, 2021-11-25:
Hi Ivan, @igrr 

No problem Ivan. I understood clearly now. Thank you for explaining specifically what version will work and the "why" aspect of the error. I will try your solution and mostly I hope it will be sorted out and will close the issue once done.

Thanks for your time
Naveen PS

naveenpsgit, 2021-11-26:
This Issue is solved

closed 2021-11-26 by naveenpsgit