espressif/esp-idf #7961: SPI Ethernet and External SPI flash are not working on the same SPI bus.

nilesh-dryad, 2021-11-24:
## Environment

- Development Kit: ESP32-S2-Kaluga-1
- Kit version (for WroverKit/PicoKit/DevKitC): V1.3
- Module or chip used: ESP32-S2-Wrover
- IDF version: v4.3.1 (Stable)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V-2A

## Problem Description

I connected external SPI flash (W25Q32JV) to ESP32S2. Standalone example just works fine on ESP32S2. I am able to read/write into file.
I also connected SPI Ethernet (WIZ850io - W5500) to ESP32S2. Standalone example also works fine with ESP32S2. I am getting IP and able to ping to my ESP32S2 from my network.

But when I try to use external SPI flash and SPI Ethernet on the same bus together. None of them are working. I am getting flash error (wl_erase_range failed) while writing into any file and ESP is also not getting any IP.

It looks like SPI flash driver does not expect any other SPI device on the same bus. Is it true?

### Expected Behavior
External SPI flash and SPI Ethernet both should work on the same SPI bus.

### Actual Behavior
Neither flash nor Ethernet is working together on the same SPI bus.

### Steps to reproduce

1. Take external flash example code and take basic Ethernet application and merge them together.
2. Test file write/read operation and ping operation together.

igrr, 2021-11-25:
@nilesh-dryad Your observation is correct, having SPI flash devices and other SPI devices on the same SPI bus is not supported yet (see note in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#initializing-a-flash-device). It is possible to have multiple SPI flash chips or multiple SPI devices on the same bus, but not both at the same time.

nilesh-dryad, 2021-11-25:
@igrr Thank you for the confirmation. I am closing this issue.

closed 2021-11-25 by nilesh-dryad

muratdemirtas, 2022-08-03:
any update on this?

nilesh-dryad, 2022-08-04:
I guess it is already closed. We can't share SPI bus with other SPI slave if it has SPI flash already on it. Of course we can do it if we write our own SPI flash driver and manage sync between two spi devices. Unfortunately I don't have that much time therefore I shared W5500 Ethernet SPI with other SPI and kept SPI flash alone on SPI bus. It worked quite well.

muratdemirtas, 2022-08-04:
> I guess it is already closed. We can't share SPI bus with other SPI slave if it has SPI flash already on it. Of course we can do it if we write our own SPI flash driver and manage sync between two [...]

Thanks for your quick reply, Can I use an sd card using with partitions while attaching other spi slaved on the bus? I am afraid because External SPI Flash and SD card use the same partitioning system backend on ESP32. Is there the same limitation for SD cards?

nilesh-dryad, 2022-08-05:
@muratdemirtas As far as you are using spi flash driver from esp component, you can't share that spi with other spi devices. You can attach multiple spi flash chips though. I don't know about SD card, I never used and looked into sd card with esp-idf. Therefore I don't know how it is driver in esp-idf.