espressif/esp-idf #7962: corrupted fame length in second httpd_ws_recv_frame call of websocket-echo-server

EN20M, 2021-11-24:
I have an ESP running as acces point and running a http web socket server unsing parts from the example codes. When sending a ws frame via this python script:
`
from websocket import create_connection
ws = create_connection("ws://192.168.4.1/ws")
print ("sending moin")
ws.send("test")
print ("Sent")
print ("Receiving...")
result =  ws.recv()
print ("Received '%s'" % result)
ws.close()
`

I get the following printout on the monitor:
`
W (70829) httpd_ws: httpd_ws_recv_frame: WS Message too long
I (70829) WebSocket: frame len is 4
W (70829) httpd_ws: httpd_ws_recv_frame: WS Message too long
E (70839) WebSocket: httpd_ws_recv_frame failed with 260
`

while debugging this problem I noticed that the first call of `httpd_ws_recv_frame` where `max_length = 0` returns as intended and is giving me the payload lenght 4 (see msg "test"), when using that lenght and calling `httpd_ws_recv_frame` the second time with that lenght and a dynamically provided buffer `frame->len` gets set to somthing larger than 4.

When I provide a static buffer and only call `httpd_ws_recv_frame` once right away with `max_len = 4` it works but I can't and don't want to use a static buffer.

ESP-YJM, 2021-11-25:
@EN20M Could you please share your IDF version and websocket server handle function code?

EN20M, 2021-11-25:
Here is the websocket handler code:
```
static esp_err_t echo_handler(httpd_req_t *req)
{
    httpd_ws_frame_t ws_pkt;
    uint8_t *buf = NULL;
    
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    
    if (req->method == HTTP_GET) {
        ESP_LOGI(TAG, "Handshake done, the new connection was opened");
        return ESP_OK;
    }
    
    ws_pkt.type = HTTPD_WS_TYPE_TEXT;
    /* Set max_len = 0 to get the frame len */
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    ESP_LOGI(TAG_WS, "frame len is %d", ws_pkt.len);
    if (ws_pkt.len) {
        /* ws_pkt.len + 1 is for NULL termination as we are expecting a string */
        buf = calloc(1, ws_pkt.len + 1);
        if (buf == NULL) {
            ESP_LOGE(TAG_WS, "Failed to calloc memory for buf");
            return ESP_ERR_NO_MEM;
        }
        ws_pkt.payload = buf;
        /* Set max_len = ws_pkt.len to get the frame payload */
        ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG_WS, "httpd_ws_recv_frame failed with %d", ret);
            free(buf);
            return ret;
        }
        ESP_LOGI(TAG_WS, "Got packet with message: %s", ws_pkt.payload);
    }
    ESP_LOGI(TAG_WS, "Packet type: %d", ws_pkt.type);
    if (ws_pkt.type == HTTPD_WS_TYPE_TEXT &&
        strcmp((char*)ws_pkt.payload,"Trigger async") == 0) {
        free(buf);
        return trigger_async_send(req->handle, req);
    }

    ret = httpd_ws_send_frame(req, &ws_pkt);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG_WS, "httpd_ws_send_frame failed with %d", ret);
    }
    free(buf);
    return ret;
}
```
as far as i know it is the same as in the provided example code and idf.py --version says I'm on `ESP-IDF v4.3.1-dirty`

EN20M, 2021-11-25:
now reding about the `dirty` tag I am wondering about it because I don't remember changing anything. Is there a way to reset the idf?

ESP-YJM, 2021-11-25:
You can use `git status` to check which file you modify and use `git checkout` to revert it.

EN20M, 2021-11-25:
```
$ git status
Not currently on any branch.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   components/mbedtls/esp_crt_bundle/cacrt_all.pem
        modified:   examples/get-started/blink/main/blink.c

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        examples/get-started/blink/blink.code-workspace
        tools/kconfig_new/esp-windows-curses/esp_windows_curses.egg-info/
```
now i do remember manipulating the cacrt_all.pem there was another issue with some negativ serial number and the described quickfix was to remove somthin in that file

ESP-YJM, 2021-11-25:
@EN20M This websock handle is based on version beyond v4.3.1, your version not support. Please update to the master version or release/v4.4 version and try.

EN20M, 2021-11-25:
why is it in the examples that come with the latest stable release if its not supported by it?

ESP-YJM, 2021-11-25:
@EN20M Which stable release version you saw the handle function support?

EN20M, 2021-11-25:
OMG now I feel really dumb. At some point while looking up stuff I must have been using code from the browser window instead of the example code from my local repo. But since it is so similar i just overlooked the differences. Thank you verry much.

closed 2021-11-25 by EN20M