espressif/esp-idf #7965: Loosing Bytes on UART after 128 Bytes received

WatDavid, 2021-11-25:
## Environment

- Development Kit: custom board
- Kit version -
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.2.2
- Build System:   CMake
- Compiler version: 8.4.0
- Operating System:     Windows
- (Windows only) environment type: -
- Using an IDE?: Yes (Visual Studio Professional 2019 version 16.1.0 with Visual GDB 5.5R5)
- Power Supply:        external 3.3V

## Problem Description
I´m loosing bytes on the uart receive on ESP32. I found out that some bytes are really missing and it looks like that the last received bytes appear at the beginning of the receive buffer.
Its interessting to say that the difference between the lost byte and the bytes doubled at the beginning of the buffer counts 128 (That means 128 Bytes are received correctly). See the example in "Actual Behavior".

### Expected Behavior
Recieve all Bytes correct over UART.

### Actual Behavior

I have a Sniffer which is between my two communication partners. So I can read what is actually on the UART wires.
In the following you can see the output of the sniffer ( = expected behavior) and the Bytes received from ESP32.
The first 5 recieved Bytes from ESP32 (67 e7 fd bd 39) are the same bytes received before on byte (cb) was lost.
It is possible that there is a overflow of the UART Hardware Buffer?
Could there be an impact if higher prior interrupts interrupting the running uart isr (uart_rx_intr_handler_default)?
Can there be a Problem with the Ringbuffer, which IDF is using for UART?

```
- Log Sniffer:    7E 00 9E 99 00    01 F8 C0 B7 6D 6F 9E 6A 90 46 36 94 29 25 57 19 16 69 00 BC EA 10 54 D9 79 91 EA C6 6B AD F5 C9 AA A2 9E CB 41 D3 52 6C 8E D6 3A 34 DB BC C9 11 A5 48 8C B4 55 09 AB 25 70 A3 A9 A3 AF 4D 2E E6 9E 8A 74 31 CC 41 11 FA 31 DF E2 67 50 F4 DF C7 5B 2C C5 BC FF ED F7 AF 24 0C F5 DA D4 92 58 6C 5B E9 D8 0F 02 27 A5 18 DF 1F 3F 5F E8 38 7D 5E E5 41 90 F2 D4 D0 9B 96 36 00 58 67 E7 FD BD 39 CB    D3 3A C5 65 3A 51 65 CD AB 1E E0 3D 2A D1 81 A7 37 A4 3F 9E 73 7E
- Log Controller: 67 e7 fd bd 39    01 f8 c0 b7 6d 6f 9e 6a 90 46 36 94 29 25 57 19 16 69 00 bc ea 10 54 d9 79 91 ea c6 6b ad f5 c9 aa a2 9e cb 41 d3 52 6c 8e d6 3a 34 db bc c9 11 a5 48 8c b4 55 09 ab 25 70 a3 a9 a3 af 4d 2e e6 9e 8a 74 31 cc 41 11 fa 31 df e2 67 50 f4 df c7 5b 2c c5 bc ff ed f7 af 24 0c f5 da d4 92 58 6c 5b e9 d8 0f 02 27 a5 18 df 1f 3f 5f e8 38 7d 5e e5 41 90 f2 d4 d0 9b 96 36 00 58 67 e7 fd bd 39       d3 3a c5 65 3a 51 65 cd ab 1e e0 3d 2a d1 81 a7 37 a4 3f 9e 73 7e
```

### Code to reproduce this issue
Uart configuration:

Init is called for UART 2 with the following parameters:

```cpp
bufferSize: 4096
baudrate: 500000
txGpio: gpio_num_t::GPIO_NUM_21
rxGpio: gpio_num_t::GPIO_NUM_33
rtsGpi: gpio_num_t::GPIO_NUM_27
ctsGpio: UART_PIN_NO_CHANGE
blockSend: true
reInit: false
```

```cpp

void Uart::Init(uint32_t const bufferSize, uint32_t const baudrate, int const txGpio, int const rxGpio, int const rtsGpio, int const ctsGpio, bool blockSend, bool reInit)
{
   m_Baud = baudrate;
   m_txGpio = txGpio;
   m_rxGpio = rxGpio;
   m_BlockSend = blockSend;
   
   esp_err_t err = ESP_OK;
   
   if (reInit)
   {
      err = uart_driver_delete(c_uartNum);
   }

   if (err == ESP_OK)
   {
      int intr_alloc_flags = 0;
      intr_alloc_flags |= ESP_INTR_FLAG_IRAM;
      intr_alloc_flags |= ESP_INTR_FLAG_LEVEL2;
      err = uart_driver_install(c_uartNum, bufferSize, bufferSize, 0, nullptr, intr_alloc_flags);
   }
  
   if (err == ESP_OK) 
   {
      uart_config_t uart_config;
      uart_config.baud_rate = baudrate;
      uart_config.data_bits = UART_DATA_8_BITS;
      uart_config.parity = UART_PARITY_DISABLE;
      uart_config.stop_bits = UART_STOP_BITS_1;
      uart_config.rx_flow_ctrl_thresh = c_flow_ctrl_thresh;
      uart_config.source_clk = UART_SCLK_APB;//UART_SCLK_REF_TICK;
      uart_config.flow_ctrl = (ctsGpio != UART_PIN_NO_CHANGE) ? UART_HW_FLOWCTRL_CTS_RTS : UART_HW_FLOWCTRL_DISABLE;

      err = uart_param_config(c_uartNum, &uart_config);
   }

   if (err == ESP_OK)
   {
      err = uart_set_pin(c_uartNum, m_txGpio, m_rxGpio, rtsGpio, ctsGpio);
   }
   
   if (rtsGpio != UART_PIN_NO_CHANGE && ctsGpio == UART_PIN_NO_CHANGE)
   {
      err = uart_set_mode(c_uartNum, UART_MODE_RS485_HALF_DUPLEX);
   }
   
   ESP_ERROR_CHECK_ASSERTION_C(err);
}
```

The Debug output is directly after the IDF function:
```cpp
uart_read_bytes(UART_NUM_2, test, size, timeout / portTICK_RATE_MS);
```

## Debug Logs

Further Logs with the same bug:
```
- Log Sniffer:    7E 00 9E    99 00 01 F6 A8 DE 38 E6 10 06 E5 42 10 E2 94 0B 38 9A DF C1 25 C9 4C EC E1 F1 E6 E3 5A B5 1D E7 1F 6F 46 C8 AA C2 9E 29 19 A6 B9 07 9B C8 05 11 02 10 55 3B 40 1F AC C0 BE 70 22 94 66 B2 81 E1 E0 56 99 F8 B1 49 8F 0E FA C9 E9 4D 17 A1 07 18 B9 9C 4C 48 36 A8 54 EB 33 EF 6D 91 78 EE FB 28 9A 83 87 50 DB 20 E2 C2 9A 96 F3 AB 4F A2 1D DA D4 C2 DB AB FE 69 51 42 BA F1 93 2C E1 78 99 81 59    42 6B 1B 26 92 29 90 DA 69 EB F5 B0 96 4D 27 6B 8C 82 9B 2A 64 34 7E
- Log Controller: 78 99 81    99 00 01 f6 a8 de 38 e6 10 06 e5 42 10 e2 94 0b 38 9a df c1 25 c9 4c ec e1 f1 e6 e3 5a b5 1d e7 1f 6f 46 c8 aa c2 9e 29 19 a6 b9 07 9b c8 05 11 02 10 55 3b 40 1f ac c0 be 70 22 94 66 b2 81 e1 e0 56 99 f8 b1 49 8f 0e fa c9 e9 4d 17 a1 07 18 b9 9c 4c 48 36 a8 54 eb 33 ef 6d 91 78 ee fb 28 9a 83 87 50 db 20 e2 c2 9a 96 f3 ab 4f a2 1d da d4 c2 db ab fe 69 51 42 ba f1 93 2c e1 78 99 81       42 6b 1b 26 92 29 90 da 69 eb f5 b0 96 4d 27 6b 8c 82 9b 2a 64 34 7e

- Log Sniffer:    7E    00 9E 99 00 01 E3 D0 D8 D4 AD 55 14 D9 EC A7 9A 8A 16 18 83 B6 77 5B F6 BA C4 CD DF 36 20 9D 15 29 80 F5 7B 0A 83 24 38 A9 21 F9 0C 83 FD 29 EB F5 8C D7 5E A5 C4 46 94 78 0C 13 E5 A5 59 E3 BA 69 9D 5C 24 7C 4C 57 7F 48 3B 8E 4C 6C 53 DF A8 20 34 26 8C 33 D1 97 7C E3 34 6F 82 BC EB 36 F6 0A 4A 9A 3D CD 29 92 CF CE F9 5C 6F BD 82 70 BE 30 93 E5 C2 A3 3A 8D B1 1D 64 E6 D7 E7 33 30 E2 C2    C2 CE A9 23 D2 12 38 F4 17 E5 31 2B 14 E2 EF 64 CC 0D C0 6E 4A 09 84 AB 7E
- Log Controller: e2    00 9e 99 00 01 e3 d0 d8 d4 ad 55 14 d9 ec a7 9a 8a 16 18 83 b6 77 5b f6 ba c4 cd df 36 20 9d 15 29 80 f5 7b 0a 83 24 38 a9 21 f9 0c 83 fd 29 eb f5 8c d7 5e a5 c4 46 94 78 0c 13 e5 a5 59 e3 ba 69 9d 5c 24 7c 4c 57 7f 48 3b 8e 4c 6c 53 df a8 20 34 26 8c 33 d1 97 7c e3 34 6f 82 bc eb 36 f6 0a 4a 9a 3d cd 29 92 cf ce f9 5c 6f bd 82 70 be 30 93 e5 c2 a3 3a 8d b1 1d 64 e6 d7 e7 33 30 e2       c2 ce a9 23 d2 12 38 f4 17 e5 31 2b 14 e2 ef 64 cc 0d c0 6e 4a 09 84 ab 7e
```

## Other items if possible

-

WatDavid, 2021-12-01:
The Problem could be solved by reducing the rx full threshold using the function uart_set_rx_full_threshold(...).
So I asume the problem was an overflow of the uart hw buffer.

closed 2021-12-01 by WatDavid

o-marshmallow, 2021-12-01:
Hi @WatDavid ,
Thank you for giving your solution! I know it's a bit late, but here is the diagram of the UART RAM as presented on page 338 of the ESP32 TRM:
[image]

So you guessed right, as the FIFOs are implemented as ring buffers, when there is an overrun (receive more than 128 bytes), the characters at the beginning will be overwritten.