espressif/esp-idf #7967: 修改工程最后生成的文件名

waqq1314, 2021-11-25:
如题，请问如何更改最后生成的文件名。
实际情况：
我这边是一个开关程序，但开关分1-4路控制，我需要将功能通过文件名区分出来，而具体路数的配置又在kconfig里面，不能直接通过project(switch+CONFIG_CHANNEL_COUNT)的方式实现。

另外能否提供一个编译前，编译完成的接口，文件名你们定，具体这个由我们来，类似很多ide里面都有的功能。
我肯定不能去改你们sdk包里面的东西，不然到时一更新啥都没了

chudsaviet, 2021-11-27:
English please.

o-marshmallow, 2022-10-11:
Hi @JustQY ,
Sorry for the delay of response. As you saw, we don't have the possiblity to rename the project with Kconfig option, however there is a workaround to still get the result you are expecting.

**NOTE:** To be clear, this is a **dirty** workaround that will currently work on IDF (tested with commit [ead10bf](https://github.com/espressif/esp-idf/commit/ead10bf01a22163286d354d1e88fb6df6ea41388)) because of the way `cmake/project.cmake` is written, so there is no guarantee this will still work in future versions.

The idea is as folow: in the `project` macro that is called from any IDF project, for example `project(hello_world)`, the executable name is store inside a variable named `project_elf`. Then, if we "forbid" any CMake function to set this variable, we can force the name we want for our final binary. In other words, we can initialize `project_elf` with the value we want and they force it to be read-only. Then, to do so your `CMakeLists.txt` should be modified to look like:
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

# Macro that will be executed everytime the variable `project_elf` is accessed
macro(set_elf_name)
  # Everytime the name is modified, set it back to what we want, here it is switch_ followed by the Kconfig option CONFIG_CHANNEL_COUNT
  set(project_elf "switch_${CONFIG_CHANNEL_COUNT}.elf")
endmacro()

# Watch the variable `project_elf` for any access, execute macro `set_elf_name` on event
variable_watch(project_elf set_elf_name)

# Initialize the project
project(hello_world)
```

Then, after compiling with `idf.py build`, you will see that your binary is named `switch_n.elf` where n is the value of Kconfig option `CONFIG_CHANNEL_COUNT`

o-marshmallow, 2022-11-08:
Closing this issue.

closed 2022-11-08 by o-marshmallow