espressif/esp-idf #7968: How do I resolve ERR_SSL_PROTOCOL_ERROR in https communication over ethernet ?

enthunilu, 2021-11-25:
## Environment

Development Kit: ESP32-Mini-1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)

Power Supply: USB
## Problem Description

//Detailed problem description goes here.
I have a simple project of loading html page. Based on example project of https_server available under esp_idf.
I have generated ssl certificate as given in the document which has validity upto 2031.
openssl req -newkey rsa:2048 -nodes -keyout prvtkey.pem -x509 -days 3650 -out cacert.pem -subj "/CN=ESP32 HTTPS server example"

When I use 
http://192.168.0.106 using chrome or any other browser I am able to load my html page.
However, when I use 
https://192.168.0.106 sometimes page gets loaded and sometimes it gives 
This site can't provide a secure connection
192.168.0.106 sent an invalid response.
ERR_SSL_PROTOCOL_ERROR and connection close.

Can anyone suggest what can lead to this error ?

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

AdityaHPatwardhan, 2021-11-26:
Hi @enthunilu, Thanks for the issue.
As the certificate is self-signed and not trusted. Most browsers tend to have a problem opening a secure connection with the https_server.
can you check this with Mozilla browzer of the exact issue is reproduced ?
Additionally Can you provide a detailed log for the `https_server` side , that would help to understand the exact cause.

Thanks,
Aditya

enthunilu, 2021-11-27:
Hi @AdityaHPatwardhan,
Thanks for the response.
I missed to update in my above post. I am using MatrixSSL instead of mbedtls. 
I checked data on wireshark and found below events,
 
3524	448.088352	192.168.0.106	192.168.0.107	TCP	54	53239 → 443 [ACK] Seq=3805 Ack=960 Win=63282 Len=0

3525	448.088951	192.168.0.106	192.168.0.107	TCP	60	443 → 53239 [RST, ACK] Seq=960 Ack=3805 Win=53270 Len=0

3526	448.089492	192.168.0.106	192.168.0.107	TCP	60	443 → 53239 [RST, ACK] Seq=960 Ack=3805 Win=53270 Len=0

//Sometimes get below message

3533	454.576308	192.168.0.107	192.168.0.106	TCP	571	[TCP Retransmission] 53247 → 443 [PSH, ACK] Seq=1 Ack=1 Win=64240 Len=517

This issue I am getting on 
1. Mozilla browzer
2. Chrome
3. MS Edge

Thanks!!

enthunilu, 2021-11-28:
Please find attached image for some more logs.

closed 2021-12-06 by enthunilu