espressif/esp-idf #7969: [Request]Adding support for release v4.4

Huckies, 2021-11-26:
Hi there, it must be great to have a m1 native toolchain to build project with, so thanks to your effort.
I'm using ESP-IDF release v4.4 cause currently it's the only version supports the arduino-as-a-component library, so no second choice.

This version uses a toolchain esp-2021r2-8.4.0 now, wish you can help with the update.

antmak, 2021-11-26:
Thank you for the reporting! Yes, we will add native macos@m1 builds in follow-up releases of our tools. Then we backport the support onto v4.4 probably.

igrr, 2021-11-26:
@Huckies I'm going to close this issue in favor of the existing one — https://github.com/espressif/esp-idf/issues/6113. Please follow the link to that issue and click "Subscribe" to get notified when an aarch64 toolchain for macOS is released.

closed 2021-11-26 by igrr