espressif/esp-idf #7972: strange bug or behavior in sd card example

ankit-thealchemist, 2021-11-26:
## Environment

- Development Kit:      |ESP32-DevKitC v4
- Kit version |v4
- Module or chip used:  |ESP32-WROOM-32D
- 
- IDF version (run ``git describe --tags`` to find it): 4.3.1
    
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: vs code with esp -idf extension
- Power Supply:         [USB]

## Problem Description

I am following the sd card example as mention in https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdspi

I just change this structure 

 spi_bus_config_t bus_cfg = {
         .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
         .sclk_io_num = PIN_NUM_CLK,
         .quadwp_io_num = -1,
         .quadhd_io_num = -1,
         .max_transfer_sz = 4000,
     };

to this
    spi_bus_config_t bus_cfg;

    bus_cfg.mosi_io_num = PIN_NUM_MOSI;
    bus_cfg.miso_io_num = PIN_NUM_MISO;
    bus_cfg.sclk_io_num = PIN_NUM_CLK;
    bus_cfg.quadwp_io_num = -1;
    bus_cfg.quadhd_io_num = -1;
    bus_cfg .max_transfer_sz = 4000;

### Expected Behavior

it should work flawlessly

### Actual Behavior

getting this error

I (331) example: Using SPI peripheral
E (341) spi: spi_bus_initialize(468): intr flag not allowed
E (341) example: Failed to initialize bus.

### Steps to reproduce

change the initialization 
// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

https://gist.github.com/ankit-thealchemist/2d5ab719de5348e33513f591580cc1ad

negativekelvin, 2021-11-26:
If you are not going to set the value of all members of spi_bus_config_t you should zero-initialize it first

`spi_bus_config_t bus_cfg = {};`

ankit-thealchemist, 2021-11-27:
> If you are not going to set the value of all members of spi_bus_config_t you should zero-initialize it first
> 
> `spi_bus_config_t bus_cfg = {};`

Strange.... Does it something basic?  Do we have any further reading on the issue?
Thanks anyways!

igrr, 2021-11-28:
@ankit-thealchemist yes, please check [C language reference regarding structure initialization](https://en.cppreference.com/w/c/language/struct_initialization) and [instructions in the IDF programming guide](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/api-conventions.html#configuration-structures). The final paragraph in that section explains the solution you need to use for C++, which is the same as @negativekelvin has pointed out.

closed 2021-11-28 by igrr

ankit-thealchemist, 2021-11-28:
> @ankit-thealchemist yes, please check [C language reference regarding structure initialization](https://en.cppreference.com/w/c/language/struct_initialization) and [instructions in the IDF [...]

Thanks Ivan for help!