espressif/esp-idf #7973: nimBLE security implementation similar way as given in Bluedroid BLE

vim-git, 2021-11-27:
## Environment

- Development Kit: ESP32-MINI-1
- Kit version 
- Module or chip used: ESP32
- IDF version v4.2.1
- Build System: Make|CMake|idf.py
- Compiler version xtensa-esp32-elf-gcc (Cross-tool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes (eclipse)
- Power Supply: USB

## Problem Description

We want to know details about security model in nimBLE.
We requires same layer of security in nimBLE as it is present in Bluedroid gatt security server.

### Expected Behavior
Following are the security parameters which we are getting used in Bluedroid security model.
Security Configuration:- 
Passkey
auth_req
icap mode
key size
auth option
oob support
init key
rsp key

We want to use same security configuration in nimBLE. But while executing nimBLE example(blehr), We have not found any security level in that and even in other examples as well.

### Actual Behavior

We required secured paring with passkey and other security parameters in nimBLE as it present in Bluedroid BLE gatt server example. 

### Steps to reproduce

1. Build ESP-IDF example for gatt security server for bluedroid stack and blehr from nimble example.
2. In Bluedroid BLE, It is asking for passkey while paring and gives secure pairing and communication between device. 
3. in nimBLE, data is getting read/write but while pairing not getting any secure pairing and communication.

Note:- To test this we have used nRF Connect app. 

### Code to reproduce this issue

Use ESP-IDF example codes. 

#include "esp_log.h"

void app_main()
{

}

rahult-github, 2021-11-27:
Hi @vim-git,

NimBLE stack does have existing API / framework for security. Can you please refer to bleprph example to check usage for same ? Menuconfig options are provided like [this](https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/nimble/bleprph/main/Kconfig.projbuild)  . You can refer to different options and enable them to test. In the sample application, upon enabling the appropriate options , the gatt characteristics are enabled with flags that require correct security level to perfom read / write .

Thanks,
 Rahul

closed 2022-05-23 by rahult-github

xark-10, 2023-06-15:
any sample implementation available for passkey based security for pairing could not find any

rahult-github, 2023-06-15:
Hi @xark-10  , 

 Can you please try selecting the BLE_SM_IO_CAP_KEYBOARD_ONLY or BLE_SM_IO_CAP_KEYBOARD_DISP for bleprph example ( assuming you already have EXAMPLE_ENCRYPTION enabled ) and check. It should go for passkey based security when remote device attempts to access a characteristic which has ENC flag set.