espressif/esp-idf #7974: Build Code for ESP32 U4WD for One Core

hdpklm, 2021-11-27:
Hello All.

I use Windows 10
vscode with arduino extension for programmer the esp32

I work with D2WD and all wotk OK

but now I need to build code for ESP32 U4WD. but its print this error

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
E (269) cpu_start: Running on single core chip, but application is built with dual core support.
E (269) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.
abort() was called at PC 0x40084805 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088784:0x3ffe3be0 0x40088a01:0x3ffe3c00 0x40084805:0x3ffe3c20 0x40079247:0x3ffe3c40 0x400792ad:0x3ffe3c70 0x400792b8:0x3ffe3ca0 0x40079465:0x3ffe3cc0 0x400806da:0x3ffe3df0 0x40007c15:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
```

I dont know how to force compiler to Build code for only one core. and I cant select Board ESP32 U4WD because the sdk only know the modules and not the chips

thank you a lo

igrr, 2021-11-28:
Hi @hdpklm, at the moment the Arduino core for ESP32 (https://github.com/espressif/arduino-esp32) doesn't have support for ESP32-U4WD. There is a feature request for adding this support: https://github.com/espressif/arduino-esp32/issues/5183. Please follow that link and click "subscribe" button to get notified about progress on that issue.

You can still compile programs for ESP32-U4WD using ESP-IDF, see https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/get-started/index.html for the getting started instructions. When using ESP-IDF, you need to use `idf.py menuconfig` or the GUI configuration editor in VS Code to enable single core mode (i.e. `CONFIG_FREERTOS_UNICORE` option). Then your program will run on ESP32-U4WD. If you wish to use Arduino libraries as part of your ESP-IDF application, check out these instructions for using arduino-esp32 as an ESP-IDF component: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#installation.

hdpklm, 2021-11-28:
Hello @igrr

 ok, thank you very much

closed 2021-12-01 by hdpklm