espressif/esp-idf #7977: using I2S_COMM_FORMAT_STAND_PCM_SHORT doesn't trigger I2S_INTR_IN_SUC_EOF

CarlosDerSeher, 2021-11-29:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- **Development Kit:**      none
- **Module or chip used:**  ESP32-WROOM-32E
- **IDF version:** v5.0-dev-489-gef98a363e3
- **Build System:** CMake
- **Compiler version:** xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0
- **Operating System:**     Windows 10
- **environment type:** ESP Command Prompt
- **Using an IDE?:** Yes
Java Runtime Version: 16.0.1+9-24
Eclipse Version: 4.20.0.v20210611-1600
Eclipse CDT Version: 10.3.1.202106191658
IDF Eclipse Plugin Version: 2.1.2.202106231127
ESP-IDF v5.0-dev-489-gef98a363e3-dirty
Python set for IDF_PYTHON_ENV: Python 3.9.6 
**- Power Supply:**         external 3.3V

## Problem Description

When configured as a I2S master with communication format I2S_COMM_FORMAT_STAND_PCM_SHORT receiving data is impossible and `i2s_read()` will time out. I2S_INTR_IN_SUC_EOF will never be set.

### Expected Behavior

`i2s_read()` returning audio data from audio codec.

### Actual Behavior

`i2s_read()` will time out, no data is returned

### Code to reproduce this issue

```cpp

void setup_dsp_i2s(void)
{
	uint32_t rate = 8000;

  i2s_config_t i2s_config0 = {
	.mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_RX,
	.sample_rate = rate,
	.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
	.channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,             // 1-channels
	.communication_format = I2S_COMM_FORMAT_STAND_PCM_SHORT,
[... 720 lines cut ...]
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE
#define CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE
#define CONFIG_TASK_WDT CONFIG_ESP_TASK_WDT
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1
#define CONFIG_TASK_WDT_TIMEOUT_S CONFIG_ESP_TASK_WDT_TIMEOUT_S
#define CONFIG_TCPIP_RECVMBOX_SIZE CONFIG_LWIP_TCPIP_RECVMBOX_SIZE
#define CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY
#define CONFIG_TCPIP_TASK_STACK_SIZE CONFIG_LWIP_TCPIP_TASK_STACK_SIZE
#define CONFIG_TCP_MAXRTX CONFIG_LWIP_TCP_MAXRTX
#define CONFIG_TCP_MSL CONFIG_LWIP_TCP_MSL
#define CONFIG_TCP_MSS CONFIG_LWIP_TCP_MSS
#define CONFIG_TCP_OVERSIZE_MSS CONFIG_LWIP_TCP_OVERSIZE_MSS
#define CONFIG_TCP_QUEUE_OOSEQ CONFIG_LWIP_TCP_QUEUE_OOSEQ
#define CONFIG_TCP_RECVMBOX_SIZE CONFIG_LWIP_TCP_RECVMBOX_SIZE
#define CONFIG_TCP_SND_BUF_DEFAULT CONFIG_LWIP_TCP_SND_BUF_DEFAULT
#define CONFIG_TCP_SYNMAXRTX CONFIG_LWIP_TCP_SYNMAXRTX
#define CONFIG_TCP_WND_DEFAULT CONFIG_LWIP_TCP_WND_DEFAULT
#define CONFIG_TIMER_QUEUE_LENGTH CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
#define CONFIG_TIMER_TASK_PRIORITY CONFIG_FREERTOS_TIMER_TASK_PRIORITY
#define CONFIG_TIMER_TASK_STACK_DEPTH CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
#define CONFIG_TIMER_TASK_STACK_SIZE CONFIG_ESP_TIMER_TASK_STACK_SIZE
#define CONFIG_TOOLPREFIX CONFIG_SDK_TOOLPREFIX
#define CONFIG_UDP_RECVMBOX_SIZE CONFIG_LWIP_UDP_RECVMBOX_SIZE
```

tpbedford, 2021-12-01:
Same is true of ESP-IDF 4.3. I've experienced the same issue. I've had some luck by deliberately misconfiguring the RX register bit **I2S_CONF_REG.I2S_RX_SHORT_SYNC**

In i2s_ll.h (ESP-IDF 4.3) I can make the below change in _i2s_ll_set_rx_pcm_short(i2s_dev_t *hw)_ by modifying _conf.rx_short_sync_ to be cleared. The physical WS clock signal on GPIO is still set to PCM SHORT thanks to the _conf.tx_short_sync_ bit setting, at least when using _i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_RX)_
```
/**
 * @brief Set I2S TX to PCM short standard
 *
 * @param hw Peripheral I2S hardware instance address.
 */
static inline void i2s_ll_set_tx_pcm_short(i2s_dev_t *hw)
{
    hw->conf.tx_short_sync = 1;
    hw->conf.tx_msb_shift = 0;
}

/**
 * @brief Set I2S RX to PCM short standard
 *
 * @param hw Peripheral I2S hardware instance address.
 */
static inline void i2s_ll_set_rx_pcm_short(i2s_dev_t *hw)
{
    //hw->conf.rx_short_sync = 1; // fails to trigger ISR on receive
    hw->conf.rx_short_sync = 0; // UNSET this to correctly trigger the ISR on receive
    hw->conf.rx_msb_shift = 0;
}

```

I see that this has been refactored in ESP-IDF 4.4 and 5.0, so perhaps there's a similar work-around that can be made to omit the call to i2s_hal_rx_pcm_cfg(hal, cfg), or after installing the driver try a call to i2s_hal_rx_set_common_mode(i2s_hal_context_t *hal, const i2s_hal_config_t *hal_cfg) to revert the short PCM setting for the RX path?

CarlosDerSeher, 2021-12-02:
Thank you for your input and indeed after changing

```cpp
void setup_dsp_i2s(void)
{
  uint32_t rate = 8000;

  i2s_config_t i2s_config0 = {
	.mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_RX,
	.sample_rate = rate,
	.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
	.channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,             // 1-channels
	.communication_format = I2S_COMM_FORMAT_STAND_PCM_SHORT,
	.dma_buf_count = 2,
	.dma_buf_len = I2S_FRAME_SIZE,
	.intr_alloc_flags = ESP_INTR_FLAG_LEVEL2 | ESP_INTR_FLAG_IRAM,	//Default interrupt priority
	.use_apll = true,
	.fixed_mclk = 0,
	.mclk_multiple = 1280,
	.bits_per_chan = 0,		   // default '0' means equal to 'bits_per_sample'
	.tx_desc_auto_clear = true  // Auto clear tx descriptor on underflow
  };

  i2s_pin_config_t pin_config0 = {
      .bck_io_num   = CONFIG_MASTER_I2S_BCK_PIN,
      .ws_io_num    = CONFIG_MASTER_I2S_LRCK_PIN,
      .data_out_num = CONFIG_MASTER_I2S_DATAOUT_PIN,
      .data_in_num  = CONFIG_MASTER_I2S_DATAIN_PIN
    };

  i2s_driver_install(I2S_NUM_0, &i2s_config0, 1, &i2s_event_queue);
  i2s_set_pin(I2S_NUM_0, &pin_config0);
  i2s_mclk_gpio_select(I2S_NUM_0, GPIO_NUM_0);

  GPIO.func_out_sel_cfg[CONFIG_MASTER_I2S_LRCK_PIN].inv_sel = 1;  // invert the output of GPIO matrix.

  i2s_set_clk(I2S_NUM_0, rate, 16, I2S_CHANNEL_STEREO);

  (&I2S0)->conf.rx_short_sync = 0;	// UNSET this to correctly trigger the ISR on receive
}
```

Notice the last line `(&I2S0)->conf.rx_short_sync = 0;	// UNSET this to correctly trigger the ISR on receive` in the function it starts to work. I'll still have to check if audio coming in sounds right, but doing loopback as in my opening post I am getting some data.

closed 2025-02-27 by Alvin1Zhang