espressif/esp-idf #7979: web page ESP IDF

SERIDJ, 2021-11-29:
Hi, 

I have developed a server web page with esp32 (esp idf V4.3 ) . we can get this page with SoftAP, Wifi STA or Ehernet connexion. 

With Ethernet connexion its work well  without any error.

Now the issue with this system is with wifi STA and SoftAP is not stable at all. i should try 10 times to get web page correctly without error.

more details : 

i have 11 files HTML and 11 files JS and 2 Png.

- max size of each file is 25ko.
- size of png1 is 24ko.
- size of png2 is 12ko.
 -All size file  is 893 ko (html.h).

/**************************************************************************************************************************/
##HTTP config 

```
    httpd_config_t config    = HTTPD_DEFAULT_CONFIG();
    config.max_uri_handlers  = 90;
    config.max_resp_headers  = 90;
    config.stack_size        = 1024*10;//4096 default to solve ***ERROR*** A stack overflow in task httpd has been detected.
    //try to increase timeout to solve : W (111473) httpd_txrx: httpd_sock_err: error in send : 11
    config.recv_wait_timeout = 5;//5 by default
    config.send_wait_timeout = 5;//5 by default
    
    /* Use the URI wildcard matching function in order to
     * allow the same handler to respond to multiple different
     * target URIs which match the wildcard scheme */
    config.uri_match_fn = httpd_uri_match_wildcard;
    // Start the httpd server

    /* URI handler for getting uploaded files */
    httpd_uri_t http_monitor = {
        .uri       = "/*",  // Match all URIs of type /path/to/file
        .method    = HTTP_GET,
        .handler   = GET_handler,
        .user_ctx  = NULL // Pass server data as context
    };
    // httpd_register_uri_handler(server, &http_navigate);

```

/**************************************************************************************************************************/
## Debug Logs

##Wifi SoftAP
```
I (133894)  GET_handler uri: /
are  authorized
UR AGAIN not Authenticated
I (133964)  GET_handler uri: /style.css
not Authenticated Redirect_to_login_page
... res = 0
I (133974)  GET_handler uri: /img_avatar2.png
not Authenticated Redirect_to_login_page
LOGIN_PNG Redirect_to_login_page 
 LOGIN_PNG res = 0
I (135004)  GET_handler uri: /script.js
not Authenticated Redirect_to_login_page
CRIPT_JS res = 0
W (135694) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (135694) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
I (138674)  GET_handler uri: /Validate_Login?pass=progiris
yes URI contains a query string
try  VALIDATE_LOGIN
I (138674) web server: Found URL query => pass=progiris
I (138674) web server: Found URL query parameter => query1=progiris
pass :progiris
password Match
I (138774)  GET_handler uri: /get_sensor.html
are  authorized
all window
GET_SENSOR_HTML res = 0
I (138804)  GET_handler uri: /style.css
are  authorized
all window
STYLE_CSS res = 0
I (138824)  GET_handler uri: /img_avatar2.png
are  authorized
all window
LOGIN_PNG
W (149544) httpd_txrx: httpd_sock_err: error in send : 11

```
/**************************************************************************************************************************/
## Wifi STA
```
I (49944)  GET_handler uri: /get_sensor.html
are  authorized
all window
GET_SENSOR_HTML res = 0
I (49974)  GET_handler uri: /style.css
are  authorized
all window
STYLE_CSS res = 0
I (49984)  GET_handler uri: /img_avatar2.png
are  authorized
all window
LOGIN_PNG
I (55944) wifi:bcn_timout,ap_probe_send_start
I (58444) wifi:ap_probe_send over, resett wifi status to disassoc
I (58444) wifi:state: run -> init (c800)
I (58444) wifi:pm stop, total sleep time: 41367025 us / 55336856 us

W (58444) wifi:<ba-del>idx
I (58454) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
W (58454) httpd_txrx: httpd_sock_err: error in send : 113
I (58464) web server: Stopping webserver
I (58464)  GET_handler uri: /Alert_img.png
are  authorized
all window
ALERT_PNG
W (58474) httpd_txrx: httpd_sock_err: error in send : 113
ALERT_PNG res = b006
W (58484) httpd_txrx: httpd_sock_err: error in recv : 113
W (58494) httpd_txrx: httpd_sock_err: error in recv : 113
W (58494) httpd_txrx: httpd_sock_err: error in recv : 113
```
/**************************************************************************************************************************/

ESP-YJM, 2021-11-30:
@SERIDJ For SoftAP, it shows `httpd_txrx: httpd_sock_err: error in send : 11`, maybe the reason is that the data send timeout, due to bad WIFI environment, you can increass the `send_wait_timeout ` from `5` to `10`. For STA, it seems that beacon timeout caused the STA disconnected with the AP and make data send error. You can test more and check `beacon time` whether could be reproduced stably.

SERIDJ, 2021-11-30:
@ESP-YJM thanks for ur reply.

when i can get  #beacon timeout ? in sdkconfig and WIFI_INIT_CONFIG_DEFAULT i didnt find .

what i find its  #WIFI_EVENT_STA_BEACON_TIMEOUT in enum #wifi_event_t file #esp_wifi_types.h !

ESP-YJM, 2021-11-30:
@SERIDJ `Beacon timeout` can be found in the console log which like this `wifi:bcn_timout,ap_probe_send_start`.

SERIDJ, 2021-11-30:
@ESP-YJM  ah oky I thought i should change it ...

so nothing change ,

I (56996)  GET_handler uri: /
not Authenticated Redirect_to_login_page
loginForm_html Redirect_to_login_page
 INDEX_LOGIN_HTML res = 0
I (57046)  GET_handler uri: /style.css
not Authenticated Redirect_to_login_page
... res = 0
I (57056)  GET_handler uri: /img_avatar2.png
not Authenticated Redirect_to_login_page
LOGIN_PNG Redirect_to_login_page 
 LOGIN_PNG res = 0
I (57096)  GET_handler uri: /script.js
not Authenticated Redirect_to_login_page
CRIPT_JS res = 0
W (57596) httpd_uri: httpd_uri: URI '/favicon.png' not found
W (57606) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
I (68136)  GET_handler uri: /Validate_Login?pass=progiris
yes URI contains a query string
try  VALIDATE_LOGIN
I (68136) web server: Found URL query => pass=progiris
I (68146) web server: Found URL query parameter => query1=progiris
pass :progiris
password Match
I (68266)  GET_handler uri: /get_sensor.html
are  authorized
all window
GET_SENSOR_HTML res = 0
I (68296)  GET_handler uri: /style.css
are  authorized
all window
STYLE_CSS res = 0
I (68316)  GET_handler uri: /Alert_img.png
are  authorized
all window
ALERT_PNG
I (74346) wifi:bcn_timout,ap_probe_send_start
I (76846) wifi:ap_probe_send over, resett wifi status to disassoc
I (76846) wifi:state: run -> init (c800)
I (76846) wifi:pm stop, total sleep time: 58803177 us / 72418691 us

W (76856) wifi:<ba-del>idx
I (76856) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
W (76866) httpd_txrx: httpd_sock_err: error in send : 113
I (76866) web server: Stopping webserver
ALERT_PNG res = b006
I (76876)  GET_handler uri: /img_avatar2.png
are  authorized
all window
LOGIN_PNG
W (76886) httpd_txrx: httpd_sock_err: error in send : 113

SERIDJ, 2021-11-30:
maybe  there is another best  method to write GET_handler  ?

esp_err_t GET_handler (httpd_req_t *req)
{

ESP_LOGI(" GET_handler uri", "%s", req->uri); 
    if (!strcmp(req->uri, INDEX_LOGIN_HTML))
    {
        httpd_resp_set_type(req, "text/html");
       // httpd_resp_send(req, index_html, INDEX_HTML_SIZE);
        esp_err_t  res =httpd_resp_send(req, loginForm_html, LOGINFORM_HTML_SIZE);

        return ESP_OK;
    }

    if (!strcmp(req->uri, GET_INDEX_HTML))
    {
        httpd_resp_set_type(req, "text/html");
        esp_err_t  res = httpd_resp_send(req, networkconf_html, NETWORKCONF_HTML_SIZE);
        printf("GET_INDEX_HTML res = %x\n",res);
        return ESP_OK;
    }

    if (!strcmp(req->uri, GET_SENSOR_HTML))
    {
        httpd_resp_set_type(req, "text/html");
        esp_err_t  res = httpd_resp_send(req, mainpage_html, MAINPAGE_HTML_SIZE);
        printf("GET_SENSOR_HTML res = %x\n",res);
        return ESP_OK;
    }
....
}

SERIDJ, 2021-11-30:
@ESP-YJM as i guessed, its problem of get req from web page to esp32, bcp i add delais that fixe problem in soft and wifi but its not a good solution , i should find best method to fixe it ..

closed 2022-07-18 by Alvin1Zhang