espressif/esp-idf #7981: The elephant in the room...

KaeLL, 2021-11-29:
> ESP8266_RTOS_SDK's framework is quite outdated and different from the current esp-idf and we are planning to migrate ESP8266_RTOS_SDK to esp-idf eventually after v2.0.0.

So? Any ETA? How about some transparency? Should we still bother? Do you even care?
Goes without saying last 2 questions are rhetorical since I'm sure being honest is legally ill-advised, but let if fly as a way of poking you guys.

wujiangang, 2021-11-30:
The framework of ESP8266_RTOS_SDK v3.x is more like esp-idf, but it's still outdated, it's under limited maintenance now. If you will use esp-idf, please follow the programming guide of ESP32 directly.

KaeLL, 2021-11-30:
@wujiangang with all due respect but your answer perfectly matches what a bot would say.
To make it easier to understand, I'm asking if esp-idf will (ever) officially support esp8286.

huming2207, 2021-11-30:
I don't think they will even add the support for ESP8266, as ESP32-S2, ESP32-C3, and ESP8685 are the new & better replacements for ESP8266 and there is no point supporting an old chip that blocks their efforts of pushing new chips to the market.

cdluv, 2021-11-30:
All I can say is you should start contributing t your cause and start a
movement!

KaeLL, 2021-12-01:
> I don't think they will even add the support for ESP8266

Hence the poking meant for them to come clean about it, though I'm not holding my breath.

> you should start contributing t your cause

I don't have the required knowledge.

huming2207, 2021-12-01:
> > I don't think they will even add the support for ESP8266
> 
> 
> 
> Hence the poking meant for them to come clean about it, though I'm not holding my breath.
> 
> 
> 
> > you should start contributing t your cause
> 
> 
> 
> I don't have the required knowledge.

No, frankly speaking, if I'm Espressif, I won't add the support for a 9+ years old chip particularly with different codebases & breaking changes. I believe they are a commercial organisation not a charity. They simply don't have the obligation to port and maintain ESP-IDF for ESP8266, and I believe their priority should focus on improvements for ESP32-C3/S3. I'd rather to see more ESP32-S3-WROVER with larger PSRAM to come and more  ESP32-C3 module or SiP with more form factors/packages options to come, instead of being an archeologist and port ESP8266. Meanwhile, ESP-IDF and ESP8266 RTOS SDK is NOT mainly for education purposes (aka. a toy) like Arduino. Lots of stuff need to be properly tested. It's not just port and leave it as-is, a lot of testing stuff need to be implemented. 

If you really want to add the support for ESP8266 (and probably ignore the testing), have a look here and try porting some stuff from RTOS SDK: https://github.com/espressif/esp-idf/tree/master/components/soc

KaeLL, 2021-12-01:
@huming2207 bro, chill the f- out and read the citation in my original post. Those are **their** words, not mine, and I'm just looking for answers regarding that. Learn to read before babbling about what you want and what you think I meant.

huming2207, 2021-12-01:
> @huming2207 bro, chill the f- out and read the citation in my original post. Those are **their** words, not mine, and I'm just looking for answers regarding that. Learn to read before babbling about what you want and what you think I meant.

Well, please go get a proper CS/CE degree before saying something like that. Your words only imply your incompetence in embedded software development to me.

I'm just saying Espressif has no obligation to make something completely new for an (almost) NRND chip, and as a major user of Espressif products for serious commercial projects instead of some random players, I hope them not wasting the time on old chips and focus on the new chips.

igrr, 2021-12-02:
@KaeLL and @huming2207,  please refrain from direct or veiled insults on this issue tracker. I do appreciate the contributions both of you made to ESP-IDF repository, but further interactions of this kind will result in a ban.

@KaeLL To address your question, esp-idf may eventually support esp8266. We can't say with full certainty that it will happen, but neither we can say with full certainty that it won't. There isn't some secret decision made inside Espressif that "we will never support esp8266 in esp-idf, but let's pretend that we will, and don't tell anyone". I _can_ say that no work is actively being done at this moment to add esp8266 support to esp-idf.

On one hand, the more time passes the less relevant esp8266 becomes for new projects. This reduces the likelihood that esp8266 support work will be important enough to be high up on the backlog of the esp-idf development team. On the other hand, structural improvements being made in esp-idf to accommodate newer chips actually make it easier to support esp8266, compared to esp-idf from a couple years back. What will be the net outcome, it is hard to say.

closed 2021-12-02 by igrr