espressif/esp-idf #7982: esp32s2 will fail on external PSRAM access for size > 4m

pavlohamov, 2021-11-30:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:  esp32s2 custom board
- Module or chip used:  esp32s2 r0 ESP-ROM:esp32s2-rc4-20191025
- IDF version v5.0-dev-518-g6ff72be660
- Build System:        idf.py
- Compiler version: xtensa-esp32s2-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:     external 5V

## Problem Description

Device uses external PS-RAM that supposed to be used as heap. New batch of devices is shipped with 8 meg (APS6404) PS-RAM. That leads to loadStoreProhibited exception on boot

### Expected Behavior
System works with whole memory region available as external mem. 

### Actual Behavior
System fails due to exception and reboots

### Steps to reproduce
1. Solder 8 meg psram to esp32s2Wrower kit
2. build and execute app that writes to external RAM

Proposed solution:
limit mmaped size 4 megs

ESP-Marius, 2021-12-06:
Hi, 

Could you describe a bit more in detail what goes wrong? (Log output, sdkconfig, etc).

The ESP32-S2 should be able to use 8MB just fine, I've got a S2 with ESP-PSRAM64H booting without issues.

pavlohamov, 2021-12-09:
> Hi,
> 
> Could you describe a bit more in detail what goes wrong? (Log output, sdkconfig, etc).
> 
> The ESP32-S2 should be able to use 8MB just fine, I've got a S2 with ESP-PSRAM64H booting without issues.

It was failing with load/store prohibited exception. On latest IDF it is working fine. Really odd. Closing issue and PR

closed 2021-12-09 by pavlohamov