espressif/esp-idf #7985: Resulting binaries: Exactly same code produces very different resulting binaries

carlessole, 2021-12-01:
- Module or chip used:  ESP32-WROVER (With ESP_PROG)
- IDF version: v4.3.1
- Build System: CMake/idf.py (usingEclipse Plugin)
- Compiler version: esp-2020r3-8.4.0
- Operating System: Windows
- (Windows only) environment type: -.
- Using an IDE?: Yes. Eclipse + ESP-IDF plugin, but also thorugh CLI
- Power Supply: USB

Hi everyone,
I have noticed a behaviour that it is not what I thought it was correct.

When I compile multiple times **exactly** the same source code. The resulting binaries:

1. Sometimes are almost the same, in exception of the compile date/time and some final CRC. That is the behaviour that I expect.
2. Sometimes the resulting binaries are very very different even the size is not the same (just few bytes). That is the behaviour that I don't understand. Is that correct? Why is happening?

I was expecting always the 1st case because then it is easier to probe that both binaries have been created from the same source code.

Here I attach some images when compilation differs a lot:
[image]
[image]
[image]
[image]

negativekelvin, 2021-12-01:
https://github.com/espressif/esp-idf/blob/9919b75ec1ecf203b54d3d057e8332d2ea60245c/Kconfig#L204-L209

igrr, 2021-12-01:
Linking https://github.com/espressif/esp-idf/issues/5251 and https://github.com/espressif/esp-idf/issues/5873.

I think in 4.3 this might be a regression in the linker script generator, where on different runs it may produce different order of libraries. This should be fixed in https://github.com/espressif/esp-idf/commit/404ee09181375b39d91a8b198bba0afc43d1b21b, will be part of the next v4.3.2 bugfix release.

Additionally in release/v4.4 there is a new option CONFIG_APP_REPRODUCIBLE_BUILD which @negativekelvin has kindly linked above, it will also make ELF files identical across builds, so event the ELF SHA256 value in binary image header will be stable.

carlessole, 2021-12-01:
Thanks @igrr amd @negativekelvin ,

@negativekelvin : I'm using v4.3.1 and so I have not that config field avialble. Moreover I'm ok that few things as paths, and date/time differs. But I see so many changes apart from these ones

@igrr : I think that my problem is probably related with the linker issue. I will try when v4.3.2 comes out to test it.

Thanks

igrr, 2021-12-21:
@carlessole v4.3.2 has been released, please give it a try and see if this issue is fixed for you.

Alvin1Zhang, 2022-01-24:
@carlessole Thanks for reporting, would you please help a try with the newly released IDF v4.3.2 and share any updates? Thanks.

closed 2022-03-08 by Alvin1Zhang