espressif/esp-idf #7990: NimBLE - bond is sometimes not saved in case of re-pairing

jspngh, 2021-12-01:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.4-dev-3703-gddc44956bf
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

I am using the `bleprph` example and use 2 Android (11) phones to connect and pair to the esp32.
I use I/O capability *DISPLAY YESNO* and have enabled *Use bonding*, *MITM security* and *Use secure connection feature*.
I attached the sdkconfig file for reference.
I made 2 slight modifications to the example code, described in the appropriate section.
The problem I am seeing is the following: bond information is not correctly saved if previously a `BLE_GAP_EVENT_REPEAT_PAIRING` event occurred that was declined by the peripheral (esp32).

In more detail:
Start by bonding 2 phones to the peripheral. It is important to use 2 phones,
since the issue does not seem to occur when using only 1 phone.
Now delete the bond information on one phone and connect again.
This will cause a `BLE_GAP_EVENT_REPEAT_PAIRING` event, since the esp32 already knows this phone.
Decline the pairing (with `key N`), this will cause a disconnect.
Try to connect again with the phone and now accept the pairing (`key Y`).
The phone is now bonded to the esp32. When I now disconnect and connect again, I observe the issue.

### Expected Behavior
The phone can successfully setup an encrypted connection to the esp32 without pairing again.

### Actual Behavior
The phone cannot use the existing bond to securely connect to the esp32. A connection is established, but no security.
The phone then deletes the bond info automatically, since it could not be used.

I think this is related to the changes made in #5530.
If I revert the function `ble_hs_resolv_list_rmv` to what is was before these changes, the issue does not seem to appear.

```cpp
int
ble_hs_resolv_list_rmv(uint8_t addr_type, uint8_t *ident_addr)
{
    int position;

    /* Remove from IRK records */
    position = ble_hs_is_on_resolv_list(ident_addr, addr_type);
    if (position) {

        memmove(&g_ble_hs_resolv_list[position],
                &g_ble_hs_resolv_list[position + 1],
                (g_ble_hs_resolv_data.rl_cnt - position) * sizeof (struct
                        ble_hs_resolv_entry));
        --g_ble_hs_resolv_data.rl_cnt;

        return 0;
    }

    return BLE_HS_ENOENT;
}
```
My guess is that because of the added call to  `ble_rpa_replace_id_with_rand_addr(&addr_type, ident_addr);`
certain info is kept, while is should be removed when the re-pairing of the phone "fails" / is declined.
When accepting the pairing at the `BLE_GAP_EVENT_REPEAT_PAIRING` event, this issue does not occur.

### Steps to reproduce

1. connect phone 1 to the esp32 and successfully pair (`key Y`) + disconnect again
2. connect phone 2 to the esp32 and successfully pair (`key Y`) + disconnect again
3. delete bond info on phone 1
4. connect phone 1 to the esp32 and decline pair on esp32 (`key N`), disconnect will happen
[... 119 lines cut ...]
    I (206751) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
    I (206751) NimBLE: e0:e2:e6:47:90:de
    I (206751) NimBLE:  our_id_addr_type=0 our_id_addr=
    I (206761) NimBLE: e0:e2:e6:47:90:de
    I (206761) NimBLE:  peer_ota_addr_type=0 peer_ota_addr=
    I (206771) NimBLE: 9c:6b:72:ae:68:44
    I (206771) NimBLE:  peer_id_addr_type=0 peer_id_addr=
    I (206781) NimBLE: 9c:6b:72:ae:68:44
    I (206781) NimBLE:  conn_itvl=6 conn_latency=0 supervision_timeout=500 encrypted=1 authenticated=1 bonded=1

    I (207081) NimBLE: connection updated; status=0 
    I (207081) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
    I (207081) NimBLE: e0:e2:e6:47:90:de
    I (207081) NimBLE:  our_id_addr_type=0 our_id_addr=
    I (207091) NimBLE: e0:e2:e6:47:90:de
    I (207091) NimBLE:  peer_ota_addr_type=0 peer_ota_addr=
    I (207101) NimBLE: 9c:6b:72:ae:68:44
    I (207101) NimBLE:  peer_id_addr_type=0 peer_id_addr=
    I (207111) NimBLE: 9c:6b:72:ae:68:44
    I (207111) NimBLE:  conn_itvl=38 conn_latency=0 supervision_timeout=500 encrypted=1 authenticated=1 bonded=1
</details>

## Other items if possible

- [x] [sdkconfig file](https://github.com/espressif/esp-idf/files/7632238/sdkconfig.zip)

flonas955, 2021-12-14:
I have opened a very similar issue [here](https://github.com/espressif/esp-idf/issues/7937). It looks just like the problem @jspngh is facing.

rahult-github, 2022-09-16:
@flonas955  @jspngh  , the changes put by [this](https://github.com/espressif/esp-nimble/commit/f322e780aadfcfdc4a21e1418f22db3216e46d97) commit and already part of release/v5.0 . Closing this issue . Please feel free to open, if issue still seen.

closed 2022-09-16 by rahult-github