espressif/esp-idf #7991: Failed to initialize SPIFFS (ESP_ERR_INVALID_STATE)

enthusisastSR, 2021-12-01:
Development Kit:  ESP32S2
Module or chip used:  Custom
Operating System:  Ubuntu
Using an IDE: Visual Studio Code
Power Supply: USB

Problem Description
Unable to upload spiffs in a custom Serial interface. 
Ends up with error: Failed to initialize SPIFFS (ESP_ERR_INVALID_STATE)

Expected Behavior
Able to upload spiffs

Actual Behavior
Error: Failed to initialize SPIFFS (ESP_ERR_INVALID_STATE)

### Code to reproduce this issue

```c
esp_vfs_spiffs_conf_t conf = {
      .base_path = "/data",
      .partition_label = "data",
      .max_files = 5,
      .format_if_mount_failed = true,
  };
 
  esp_err_t ret = esp_vfs_spiffs_register(&conf);
  if (ret != ESP_OK)
  {
    if (ret == ESP_FAIL)
    {
      ESP_LOGE(TAG, "Failed to mount or format filesystem");
    }
    else if (ret == ESP_ERR_NOT_FOUND)
    {
      ESP_LOGE(TAG, "Failed to find SPIFFS partition");
    }
    else
    {
      ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
    }
    return;
  }
  size_t total = 0, used = 0;
  ret = esp_spiffs_info(conf.partition_label, &total, &used);
  if (ret != ESP_OK)
  {
    ESP_LOGE(TAG, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
  }
  else
  {
    ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
  }
 
  if (l_Addr == 0x1E800)
  {
    FILE *f1 = NULL;
    f1 = fopen("/data/mem.bin", "w");
    ESP_LOGI(TAG, "file opened to append");
    if (f1 != NULL)
    {
      fwrite(ringBuf_begin(pCmd->data) + 9, l_Len, 256, f1);
      ESP_LOGI(TAG, "firmware appended to flash");
    }
    fclose(f1);
    ESP_LOGI(TAG, "file appended and closed");
    esp_vfs_spiffs_unregister(conf.partition_label);
  }
```
Debug Logs

```
I (21) boot: ESP-IDF v4.4-dev-3401-gb86fe0c66c-dirty 2nd stage bootloader
I (21) boot: compile time 11:26:14
I (22) boot: chip revision: 0
I (26) boot.esp32s2: SPI Speed      : 80MHz
I (31) boot.esp32s2: SPI Mode       : DIO
I (36) boot.esp32s2: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (64) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  3 factory          factory app      00 00 00010000 00080000
I (86) boot:  4 ota_0            OTA app          00 10 000d0000 00080000
I (94) boot:  5 ota_1            OTA app          00 11 00190000 00080000
I (101) boot:  6 data             Unknown data     01 fe 00250000 00080000
I (109) boot:  7 hwinfo           WiFi data        01 02 002d0000 00003000
I (117) boot: End of partition table
I (121) boot: Defaulting to factory image
I (125) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0bb34h ( 47924) map
I (144) esp_image: segment 1: paddr=0001bb5c vaddr=3ffbf870 size=01da4h (  7588) load
I (146) esp_image: segment 2: paddr=0001d908 vaddr=40022000 size=02710h ( 10000) load
I (153) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=25cb8h (154808) map
I (190) esp_image: segment 4: paddr=00045ce0 vaddr=40024710 size=0b154h ( 45396) load
I (202) esp_image: segment 5: paddr=00050e3c vaddr=50000000 size=00010h (    16) load
I (208) boot: Loaded app from partition at offset 0x10000
I (209) boot: Disabling RNG early entropy source...
I (222) cach3�~���~�Սѥ���cache  : size 8KB, 4Ways, cache line size 32Byte
I (222) cpu_start: Pro cpu up.
I (234) cpu_start: Pro cpu start user code
I (234) cpu_start: cpu freq: 160000000
I (234) cpu_start: Application information:
I (238) cpu_start: Project name:     esp32s2-c
I (243) cpu_start: App version:      1
I (248) cpu_start: Compile time:     Dec  1 2021 11:26:09
I (254) cpu_start: ELF file SHA256:  89846084c0da4df3...
I (260) cpu_start: ESP-IDF:          v4.4-dev-3401-gb86fe0c66c-dirty
I (267) heap_init: Initializing. RAM available for dynamic allocation:
I (274) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (280) heap_init: At 3FFC58F8 len 00036708 (217 KiB): DRAM
I (287) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (293) spi_flash: detected chip: generic
I (298) spi_flash: flash io: dio
I (302) cpu_start: Starting scheduler on PRO CPU.
```
 FW and data:
os.system("~/esp/esp-idf/components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py generate factory.csv factory.bin 12288")
os.system("esptool.py write_flash 0x2d0000 factory.bin")

o-marshmallow, 2021-12-02:
Hi @enthusisastSR ,
I am not sure to understand, you say you cannot upload your spiffs partition to your board but the debug log shows the line:
```
I (101) boot:  6 data             Unknown data     01 fe 00250000 00080000
```
which is the partition you are trying to open from your code.

I also noticed that you are flashing `factory.bin` to the address `0x2d0000`, but the partition table shows that this address corresponds to hwinfo partition:
```
I (109) boot:  7 hwinfo           WiFi data        01 02 002d0000 00003000
```

Is `factory.bin` the data partition you are trying to open in your code? What about `/data` partition? Where does it come from? If it's your spiffs data partition, then in the log, we can see that `/data` partition is of type `phy` (0x1) instead of `nvs` (0x2).
So can you please show us your `factory.csv` file and your partition CSV file (the one that contains your project partitions)?

enthusisastSR, 2021-12-02:
Thank you @o-marshmallow. I have it working now. The problem was it was already initialized in my repository. So since I was trying to initialize it again I was running into this error. /data is custom partition type.

closed 2021-12-02 by enthusisastSR

o-marshmallow, 2021-12-03:
@enthusisastSR Great, thanks for coming back with the solution!