espressif/esp-idf #7992: Could not install package 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1' for 'linux_x86_64' system

SebiTimeWaster, 2021-12-01:
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
Error: Could not install package 'espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1' for 'linux_x86_64' system
The terminal process "platformio 'run'" terminated with exit code: 1.
```

it would be nice if the tool would tell me WHY it can't install it...

SebiTimeWaster, 2021-12-01:
also what does that mean:
```
Updating platformio/tool-scons                4.40200.0 @ ~4.40200.0             [Incompatible 4.40300.1]
```

igrr, 2021-12-01:
Hi @SebiTimeWaster, from the log it looks like you might be using platformio? I think it might be an issue to report on platformio forum or on the issue tracker (https://github.com/platformio/platform-espressif32/issues).

Alternatively you may use ESP-IDF without platformio, please see the getting started guides at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html.

closed 2021-12-01 by SebiTimeWaster