espressif/esp-idf #7993: ESP32-S3 ESP-IDF Bootloader Issue for Micropython

TGiles1998, 2021-12-01:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: ESP32S3-DevKitC-1
- Version: N8R8
- Module or chip used: ESP32S3-N8R8
- IDF version: v4.4-dev-3703-gddc44956bf
- Build System:         [Make|CMake|idf.py]
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:  macOS
- Using an IDE?: No from Bash
- Power Supply: USB

## Problem Description

Flashed the ESP32-S3 micropython onto the board and when connecting to a REPL. Either Screen or Pymakr results in:

Error:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0xf60
load:0x403b6000,len:0x978
load:0x403ba000,len:0x2c80
entry 0x403b616c
W (25) bootloader_random: RNG for ESP32-S3 not currently supported
W (238) bootloader_random: RNG for ESP32-S3 not currently supported

This seems to be an issue to do with the ESP-IDF.

Another forum talks about having to disable the SPIRAM in the menu config and I also did that as well but this did not help either. https://issueexplorer.com/issue/espressif/esp-dl/69

### Expected Behaviour

Should produce a working REPL e.g. >>>

### Actual Behavior

Error:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0xf60
load:0x403b6000,len:0x978
load:0x403ba000,len:0x2c80
entry 0x403b616c
W (25) bootloader_random: RNG for ESP32-S3 not currently supported
W (238) bootloader_random: RNG for ESP32-S3 not currently supported

### Steps to reproduce

1. Use ESP-IDF v4.4
2. Go to https://github.com/micropython/micropython
3. Fork
4. Then run: git submodule update --init
5. make -C mpy-cross V=1 clean all
6. cd ports/esp32
7. make submodules
8. make BOARD=GENERIC_S3
9. esptool.py -p /dev/cu.SLAB_USBtoUART -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 build-GENERIC_S3/bootloader/bootloader.bin 0x8000 build-GENERIC_S3/partition_table/partition-table.bin 0x10000 build-GENERIC_S3/micropython.bin
10. Open screen /dev/cu.SLAB_USBtoUART 115200
11. Press Enter Key
12. Press reset button on the dev board

Produces 

Error:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0xf60
load:0x403b6000,len:0x978
load:0x403ba000,len:0x2c80
entry 0x403b616c
W (25) bootloader_random: RNG for ESP32-S3 not currently supported
W (238) bootloader_random: RNG for ESP32-S3 not currently supported

igrr, 2021-12-01:
Hi @TGiles1998, I see you have opened the issue on micropython repository as well: https://github.com/micropython/micropython/issues/8053.

Do I understand you correctly that you are expecting the REPL to appear on UART console?

This isn't how GENERIC_S3 board is configured in micropython:

* mpconfigboard.cmake adds boards/sdkconfig.usb to SDKCONFIG_DEFAULTS: https://github.com/dpgeorge/micropython/blob/da4593f937003e5593725bfa61f727d429a1e061/ports/esp32/boards/GENERIC_S3/mpconfigboard.cmake#L3-L7
* sdkconfig.usb enabled USB CDC console: https://github.com/dpgeorge/micropython/blob/da4593f937003e5593725bfa61f727d429a1e061/ports/esp32/boards/sdkconfig.usb

So it looks like the REPL will be on USB CDC, not on UART.

TGiles1998, 2021-12-01:
Hi,

Thank you very much for your reply. Yes that was what I was expecting. I didn't realise that other REPL configurations were available. I will change this on my fork to UART and try and get it work that way.

TGiles1998, 2021-12-01:
Hi,

Yes I just removed the boards/sdkconfig.usb and this reset it back to UART and its working perfectly. Thank you very much!

closed 2021-12-01 by igrr

ruihua-code, 2022-04-08:
> Hi,
> 
> Yes I just removed the boards/sdkconfig.usb and this reset it back to UART and its working perfectly. Thank you very much!

I have the same problem, how to solve it，thk.