espressif/esp-idf #7994: first stage bootloader failure

phatpaul, 2021-12-01:
## Environment

- Module or chip used:  ESP32-WROVER-E (16M)
- IDF version: 3.3.5
- Build System:  Make
- Compiler version: 1.22.0-97-gc752ad5d
- Operating System:   Windows
- (Windows only) environment type: MSYS2 mingw32
- Power Supply:   external 3.3V

Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:e2:e6:4b:be:f8

## Problem Description

We've recently gotten at least 3 bricked devices returned from customers with exactly the same symptoms.
ESP module does not boot from flash.  It seems to be a failure of the first-stage bootloader.

This was reported a few months ago [on esp32.com](https://esp32.com/viewtopic.php?f=12&t=22938) but didn't get any attention from Espressif.  I then filled-out [the form here](https://www.espressif.com/en/contact-us/technical-inquiries/software-bug-report) a couple weeks ago:  But I didn't even get an email confirmation that it was received.

WiFive pointed out that the garbage load address 0xff001cff looks like a left-shifted version of the correct 0x3fff001c

### Steps to reproduce

I'm unable to reproduce the issue in the lab.  It has occurred in a small percentage ( ~ 0.1%) of units which have been sold and used by customers for up to 3 years.

### Code to reproduce this issue
My application is based on [this project](https://github.com/chmorgan/esphttpd-freertos ), which includes a OTA update mechanism.

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xff001cff,len:1809471
1150 mmu set 00010000, pos 00010000
1150 mmu set 00020000, pos 00020000
1150 mmu set 00030000, pos 00030000
1150 mmu set 00040000, pos 00040000
1150 mmu set 00050000, pos 00050000
1150 mmu set 00060000, pos 00060000
1150 mmu set 00070000, pos 00070000
1150 mmu set 00080000, pos 00080000
1150 mmu set 00090000, pos 00090000
1150 mmu set 000a0000, pos 000a0000
1150 mmu set 000b0000, pos 000b0000
1150 mmu set 000c0000, pos 000c0000
1150 mmu set 000d0000, pos 000d0000
1150 mmu set 000e0000, pos 000e0000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

## Other items

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7634874/sdkconfig.txt)

igrr, 2021-12-01:
Hi @phatpaul,

Could you please try to gather a bit more information, it will help us narrow down the issue.

1) could you use `esptool.py read_flash_status` to examine flash chip status register on the unit which has this issue, and post the output here?
2) could you use `esptool.py read_flash` to dump the the data between 0x1000 — 0x1030? It was mentioned by another user on the forum that the data in flash was correct, just to double check it's the same in your case.

phatpaul, 2021-12-01:
Hey Ivan,
I did previously dump all of the parts of flash and compared with a working unit.  My notes indicate that they were identical in the bootloader (0x1000 - 0x6300) and factory app (0x10000 - 0x40000) sections.
I only did this on the first returned unit.  I'll do it again on another failed unit to confirm.

But I didn't try the `read_flash_status`.  I'll do that when I'm back in my lab on Friday.
Thanks

fizzyade, 2022-01-09:
I am experiencing what I think is the same problem, I have 4 units built and under test for less than 24 hours and I noticed that one stopped working reporting data, upon connecting it to usb I could see it stuck in a boot loop and so I've been trying to figure out what the problem is all day without success.

When it boots it shows this (this is a repeated message, but there's an identical one on reset giving the power on reset reason code)

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7084
ho 0 tail 12 room 4
load:0x07800000,len:3662912
ets Jul 29 2019 12:21:46
```

note the load address, this is wrong, a working board shows:

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7084
ho 0 tail 12 room 4
load:0x40078000,len:14308
load:0x40080400,len:3716
0x40080400: _init at ??:?
```

Like the OP, the load: address that should be 0x40078000 seems to be a byte out on the broken one.

I have refreshed many times today to try and resolve this, but nothing works, I can verify the flash contents with:

```
esptool.py verify_flash --diff yes 0x1000 build/bootloader/bootloader.bin

esptool.py v3.1-dev
Found 5 serial ports
Serial port /dev/cu.wchusbserial1450
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:ab:95:47:29:18
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Verifying 0x6270 (25200) bytes @ 0x00001000 in flash against build/bootloader/bootloader.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...

esptool.py verify_flash --diff yes 0x10000 build/hugh.bin

esptool.py v3.1-dev
Found 5 serial ports
Serial port /dev/cu.wchusbserial1450
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:ab:95:47:29:18
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Verifying 0xeda20 (973344) bytes @ 0x00010000 in flash against build/hugh.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...

esptool.py verify_flash --diff yes 0x8000  build/partition_table/partition-table.bin

esptool.py v3.1-dev
Found 5 serial ports
Serial port /dev/cu.wchusbserial1450
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:ab:95:47:29:18
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Verifying 0xc00 (3072) bytes @ 0x00008000 in flash against build/partition_table/partition-table.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...
```

And as you can see from the output, the contents match what was written.

And here's the read-flash_status:

```
sptool.py read_flash_status
 
esptool.py v3.1-dev
Found 5 serial ports
Serial port /dev/cu.wchusbserial1450
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:ab:95:47:29:18
Uploading stub...
Running stub...
Stub running...
Status value: 0x7bfc
Hard resetting via RTS pin...
````

and the flash id:

```
esptool.py flash_id     
    
esptool.py v3.1-dev
Found 5 serial ports
Serial port /dev/cu.wchusbserial1450
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:ab:95:47:29:18
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4018
Detected flash size: 16MB
Hard resetting via RTS pin...
```

negativekelvin, 2022-01-09:
@fizzyade your strapping config differs between your boards boot:0x13 vs boot:0x33.  It may not be relevant but do you have flash voltage configured in efuse?

fizzyade, 2022-01-09:
> @fizzyade your strapping config differs between your boards boot:0x13 vs boot:0x33.  It may not be relevant but do you have flash voltage configured in efuse?

Yes, sorry, the same error occurs regardless of whether GPIO12 is left floating or tied, once the board died after about 8 hours of trying to figure out what was going on, I tried strapping GPIO12 because there was nothing left to lose at that point, I just inadvertently copied the text and pasted it here from a different strapping test.

When the board failed and I plugged it into the PC, the strapping was 0x33.

The flash actually reads and writes correctly regardless of the strapping.

It's the offset of 1 byte that was the original posters fault they were reporting, I'm also seeing this, although I've since destroyed that module now and replaced it with a different one that booted straight up with the same code.

negativekelvin, 2022-01-10:
> The flash actually reads and writes correctly regardless of the strapping.

@fizzyade ok well you should still make sure you are running it at the correct voltage spec.  boot:0x33 means you are running it at 1.8v unless you override it with efuse setting.

fizzyade, 2022-01-10:
> > The flash actually reads and writes correctly regardless of the strapping.
> 
> 
> 
> @fizzyade ok well you should still make sure you are running it at the correct voltage spec.  boot:0x33 means you are running it at 1.8v unless you override it with efuse setting.

Yes, as I said, I tried everything including running at 1.8 and 3.3 volts, I just happened to copy the results from that experiment, by the time I'd got to this point, I was clutching at straws and tried everything, including the changing that level.

By the time i changed the strapping I had spent at least 6 hours trying to get past the first stage bootloader.

No e-fuses have been changed.  The system was up and collecting and sending data for about 12 hours before it stopped.

The error I'm seeing existed when both a working and this dead board were connected in exactly the same way with regards to strapping and well, everything, as they are identical.

phatpaul, 2022-01-19:
Pardon the delay, here's the extra info you requested @igrr. 
### Failed unit 1 (MAC: e0:e2:e6:4b:be:f8)

Unit 1 try to boot:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xff001cff,len:1809471
1150 mmu set 00010000, pos 00010000
1150 mmu set 00020000, pos 00020000
1150 mmu set 00030000, pos 00030000
1150 mmu set 00040000, pos 00040000
1150 mmu set 00050000, pos 00050000
1150 mmu set 00060000, pos 00060000
1150 mmu set 00070000, pos 00070000
1150 mmu set 00080000, pos 00080000
1150 mmu set 00090000, pos 00090000
1150 mmu set 000a0000, pos 000a0000
1150 mmu set 000b0000, pos 000b0000
1150 mmu set 000c0000, pos 000c0000
1150 mmu set 000d0000, pos 000d0000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xff001cff,len:1809471
1150 mmu set 00010000, pos 00010000
1150 mmu set 00020000, pos 00020000
1150 mmu set 00030000, pos 00030000
1150 mmu set 00040000, pos 00040000
1150 mmu set 00050000, pos 00050000
1150 mmu set 00060000, pos 00060000
1150 mmu set 00070000, pos 00070000
1150 mmu set 00080000, pos 00080000
1150 mmu set 00090000, pos 00090000
1150 mmu set 000a0000, pos 000a0000
1150 mmu set 000b0000, pos 000b0000
1150 mmu set 000c0000, pos 000c0000
1150 mmu set 000d0000, pos 000d0000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2)
)
waiting for download
```
Unit 1 read_flash_status:
```
# python esp-idf/components/esptool_py/esptool/esptool.py -p COM42 read_flash_status
esptool.py v2.9-dev
Serial port COM42
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:e2:e6:4b:be:f8
Uploading stub...
Running stub...
Stub running...
Status value: 0x5aec
Hard resetting via RTS pin...
```
Unit 1 flash_id:
```
# python esp-idf/components/esptool_py/esptool/esptool.py -p COM42 flash_id
esptool.py v2.9-dev
Serial port COM42
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:e2:e6:4b:be:f8
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4018
Detected flash size: 16MB
Hard resetting via RTS pin...
```

I also did `python esp-idf/components/esptool_py/esptool/esptool.py -p COM42 -b 921600 read_flash 0x1000 0x6300 flash_dump-bootloader.bin` and compared the dump to what I had programmed at production time.  There are no differences.

### Failed unit 2 (MAC: e0:e2:e6:4b:ba:2c)
Unit 2 (failed) boot message is the same as unit 1.

Unit 2 read_flash_status is different:
```
# python esp-idf/components/esptool_py/esptool/esptool.py -p COM42 read_flash_status
esptool.py v2.9-dev
Serial port COM42
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: e0:e2:e6:4b:ba:2c
Uploading stub...
Running stub...
Stub running...
Status value: 0x7bfc
Hard resetting via RTS pin...
```
Unit 2 flash_id produced the same result as unit 1.
Unit 2 Dump and compare of bootloader is also identical to production file as in unit 1.

Do you have any idea of what's going wrong yet?

ginkgm, 2022-01-26:
Hi @phatpaul ,

Thanks for providing the information.

[image]
[image]

According to your information, there seems to a quite a few status register bits being written to 1 for some reasons. You can see the difference if you read the status of a working unit. The status should be something like 0x0200. (only QE is expected to be written).

Though not printed, I doubt SR3 is also been written to something bad, in which there are DC[1, 0] bits used to control the timing. When these bits are set, the ROM will not be able to read the flash correctly.

[image]

Could you confirm this by running `esptool.py read_flash_status --bytes 3`?

fizzyade, 2022-01-26:
What is the cause of this status register being bad? 

Both the OP and myself have identical status registers.

In my case, the board was working and was left alongside 3 other identical systems, all battery powered, and were working for 12 or so hours without being touched, as they were publishing measurements via MQTT, I noticed that suddenly at 10pm one unit stopped working.

When I removed the unit and connected it to USB, I saw the boot loop.  I spent hours trying to resolve it, but eventually the only solution was to put the board in the bin, it was dead.

Like the OP I appeared to be seeing an offset of one byte when the first stage bootloader tried to load the partitions, obviously the second one caused a problem because the length and address were erroneous, causing the bootloader to reset ad infinitum.

What's most concernig about all of this is that it happened on a unit within a few hours of powering on, and without any interaction or handling of any kind, it died.

phatpaul, 2022-01-28:
We just got another unit returned bricked.  After cutting it open for analysis, the serial port dumps the same output as the previous failed units.

As requested I ran the additional command 
```
python esp-idf/components/esptool_py/esptool/esptool.py -p COM42 read_flash_status --bytes 3
...
Status value: 0xe37bfc
```

This is the 4th failed unit we have received with this issue.  This represents most of our field failures.  Is there any further troubleshooting you would like me to do?  I'll be happy to send in the units for failure analysis.
Thanks,
Paul

ginkgm, 2022-01-30:
Hi @fizzyade , @phatpaul 
Thanks for providing these information.

Actually we have also observed some flash status written to all 1's issues from different vendors. These issues are reported more on applications whose power supply is not stable. I guess it's because when the status is being written, there is an intermediate state that all bits are 1's. But if power drop at this moment, the flash can't set the status to the correct value. ESP ROM doesn't have the ability to restore the DC bits (because these bits are vendor-specific), so when the timing is incorrect, it will be not able to boot any more.

Fixing random issue is long process. If you are urgent about mass production, I propose using SRP to protect these bits: setting SRP1 = 1 while keeping SRP0 = 0 at app startup. This will forbid further SR change until next power cycle.

[image]

btw, I'm also interested in whether they can be recovered by factory fix: running `esptool.py write_flash_status --bytes 3 --non-volatile 0`

fizzyade, 2022-01-30:
Hi, thanks for the reply.

As I said, the unit was powered by battery and was working and reporting data every 30 minutes, it had been running under test for somewhere around 10-12 hours, the following day I was looking at the data that had been published (MQTT) via influxdb and grafana and noticed that this one unit had been running fine from battery, suddenly stop reporting.

The unit had no connection to it other than the battery power, nothing was in programming mode and the unit had been running fine up until it failed.

Once I pulled the unit, I plugged it into USB where I was greeted by this boot loop, I spent a day trying to recover it and failing before I found this report here which was describing exactly what I was seeing here.

As far as I'm concerned, nothing should have been changing status bytes, I assume that should only be possible in programming mode.

The flash read out correctly, there was no problem there.

I'm using these in a hobby situation, but I have also plans to use them in a commercial situation, and seeing a failure like this after 12 hours has basically stopped me exploring that any further until there is some sort of explanation and mitigation of whatever caused this problem.

I'm even more nervous as the OP now has multiple units with this problem and like me, I can only assume that there are no programming operations actually being done.

I don't believe I have missing anything regarding the layout/design, I have tried both 1.8 and 3.3 strapping, and at least on my board there was no difference, both worked and I was able to read and write flash correctly regardless of the voltage strapping. (These tests were done after the unit failed towards the end of the day of trying to resolve it)

If there is a possibility of causing an issue like this with strapping pins, then if you tell me what I should be looking for I can compare against my design.

I'm not able to perform any tests on the dead module because I got so frustrated with the problem that I ended up physically destroying it in the end.

> Hi @fizzyade , @phatpaul
>
> Thanks for providing these information.
>
>
>
> Actually we have also observed some flash status written to all 1's issues from different vendors. These issues are reported [...]

fizzyade, 2022-01-30:
> Hi @fizzyade , @phatpaul Thanks for providing these information.
>
> Actually we have also observed some flash status written to all 1's issues from different vendors. These issues are reported more on [...]

Also, this protection of the status register appears to be some form of setting on bits when the application has been booted.  What happens if the power issue you mention occurs before then and results in the dead flash.

If I write to these bits once from the application, are you saying they will remain in that state until power is physically removed, is power removal the only thing that can reset them?  I ask as I used deep sleep, and wake up every 30 minutes, so if I set these protection bits after the first boot, will they remain set when the device wakes from deep sleep or will the deep sleep have caused them to reset?

fizzyade, 2022-01-31:
> No luck with write_flash_status:
> # python esp-idf/components/esptool_py/esptool/esptool.py -p COM42 write_flash_status --bytes 3 --non-volatile 0
> Initial flash status: 0xe37bfc
> Setting flash status: 0x000000
> After flash status:   0xe37bfc
> Unit still won't boot after power-cycle.
> I don't have any experience working directly with NAND flash chip, so pardon my naive questions:
> You said:
> • Why is it that I can still read and program the flash device using the esptool.py, but the bootloader can't read it? Don't they use the same (ROM) mechanism internally?
> You suggested:
> • > Would your suggestion cause any side-effects like not being able to write to the flash data? If not, then why doesn't the bootloader do this be default?
> • > How is it that the flash status bytes don't reset on power-cycle? Are they non-volatile? Why is the ESP frequently changing them?
> • > Please provide me some example C code to achieve the SRP lock. (ideally working within ESP-IDF 3.3)

This made me feel slightly better that I destroyed my failed unit, with the suggestion not working.

I really don’t understand how this “corruption” has happened, my device boots, reads some sensors, publishes it via MQTT and then goes into deep sleep until it wakes next.

I am not writing to flash or any non volatile storage in the chip itself at any point, unless it’s happening somewhere outside of my code.  The fact that this has happened, and that Paul has multiple units exhibiting this fault is making me very nervous about using this in the future.

Is there a reason why when the first stage bootloader reads the partition map that both myself and Paul are seeing a shift of one byte which results in the bootloader attempting to load something from an address and size that is garbage.

I can’t understand how exactly this failure mode is even possible, because if I read the flash I can verify it is correct, everything appears to be in the right place.

I wonder how prevalent this issue actually is, I did find a number of posts elsewhere where other people had his this issue but there was either no resolution or no replies.

Failing to boot the first stage bootloader when the flash is correct is a very worrying problem.

igrr, 2022-01-31:
(Our team in Shanghai is on Lunar New Year holiday this week, so I'll reply to some of the questions raised in the meantime.)

> Why is it that I can still read and program the flash device using the esptool.py, but the bootloader can't read it? Don't they use the same (ROM) mechanism internally?

The most likely reason is that different SPI flash modes are used in these two cases. I think esptool is using the FASTRD mode while ROM is using DIO mode (`mode:DIO, clock div:1` in your boot log).

> Would your suggestion cause any side-effects like not being able to write to the flash data? If not, then why doesn't the bootloader do this be default?

The status register is written at startup (in the 2nd stage bootloader) to enable QIO mode, if the bootloader was built with QIO mode enabled. This also happens in the application startup, if the mode is still DIO but the application was built with QIO mode enabled. (The latter step is there so that you can initially ship the device with 2nd stage bootloader & app both using DIO mode, and later upgrade the application enabling QIO mode for it.)

For this reason, the ROM bootloader can't enable Power Supply Lock-Down of the status registers, since the 2nd stage bootloader or the application may decide to change the operating mode of the flash chip.
Enabling this feature after the app has started should be possible, though. One thing I'll leave to @ginkgm to confirm is how this will interact with Write Enable (WEL) bit (which is in SR0) — looks like we would need to set WEL before writing SRP?

> How is it that the flash status bytes don't reset on power-cycle? Are they non-volatile? Why is the ESP frequently changing them?

Status registers of an SPI flash chips have both volatile and non-volatile bits. For example, QIO mode enable bit and various write/erase protection bits are all non-volatile. The only bit which is changed frequently in recent ESP-IDF versions is WEL (write enable). WEL is disabled between flash program or erase operations.

> I am not writing to flash or any non volatile storage in the chip itself at any point, unless it’s happening somewhere outside of my code.  The fact that this has happened, and that Paul has multiple units exhibiting this fault is making me very nervous about using this in the future.

When you use Wi-Fi on ESP32, by default some connection parameters (such as SSID, password, BSSID, and so on) are saved into flash using [NVS library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html). More on NVS usage by Wi-Fi driver [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-nvs-flash). As mentioned above, writing into flash requires writing to the flash status register ("write enable", WEL bit).

>  Is there a reason why when the first stage bootloader reads the partition map that both myself and Paul are seeing a shift of one byte which results in the bootloader attempting to load something from an address and size that is garbage.

It seems that some of the bits in SR3 register which affect the timing of DIO mode are erroneously set. We don't know the cause for sure yet, @ginkgm suggests above that the following could be the reason:

> These issues are reported more on applications whose power supply is not stable. I guess it's because when the status is being written, there is an intermediate state that all bits are 1's. But if power drop at this moment, the flash can't set the status to the correct value.

Note that "power drop" above does not necessarily mean that the power supply is completely lost. If the supply voltage drops below the operating range of the SPI flash chip (2.7-3.6V), the issue may also happen. At the same time, ESP32 may keep working because its supply voltage range is wider (2.3-3.6V).

> I can’t understand how exactly this failure mode is even possible, because if I read the flash I can verify it is correct, everything appears to be in the right place.

The most likely reason is that the ROM bootloader and esptool are using different SPI flash modes. esptool is using the FASTRD mode while ROM is using DIO mode. It looks like the timing in DIO mode could be affected by some of the status register bits which get erroneously set. Therefore the reading process in ROM bootloader is affected, but esptool (which is using the slower "FASTRD" mode) is not.

negativekelvin, 2022-01-31:
W25Q128FV datasheet shows the "dummy control" bits as reserved and says they can't be written during a WSR operation anyway so if they are getting set erroneously it would be due to some internal glitch in the flash chip, not over the spi bus.  But it should be possible to observe the effect on a spi scope capture if it is happening.  Another test would be swap the flash chips between a known good and bad module and see what happens.

igrr, 2022-02-01:
I think mfg=20h id=4018 is XM25QH128. Its datasheet does show the DC0 & DC1 bits in SR3 ([this diagram posted above](https://user-images.githubusercontent.com/18559742/151087513-7ba370d8-39a5-4944-abda-7927fc3fa8ab.png)).

When DC1=1 and DC0=1 (as seen in flash status readout), there are 8 dummy clock cycles in DIO mode instead of the default 4 clock cycles. However the number of dummy clock cycles in FASTRD doesn't change, it's 8 cycles regardless of DC bits setting. This results in data being "shifted" in DIO mode, but not FASTRD.

> No luck with write_flash_status:

Unfortunately this didn't work due to the above mentioned SRP0 and SRP1 bits already being set to 1, 1 — which means that the status register is permanently locked (OTP mode).

There is no way the flash chip can be recovered, but we should find a way to prevent this happening on more devices.

The status register value — 0xe37bfc shows that indeed, as @ginkgm has suggested, all bits of all status registers got set to 1:

SR3 = 0xe3: HOLD/RST, DRV1, DRV0, DC1, DC0 bits set, remaining reserved bits are read back as 0
SR2 = 0x7b: CMP, LB3-LB1, QE, SRP1 bits set, remaining reserved bits are read back as 0
SR1 = 0xfc: SRP0, SEC, TB, BP2-BP0 bits set, WEL and BUSY are read back as 0.

@phatpaul could you please check, does your application sdkconfig have brownout detector enabled, and if yes, at which threshold? (`CONFIG_ESP32_BROWNOUT_DET_LVL` option).

negativekelvin, 2022-02-01:
>I think mfg=20h id=4018 is XM25QH128

Ah ok.  

>The status register is written at startup (in the 2nd stage bootloader) to enable QIO mode

Datasheet says:
>When the QE bit is set to a 1(factory default for Quad Enabled part numbers with ordering option “Q”), the Quad IO2 and IO3 pins are enabled, Andrew /WP and /HOLD functions are disabled....QE bit is required to be set to a 1 before issuing an “Enter QPI (38h)” to switch the device from Standard/Dual/Quad SPI to QPI

So is there any need to toggle this bit?

>The only bit which is changed frequently in recent ESP-IDF versions is WEL (write enable). WEL is disabled between flash program or erase operations.

Datasheet says: 

> Write Enable Latch (WEL) is a read only bit

So is there any reason the status registers can't be OTP to a good value?

citiq, 2022-03-09:
For the record, our ESP32 module seems to be bricked the same way - having the 16MB flash chip with OTP lock:

```
# esptool flash_id
esptool.py v3.1
…
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:3c:c6:2c:5b:44
Stub is already running. No upload is necessary.
Manufacturer: 20
Device: 4018
Detected flash size: 16MB
```

Status value has all bits set:
```
# esptool read_flash_status --bytes 3
…
Status value: 0xe37bfc
```

> There is no way the flash chip can be recovered, but we should find a way to prevent this happening on more devices.

> "using SRP to protect these bits: setting SRP1 = 1 while keeping SRP0 = 0 at app startup. This will forbid further SR change until next power cycle."

(How) Can we set or reset the SRP bits from userspace?

negativekelvin, 2022-03-09:
spi_flash_common_write_status_16b_wrsr

phatpaul, 2022-03-15:
@igrr said: could you please check, does your application sdkconfig have brownout detector enabled, and if yes, at which threshold? (CONFIG_ESP32_BROWNOUT_DET_LVL option).
```
CONFIG_BROWNOUT_DET=y
CONFIG_BROWNOUT_DET_LVL=3
```

But I have also added an external Voltage Supervisor because we had some issues with brownout causing the ESP to lockup (recoverable with a power-cycle).
[image]

persan666, 2022-03-30:
I have the exact same problem. When reading the status register I get:
# esptool.py read_flash_status --bytes 3
…
Status value: 0xe37bfc

I have now 17 units returned from customer with this problem.

I am using a esp32-Wrover-E with esp-idf 4.4 on a battery powered custom board. We have earlier used a esp32-Wrover-B with a external reset circuit, resetting at 2.7V. The internal brown-out is also set to 2.43V and only used if the external circuit fails. For some reason we have failed to notice that the supported supply voltage has changed from 2.7V to 3.0V. Could it be so that when the battery is near empty and the 3.3V regulator is unable to supply a voltage above 3.0V, a write error may occur to the status register and it gets locked?.  

My hope to save units already delivered was to change the brown-out detection level to 3.0V but the highest value one can set is 2.8V. Shouldn't it be possible to set 3.0V?

I am able to flash and verify data with the esptool that is using FASTRD. Is it possible in some way to force the bootloader and the application to use FASTRD instead of DIO?

This is very alarming and i am very worried, I have 10000+ delivered to customers.

persan666, 2022-04-04:
Any news on this? What is the proposed way to handle already delivered devices? Increase the brownout detection level to 2.8V? Write to the status register to lock it in a working state?
All comments are appreciated

persan666, 2022-04-27:
@negativekelvin Would you suggest that the solution to the problem would be to implement the code snippet that @winzkigermany got from Espressif? If yes, where should it be injected? I can see from the code snippet that it is called in the beginning of main. It also suggest to put the main function i IRAM. Maybe the snippet is meant to be run as a separate program in production, before the real program is flashed. But for already delivered devices we need an implementation that we can upgrade the devices with.

winzkigermany, 2022-04-27:
I call xmc_flash_fix() in my app_main(), quite early. After this fix has run once in the lifetime of the XMC flash chip, you're done. The code is not beautiful, but hey, the problem isn't either.

Beware I: If you NEED some write protect bits of the flash chip for custom write protection, you should obviously not run this code unaltered. This is just a direct copy&paste from my dev project with NO guarantee for completeness or fit for any purpose.

Beware II: We're working ONLY with WROVER-E modules, so I can't say anything about other ESP versions.

Take this with a grain of salt and good luck for your projects!

`/*
 * This file contains the XMC flash fix against status register corruption
 */

#include "esp_log.h"        // required for esp_log_timestamp()

#include "m3u_processor.h"
#include "esp_task_wdt.h"
#include "esp_spi_flash.h"
#include "esp32/rom/spi_flash.h"

#include "xmc_flash_fix.h"

static const char *TAG = "xmc_flash_fix";

/**
 *  Copy from execute_flash_command() since it is static function
 */
#define SPIFLASH SPI1
static IRAM_ATTR uint32_t bootloader_execute_flash_command(uint8_t command, uint32_t mosi_data, uint8_t mosi_len, uint8_t miso_len)
{
    uint32_t old_ctrl_reg = SPIFLASH.ctrl.val;
#if CONFIG_IDF_TARGET_ESP32
    SPIFLASH.ctrl.val = SPI_WP_REG_M; // keep WP high while idle, otherwise leave DIO mode
#else
    SPIFLASH.ctrl.val = SPI_MEM_WP_REG_M; // keep WP high while idle, otherwise leave DIO mode
#endif
    SPIFLASH.user.usr_dummy = 0;
    SPIFLASH.user.usr_addr = 0;
    SPIFLASH.user.usr_command = 1;
    SPIFLASH.user2.usr_command_bitlen = 7;

    SPIFLASH.user2.usr_command_value = command;
    SPIFLASH.user.usr_miso = miso_len > 0;
#if CONFIG_IDF_TARGET_ESP32
    SPIFLASH.miso_dlen.usr_miso_dbitlen = miso_len ? (miso_len - 1) : 0;
#else
    SPIFLASH.miso_dlen.usr_miso_bit_len = miso_len ? (miso_len - 1) : 0;
#endif
    SPIFLASH.user.usr_mosi = mosi_len > 0;
#if CONFIG_IDF_TARGET_ESP32
    SPIFLASH.mosi_dlen.usr_mosi_dbitlen = mosi_len ? (mosi_len - 1) : 0;
#else
    SPIFLASH.mosi_dlen.usr_mosi_bit_len = mosi_len ? (mosi_len - 1) : 0;
#endif
    SPIFLASH.data_buf[0] = mosi_data;

    if (g_rom_spiflash_dummy_len_plus[1]) {
        /* When flash pins are mapped via GPIO matrix, need a dummy cycle before reading via MISO */
        if (miso_len > 0) {
            SPIFLASH.user.usr_dummy = 1;
            SPIFLASH.user1.usr_dummy_cyclelen = g_rom_spiflash_dummy_len_plus[1] - 1;
        } else {
            SPIFLASH.user.usr_dummy = 0;
            SPIFLASH.user1.usr_dummy_cyclelen = 0;
        }
    }

    SPIFLASH.cmd.usr = 1;
    while (SPIFLASH.cmd.usr != 0) {
    }

[... 85 lines cut ...]
    esp_err_t retStatus = ESP_FAIL;

    // if we have an XMC flash chip susceptible to status register corruption,
    // flash the status registers properly and lock them afterwards.
    if( g_rom_flashchip.device_id == 0x204018 ){

        TaskHandle_t cur_core_idle_handle = xTaskGetIdleTaskHandle();
        retStatus = esp_task_wdt_delete(cur_core_idle_handle);  //Disable watchdog
        if (retStatus == ESP_OK) {

            g_flash_guard_default_ops.start(); //Disable cache
            retStatus = esp_xmc_16m_flash_sr_otp(&g_rom_flashchip);
            g_flash_guard_default_ops.end();  //Enable cache

            esp_task_wdt_add(cur_core_idle_handle);
        }

        printf( "\nI (%u) %s: Flash chip id is 0x204018, esp_xmc_16m_flash_sr_otp() return code is: %u\n", (unsigned) esp_log_timestamp(), TAG, retStatus);
    } else {
        printf( "\nI (%u) %s: Flash chip is NOT 0x204018\n", (unsigned) esp_log_timestamp(), TAG);
    }

}

`

fizzyade, 2022-04-27:
Come on espressif, you need to not only provide a solution here in this issue (actually, it’s way overdue) that prevents this from happening.

As a hobbyist, the fact I saw this within 12 hours of my initial soak test of 4 units has put me of considering using this chip in the future, certainly not for any commercial projects, and I’ll seriously think twice for my own projects as populating boards by hand is time consuming and having a dead module is not something I want to contend with.

Adrian

persan666, 2022-04-28:
Thanks! @winzkigermany for sharing your implementation. 
I have made some test but I struggle with that I get the following Watchdog reset:
`rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`
It seems to be happening when the following code is called:
`g_flash_guard_default_ops.start(); //Disable cache`
If I print something before and after, I only see the first message

winzkigermany, 2022-04-28:
You're welcome. 
Are you on idf 4.4.x ?
I'm struggling with some (heap corruption?) problem there right now:
https://github.com/espressif/esp-idf/issues/8558

Don't know how to work around that bug atm.
The Code I pasted above runs on 4.3.x without problems.

persan666, 2022-04-28:
Aha, yes I am running 4.4

persan666, 2022-04-28:
@winzkigermany have you experienced that this problem can occur even if the VCC/Enable is above 3.0v?

winzkigermany, 2022-04-28:
We did not try to reproduce it under lab conditions (and we don't want to wreck even more of our boards), because we were busy finding a quick fix with 3000 modules in the field already. But since the majority of bricked modules occurred after the typical lifetime of a bunch of batteries in our product, I suspect the bricking *might* happen when VCC is below some threshold.

persan666, 2022-05-03:
Any updates on this?

phatpaul, 2022-05-03:
I'm glad to see there is hope in preventing this.  Thanks @winzkigermany for sharing your code.  I need to study it and fully understand what it is doing and adapt it to my needs.  Should I assume that I should permanently write the same value that I read with `esptool read_flash_status --bytes 3`?  (a good unit reads: 0x400200)  
Also, I'm still on IDF 3.5, so hopefully this code works without too much hassle...

winzkigermany, 2022-05-04:
@phatpaul 
no, writing 0x400200 does not do the trick.
The problem is that somehow the registers SR1..SR3 get overwritten with garbage.
This garbage also **write protects** those registers once and for all. _THAT_ is the real problem.

So in my code, I check if it is a flash chip that's susceptible to this problem (chip id 0x204018),
then I write the 3 registers, and ALSO LOCK them forever, but this time with sensible values.
So I slam the door shut before the bug can do it.

In my case, the correct register values are:
```
uint8_t new_sr1_status = 0x80;  // Set SRP0, clear other protect bits
uint8_t new_sr2_status = 0x03;  // Set QE and SRP1      // we *could* just update sr2_status. But we don't.     =  ((uint8_t)sr2_status | SR2_SRP1_MASK | SR2_QE_MASK);
uint8_t new_sr3_status = 0x60;  // Set output driver strength to 25%, which is the default (and good enough for us)
```

To get into my code, start with 
xmc_flash_fix()
which simply calls the helper methods that are declared above it.

hth
Rainer

fizzyade, 2022-05-04:
It’s incredible that there has been no further comment here from espressif for a bug that bricks devices.

I dread to think how many devices out there have bricked themselves, I happened across the bug on just the third device I programmed for a soak test, it died within 12 hours.

Come on, this is poor, if they know of this workaround (which was supplied by them privately to somebody who hit it) then why in earth has this not been updated, why is there not a massive warning about this problem and that you really need to program those bits to prevent it, and that those bits are write once, so choose their state wisely.

It shouldn’t be us sitting here discussing a bug that bricks devices and having to figure out how to apply the secret workaround.

Why is there no protection mechanism around those bits that require you to unlock via a command before you are able to write them, this is something that I have seen across many microcontrollers when it comes to “fuses”.

I can’t believe there’s a bug report here, that is so fatal for devices that its 5 months old with no official solution, with the occasional reply before they disappear for slither month or two, I’d be bending over backwards to provide a fix or workaround if I had such a fatal problem.

Adrian

persan666, 2022-05-05:
I have made some tests with idf 4.4 and I was unable to run @winzkigermany example code due to that it failed to disable flash cache for unknown reason. However I was able to run the code before the task scheduler started by injected it in the file esp_flash_api.c in the function as below. The result of the status registers has been verified with this esptool command:

**esptool.py read_flash_status --bytes 3**

Sorry for the formating of the code, it seems to be some problem with the code snippet tool.  

`static esp_err_t IRAM_ATTR read_id_core(esp_flash_t* chip, uint32_t* out_id, bool sanity_check)
{
    bool installed = esp_flash_chip_driver_initialized(chip);
    esp_err_t err = rom_spiflash_api_funcs->start(chip);
    if (err != ESP_OK) {
        return err;
    }
    esp_err_t retStatus = ESP_FAIL;

    // if we have an XMC flash chip susceptible to status register corruption,
    // flash the status registers properly and lock them afterwards.
    if( g_rom_flashchip.device_id == 0x204018 ){
        retStatus = esp_xmc_16m_flash_sr_otp(&g_rom_flashchip);
    }

    esp_err_t (*read_id_func)(void*, uint32_t*);
    void* read_id_arg;
    if (installed && chip->chip_drv->read_id) {
        read_id_func = (void*)chip->chip_drv->read_id;
        read_id_arg = (void*)chip;
    } else {
        //default option if the chip is not detected/chosen yet.
        read_id_func = (void*)chip->host->driver->read_id;
        read_id_arg = (void*)chip->host;
    }

    // Inner function fails if it sees all-ones or all-zeroes.
    err = read_id_func(read_id_arg, out_id);

    if (sanity_check && err == ESP_OK) {
        // Send RDID command twice, check for a matching result and retry in case we just powered on
        uint32_t new_id;
        err = read_id_func(read_id_arg, &new_id);
        if (err == ESP_OK && (new_id != *out_id)) {
            err = ESP_ERR_FLASH_NOT_INITIALISED;
        }
    }

    esp_err_t esp_err = ESP_FAIL;

    esp_err = rom_spiflash_api_funcs->end(chip, err);

    printf("retStatus = 0x%x\n", retStatus);
    printf("sr1_status = 0x%x\n", sr1_status);
    printf("sr2_status = 0x%x\n", sr2_status);

    return esp_err;
}`

phatpaul, 2022-05-09:
I was able to reproduce the issue in the lab.
I probed several units that were in process of assembly with `esptool.py flash_id` and found a mix of Manufacturer codes '0xC8' and '0x20'.  

I took one with flash id '0x20' and powered with my lab power supply at about 4.5V.  The 3.3V buck regulator begins to drop-out at that low input voltage.  The ESP32 is attempting to start-up but when it draws a bit more current, the VCC drops and the BD5228-TR voltage supervisor shuts-off the EN pin.  Once the ESP32 is stopped, the VCC recovers and the cycle repeats every second or so.
After about 20 seconds, I tested the unit at normal input voltage and it was indeed bricked with the same symptoms as above.

Then I took a fresh unit (also with flash id 0x20) I tried to implement the code example provided by @winzkigermany but I'm having trouble getting it working on my IDF v3.3.6.  I had to swap some header includes: 
```
//#include "m3u_processor.h" // ??
#include "esp_task_wdt.h"
#include "esp_spi_flash.h"
#include "rom/spi_flash.h"  // was #include "esp32/rom/spi_flash.h"
#include "soc/spi_struct.h" // also seems needed in IDF 3.3
extern uint8_t g_rom_spiflash_dummy_len_plus[]; // also seems needed in IDF 3.3
```
That compiled OK, but when run it crashes:
```
W (156) xmc_flash_fix: Flash chip id is 0x204018, attempting to apply workaround.
Guru Meditation Error: Core  0 panic'ed (Double exception)
Core 0 register dump:
PC      : 0x400803c0  PS      : 0x00050736  A0      : 0x8010ce05  A1      : 0x3fffffa8  
0x400803c0: _DoubleExceptionVector at /home/xxx/tools/esp-idf/components/freertos/xtensa_vectors.S:543

A2      : 0x40000018  A3      : 0x00000000  A4      : 0x3ffcaebc  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x3ffb7fec  A8      : 0x800881a8  A9      : 0xd1126148  
A10     : 0x00000006  A11     : 0x3ffcaec0  A12     : 0x00003400  A13     : 0x51384128  
A14     : 0x3ffbbe04  A15     : 0x3ffc8fe4  SAR     : 0x00000018  EXCCAUSE: 0x00000002  
EXCVADDR: 0xd1126148  LBEG    : 0x40092341  LEND    : 0x40092351  LCOUNT  : 0xfffffffc  
0x40092341: strlen at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84

0x40092351: strlen at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96

ELF file SHA256: 5194fd04eaec9eec

Backtrace: 0x400803bd:0x3fffffa8 |<-CORRUPTED
0x400803bd: _UserExceptionVector at ??:?

Rebooting...

```

So I just set it with:
`esptool.py write_flash_status --bytes 3 --non-volatile 0x600380`
And I repeated the low-voltage brownout test the same as above.  I let it go for 20 minutes and still not bricked!

So that confirms the solution for me, now I just could use some help porting the code to work in my project.

persan666, 2022-05-10:
@phatpaul Did you manage to see at which voltage level the flash got bricked? I wonder if the built in brownout detect level of 2.8 is enough to prevent bricking the flash?

winzkigermany, 2022-05-10:
@persan666 It's not. At least it was not for us in the field. You would not want to rely on that and surprisingly end up with bricked devices in the field.

phatpaul, 2022-05-10:
I was watching the VCC on my oscilloscope, and it was lower (~3.0V) than normal (3.3V) because of the drop-out of the regulator (the lab power-supply is set to 4.57V, which is not high enough to run normally).  Then there is a repeating depression in the voltage (which I assume is caused by the ESP32 booting up and using lots of current) that has a minimum level of 2.8V (which is the threshold of the voltage supervisor chip on my custom board).  Then the voltage rebounds back to ~3.0V and the cycle repeats.

[image]

[image]

LukasDashbike, 2022-05-12:
thanks for this solution. we had this problem with quite a few devices and i was able to rescue some of them (the ones that didnt have both SRP1 and SRP0 set) with
```
esp_tool write_flash_status
```
but it has to be noted that doing this with a single command can leave the 3. byte of the status register unchanged. I assume this is because the first two bytes get written first and after that the write protection is already in effect.
So if you want to change the 3.byte as well you have to do this in 2 turns:
for example if your status registers read 0x430200 and you want to change it to 0x600380:
```
esptool.py write_flash_status --bytes 3 --non-volatile 0x600200
esptool.py write_flash_status --bytes 3 --non-volatile 0x600380
```

phatpaul, 2022-05-13:
@winzkigermany I was playing with your code example and noticed that I got some crashes with this message:
`Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)`

So I realized that you disabled the cache with `g_flash_guard_default_ops.start(); //Disable cache` but then called some functions that were not placed in IRAM.  
Also, the main function xmc_flash_fix(void) should be in IRAM itself since it contains the guards and thus has code which runs between the guards.

In IDF 3.x there is a file called components/spi_flash/spi_flash_rom_patch.c that appears to override some ROM functions but doesn't include IRAM_ATTR.  It seems I can't call those while the cache is disabled, which seems useless.

So I moved the cache disable/enable guards around the inner-most function bootloader_execute_flash_command()
This prevented the crashes, but the end result is that the status bytes are not changed.

phatpaul, 2022-05-13:
Well, I made some progress.  I was inspired by @persan666 idea to execute the workaround before the scheduler starts.  But I didn't want to hack the IDF files if possible.

So I add this to main.c:
```c
void start_cpu0(void) __attribute__((used)); // override the weak definition from cpu_start.c
extern void start_cpu0_default(void) IRAM_ATTR __attribute__((noreturn));
void start_cpu0(void)
{
    xmc_flash_fix_early();
    start_cpu0_default();
}
```

And here is xmc_flash_fix.c

```c
/*
 * This file contains the XMC flash fix against status register corruption.
 * See:
 * https://github.com/espressif/esp-idf/issues/7994
 * https://en.hoerbert.com/technology/esp32-critical-fatal-problem-source-in-some-wrover-e-modules/
 *
 * Hacked to run before scheduler is started.
 * // add this to main.c:
void start_cpu0(void) __attribute__((used)); // override the weak definition from cpu_start.c
extern void start_cpu0_default(void) IRAM_ATTR __attribute__((noreturn));
void start_cpu0(void)
{
    xmc_flash_fix_early();
    start_cpu0_default();
}
 */

#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG
#include "esp_log.h" // required for esp_log_timestamp()
#include "esp_err.h"

#include "esp_idf_version.h"

#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0)
// enable functionality present in IDF v4.0
#include "esp32/rom/spi_flash.h"
#elif ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(3, 0, 0)
#include "rom/spi_flash.h"
#include "soc/spi_struct.h"
extern uint8_t g_rom_spiflash_dummy_len_plus[]; // also seems needed in IDF 3.3
#endif

#include "xmc_flash_fix.h"

static const char *TAG = "xmc_flash_fix";

/**
 *  Copy from execute_flash_command() since it is static function
 */
#define SPIFLASH SPI1
static IRAM_ATTR uint32_t bootloader_execute_flash_command(uint8_t command, uint32_t mosi_data, uint8_t mosi_len, uint8_t miso_len)
{

    uint32_t old_ctrl_reg = SPIFLASH.ctrl.val;
#if CONFIG_IDF_TARGET_ESP32
    SPIFLASH.ctrl.val = SPI_WP_REG_M; // keep WP high while idle, otherwise leave DIO mode
#else
    SPIFLASH.ctrl.val = SPI_MEM_WP_REG_M;       // keep WP high while idle, otherwise leave DIO mode
#endif
    SPIFLASH.user.usr_dummy = 0;
    SPIFLASH.user.usr_addr = 0;
    SPIFLASH.user.usr_command = 1;
    SPIFLASH.user2.usr_command_bitlen = 7;

[... 163 lines cut ...]
{
    // if we have an XMC flash chip susceptible to status register corruption,
    // flash the status registers properly and lock them afterwards.
    uint32_t flash_id = esp_xmc_read_flash_id();
    ESP_EARLY_LOGD(TAG, "Flash chip id is %06X", flash_id);
    if (flash_id == 0x204018)
    {
        esp_err_t retStatus = esp_xmc_16m_flash_sr_otp(&g_rom_flashchip);
        ESP_EARLY_LOGD(TAG, "esp_xmc_16m_flash_sr_otp() return code is: %u", retStatus);
    }
    else
    {
        ESP_EARLY_LOGD(TAG, "Flash chip is NOT 0x204018, skip workaround.");
    }
}
```

Now this runs without crashing, but it still doesn't seem to actually write the status bytes.

But I was able to change the status bits by hacking this call into the IDF function https://github.com/espressif/esp-idf/blob/cc5440f6a2df826dc9a9ded571ad20b5c30a4b89/components/bootloader_support/src/flash_qio_mode.c#L125
Which is called from https://github.com/espressif/esp-idf/blob/cc5440f6a2df826dc9a9ded571ad20b5c30a4b89/components/bootloader_support/src/bootloader_flash_config.c#L33 which is called from https://github.com/espressif/esp-idf/blob/cc5440f6a2df826dc9a9ded571ad20b5c30a4b89/components/esp32/cpu_start.c#L441

So I'm guessing some necessary initialization happens between the start of start_cpu0_default() and when it calls bootloader_flash_update_id() that allows writing to these status bytes??

Well, I'm done till Monday.  Any ideas welcome.

winzkigermany, 2022-05-14:
Hey @phatpaul , I'm sorry. I just took it just to the "works for me" (TM) level, but your input is of course very important. At the moment, I can't add much to this issue due to time constraints.

phatpaul, 2022-05-16:
I've tried all manner of cache disabling functions that I can find in the sdk.  They either cause a crash or don't seem to set the status bytes.  I think my initial success when hacking it into bootloader_read_flash_id(void) is because that function is also called by the 2nd-stage bootloader code.  And this status byte writing code worked from there.  The later invocation from cpu_start.c just confirmed that it was already written properly.  But I want an OTA firmware image that will fix this bug without touching the bootloader.
@igrr do you have any ideas why I can't manage to write the status bits within application code on IDF 3.3.6?

I can't justify spending more time hacking on this.  (I should have spent that time porting my project to IDF 4.x.)
In the mean-time I may just use the serial flasher tool esp_tool.py to set the status bytes in my production programming script.  That obviously won't help any units already deployed, but at least will fix newly produced ones.  Does my script need to check that it is the chip_id 0x20 before setting the status bytes?  I've seen a mix of 0x20 and 0xC8 chips on my production line.  What would be the harm of writing the same values regardless of the chip manufacturer?

fizzyade, 2022-05-16:
I'm staggered by the radio silence from Espressif on this fault.

vicatcu, 2022-05-17:
I use ESP32-WROOM-32D (not WROVER-E) modules in my products. The firmware is built using the Arduino-ESP32 board support package. I've gotten a handful of bricked units back from customers whose failure mode sounds quite similar to this. Is the scope of this issue limited to WROVER-E modules, or could this be affecting ESP32-WROOM-32D modules as well? Are the steps to confirm the same (`esptool.py read_flash_status --bytes 3`)? Is the firmware patch solution the same?

vicatcu, 2022-05-19:
Step 1. Is there a foolproof procedure that I can do, e.g. using esptool.py, to inoculate my existing modules from this problem?
Step 2. We really need a sanctioned solution from Espressif that we can put into an OTA image that hot-fixes it for fielded units.

Can anyone help with Step 1 at least?

LukasDashbike, 2022-05-19:
> Step 1. Is there a foolproof procedure that I can do, e.g. using esptool.py, to inoculate my un-fielded modules from this problem? Step 2. We really need a sanctioned solution from Espressif that we can put into an OTA image that hot-fixes it for fielded units.
> 
> Can anyone help with Step 1 at least?

yes. but only if the Status Register Protection Bits have not been corrupted yet. after applying this fix none of our "unfielded" devices had this issue again:
```
#!/bin/bash

# xmc flash fix for chip id 0x204018
if python3 esptool/esptool.py flash_id | grep -q "Manufacturer: 20" && python3 esptool/esptool.py flash_id | grep -q "Device: 4018"; then
    python3 esptool/esptool.py write_flash_status --bytes 3 --non-volatile 0x600200
    python3 esptool/esptool.py write_flash_status --bytes 3 --non-volatile 0x600380
fi
```
this can also fix devices that already have this problem if SRP0 and SRP1 bits have not been set in the status register (like described above).

vicatcu, 2022-05-19:
> yes. but only if the Status Register Protection Bits have not been corrupted yet. after applying this fix none of our "unfielded" devices had this issue again:
>
> ```
> #!/bin/bash
>
> # xmc flash fix for [...]

@LukasDashbike do you have any idea if it's safe to apply this fix on ESP32-WROOM-32D modules or if there is anything that makes it special to WROVER-E modules. I suspect it's a 'universal' fix from what I've read here, but would be good to hear an independent opinion(s).

LukasDashbike, 2022-05-20:
@vicatcu i havent tested it with other modules than the wrover-e. But if the flash chip id is the same i couldnt think of a reason this fix shouldnt work.

vicatcu, 2022-05-20:
Well, I can confirm that this board I have jigged up right now has an ESP32-WROOM-32D module on it that gives this result:
```
$ ./esptool.exe flash_id 
esptool.py v3.1
Found 1 serial ports
Serial port COM4    
Connecting....      
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:91:51:37:92:94
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...
```

and ... 

```
$ ./esptool.exe read_flash_status --bytes 3
esptool.py v3.1
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:91:51:37:92:94
Uploading stub...
Running stub...
Stub running...
Status value: 0xe37bfc
Hard resetting via RTS pin...
```

vicatcu, 2022-05-21:
@LukasDashbike Also notable that this one above was Device: 4017, not 4018. So it's possible that the aperture of that script needs to be  widened?

vicatcu, 2022-05-21:
@igrr (or anyone else) do you happen know if those status bits are applicable to all 0x20 Manufacturer chips, or some subset of the possible Device values (i.e. not just 4018)?

vicatcu, 2022-05-23:
Continuing to answer my own questions here... Looking at the XM25QH device family on the XMC website and combing over the datasheets suggests that the Device Identification changes with memory size, so I'm kind of assuming looking for a specific device ID (4018) is overly specific.

```
XM25QH32B = 32Mbit = 4MB = Device ID 0x6016
XM25QH32C = 32Mbit = 4MB = Device ID 0x4016
XM25QH64B = 64Mbit = 8MB = Device ID 0x6017
XM25QH64C = 64Mbit = 8MB = Device ID 0x4017
XM25QH128B = 128Mbit = 16MB = Device ID 0x6018
XM25QH128C = 128Mbit = 16MB =Device ID 0x4018
```
This footnote explains the 0x40 vs 0x60: 
> Data2 40h is for SPI mode and the 60h is for QPI mode.

Is there some reason to believe only certain memory sizes are susceptible?

mgigante, 2022-05-24:
Hi all,
at my company we are facing the same issue on chip modules with flash IDs

Manufacturer: 20
Device: 4017

and 

Manufacturer: 20
Device: 4018

We got bricked ESP32 on custom built boards based on ESP32-D0WD chip module, equipped with 16MiB flash.

phatpaul, 2022-05-24:
@vicatcu that makes sense.  My script (inspired by @LukasDashbike script above) is just checking for "Manufacturer: 20" - see attached [xmc-flash-fix.sh.txt](https://github.com/espressif/esp-idf/files/8763739/xmc-flash-fix.sh.txt).  If the status registers have the same layout for all sizes of the XM25QH chip then I don't see the harm in setting the values permanently on all sizes.  (Anyway, I only use the 16M flash in my product, so I'm not worried.) 

The bash script seems to work, but only prevents bricking newly built units.  I really need an OTA fix for deployed units.

Apparently, Espressif isn't very concerned about a bricked dev board for hobby use ( sorry @fizzyade ).  But several of us are using this for commercial products at significant quantities.  I guess we should present some numbers to Espressif so they will take this seriously.  I.e. units sold and the cost of field failures.
BTW, does anyone know a higher-up contact at Espressif that we can petition for help?

My numbers: 

- Part: ESP32-WROVER-E-N16R8
- Qty.: 5k per year @ $3.90 = $19,500 /y.  (Projected to increase 20% per year)
- Units already sold/deployed: 
~15k 
- Field failures: 7 confirmed with this failure mode.  Several more unconfirmed.  (it's a pain to cut open our sealed device to get to the serial port to confirm the failure mode, so I haven't confirmed every one)
- Cost of each field failure: device unit cost ~ $100 (our cost), but installation cost is much more.  And our reputation is damaged.

@mgigante, @vicatcu, @winzkigermany, @persan666  Are you using this professionally and can you share your numbers?

igrr, 2022-05-24:
Hi all, I'm sorry for the lack of updates on this issue from Espressif side. Thanks to everyone who posted here. This was really helpful for getting more attention to this issue! Espressif will publish an advisory on this topic.

On ESP-IDF side, there are some changes already present in IDF v4.4 (or later), v4.3.2 and v4.0.4 which greatly reduce the number of WRSR operations. More details (commit IDs) and backports to release/v4.2 and release/v4.1 will follow. These improvements are to the 2nd stage bootloader, so are applicable to new products. For the existing products, an example on how to lock the SR from the app will be provided.

In the meantime, for replacements, RMA or other business related issues, please contact sales at espressif.com.

vicatcu, 2022-05-24:
@phatpaul yes, I'm using modules commercially and my quantities are on par with yours.

[35 more replies]

closed 2022-07-11 by ginkgm