espressif/esp-idf #7996: ESP32-S3-WROOM-2 Octal Flash

TGiles1998, 2021-12-01:
## Environment

- Development Kit: ESP32-S3-DevKitC-1
- Kit version: ESP32-S3-DevKitC-1
- Module or chip used: ESP32-S3-WROOM-2
- IDF version: v4.4-dev-3703-gddc44956bf
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Tried to use one of the ESP-IDF example projects; Blink and flash it to the ESP32-S3-WROOM-2.

//Detailed problem description goes here.

### Expected Behavior

Expected to see the onboard LED flash as per the usual Blink Project outcome.

### Actual Behavior

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x1710
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2ed8
entry 0x403b61c0
I (36) boot: ESP-IDF v4.4-dev-3703-gddc44956bf 2nd stage bootloader
I (36) boot: compile time 23:43:05
I (36) boot: chip revision: 0
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : SLOW READ
I (50) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
W (60) bootloader_random: RNG for ESP32-S3 not currently supported
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 00100000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=089b0h ( 35248) map
I (121) esp_image: segment 1: paddr=000189d8 vaddr=3fc91b70 size=02824h ( 10276) load
I (124) esp_image: segment 2: paddr=0001b204 vaddr=40374000 size=04e14h ( 19988) load
I (135) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18a00h (100864) map
I (162) esp_image: segment 4: paddr=00038a28 vaddr=40378e14 size=08d50h ( 36176) load
I (172) esp_image: segment 5: paddr=00041780 vaddr=50000000 size=00010h (    16) load
I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
W (181) bootloader_random: RNG for ESP32-S3 not currently supported
E (199) Octal Flash: Detected MXIC Flash, but memory type is not Octal
E (200) Octal Flash: No detected Flash chip, please check the menuconfig to see if the chip is supported

abort() was called at PC 0x40381785 on core 0
0x40381785: esp_opiflash_init at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/spi_flash/esp32s3/spi_flash_oct_flash_init.c:233 (discriminator 3)

Backtrace:0x40375af2:0x3fceb2100x4037a2a1:0x3fceb230 0x4037fd06:0x3fceb250 0x40381785:0x3fceb2c0 0x40378134:0x3fceb2f0 0x40375370:0x3fceb310 0x403bb02c:0x3fceb340 0x403bb495:0x3fceb380 0x403b6229:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
0x40375af2: panic_abort at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/esp_system/panic.c:402

0x4037a2a1: esp_system_abort at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/esp_system/esp_system.c:121

0x4037fd06: abort at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/newlib/abort.c:46

0x40381785: esp_opiflash_init at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/spi_flash/esp32s3/spi_flash_oct_flash_init.c:233 (discriminator 3)

[... 130 lines cut ...]
I (124) esp_image: segment 2: paddr=0001b204 vaddr=40374000 size=04e14h ( 19988) load
I (135) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=18a00h (100864) map
I (162) esp_image: segment 4: paddr=00038a28 vaddr=40378e14 size=08d50h ( 36176) load
I (172) esp_image: segment 5: paddr=00041780 vaddr=50000000 size=00010h (    16) load
I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
W (181) bootloader_random: RNG for ESP32-S3 not currently supported
E (199) Octal Flash: Detected MXIC Flash, but memory type is not Octal
E (200) Octal Flash: No detected Flash chip, please check the menuconfig to see if the chip is supported

abort() was called at PC 0x40381785 on core 0
0x40381785: esp_opiflash_init at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/spi_flash/esp32s3/spi_flash_oct_flash_init.c:233 (discriminator 3)

Backtrace:0x40375af2:0x3fceb2100x4037a2a1:0x3fceb230 0x4037fd06:0x3fceb250 0x40381785:0x3fceb2c0 0x40378134:0x3fceb2f0 0x40375370:0x3fceb310 0x403bb02c:0x3fceb340 0x403bb495:0x3fceb380 0x403b6229:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
0x40375af2: panic_abort at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/esp_system/panic.c:402

0x4037a2a1: esp_system_abort at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/esp_system/esp_system.c:121

0x4037fd06: abort at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/newlib/abort.c:46

0x40381785: esp_opiflash_init at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/spi_flash/esp32s3/spi_flash_oct_flash_init.c:233 (discriminator 3)

0x40378134: spi_flash_init_chip_state at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/spi_flash/flash_ops.c:176

0x40375370: call_start_cpu0 at /Users/thomasgiles/esp-VS/esp-idf_011221/esp-idf/components/esp_system/port/cpu_start.c:386

mythbuster5, 2021-12-02:
@TGiles1998 HI, thanks for your report. I almost know the issue why. But I have a question that is this flash comes with the module when you bought or you once changed the flash by yourself?

TGiles1998, 2021-12-02:
Hi, It is the flash that came with the module I bought,
Thank you for your reply,

mythbuster5, 2021-12-09:
@TGiles1998  Sorry for any inconvenience caused. The patch will solve this problem. T和samples are provided in a small limited quantity at the moment. So, IDF support will release then it comes into mass production. If any more questions, please let me know.

[opi_flash_MX25UW12345G.txt](https://github.com/espressif/esp-idf/files/7681818/opi_flash_MX25UW12345G.txt)

mythbuster5, 2021-12-13:
@TGiles1998  If this patch solves your issue, this task will be closed soon. Thanks!

TGiles1998, 2021-12-13:
Hi, thank you very much for the patch, where should i place this in the idf repository? Thanks again!

mythbuster5, 2021-12-15:
@TGiles1998  1. You can change the name to ***.patch, and apply the patch with git directly.  2. You can also manually modify with the information in patch, in file `spi_flash_oct_flash_init.c`

philippe44, 2022-01-13:
The current 4.4 which seems to have that patch still gives me a reboot when I activate octal SPI flash (WROOM D3N8R8). Workd normally otherwise with SPIRAM set to octal

closed 2022-03-08 by espressif-bot