espressif/esp-idf #7997: TCP data Packet loss

omairakash, 2021-12-02:
Development Kit: ESP32-WROOM-32D (ESP32_DEVKITC_V4)
Kit version v4
Kit version N/A
Module or chip used: ESP32-WROOM-32D
IDF version ESP-IDF v4.4
Build System: CMake
Operating System: [Windows]
(Windows only) environment type: ESP Command Prompt
Using an IDE?: No
Power Supply: USB

## Problem Description
I am using ESP Dev kit for testing wifi. i have used wifi and tcp examples and modified it a little to make it work for my project. I am missing about 20 out of 100 packets. i have attached wireshark and packetsender log. I have also attached sdkconfig file used for this project. 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
I am only sharing the tcp part of the code as other part is sensitive. My device connects to a wifi and then calls this function to send tcp data. for debug or testing purpose i am sending it in a loop.

```cpp
static void tcp_client_task()
{
    char rx_buffer[128];
    char addr_str[128];
    int addr_family;
    int ip_protocol;
	char tcpPayload[1024]="HELLO, THIS IS A TEST MESSEGE";
	//printf("tcp payload\n%s",tcpPayload);
        struct sockaddr_in dest_addr;
        dest_addr.sin_addr.s_addr = inet_addr((const char*)serverip);
	    ESP_LOGI(TAG, "this is serverip  %u", dest_addr.sin_addr.s_addr);
        dest_addr.sin_family = AF_INET;
        dest_addr.sin_port = htons(serverPort);
        addr_family = AF_INET;
        ip_protocol = IPPROTO_IP;
        inet_ntoa_r(dest_addr.sin_addr, addr_str, sizeof(addr_str) - 1);
		
        int sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
            //break;
        }
        //ESP_LOGI(TAG, "Socket created, connecting to %s:%d", serverIP, serverPort);
		
        int err = connect(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
        if (err != 0) {
            ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
            //break;
        }
        ESP_LOGI(TAG, "Successfully connected");
            int XY = send(sock, tcpPayload, strlen(tcpPayload), 0);
            if (XY < 0) {
                ESP_LOGE(TAG, "Error occurred during sending: errno %d", errno);
            }

        if (sock != -1) {
            ESP_LOGE(TAG, "Shutting down socket and restarting...");
            shutdown(sock, 0);
            close(sock);
        }
[missed packets packetsender.log](https://github.com/espressif/esp-idf/files/7638132/missed.packets.packetsender.log)
![sinppet2wireshark](https://user-images.githubusercontent.com/81243896/144340128-6af09a77-2f98-48ed-8810-51e6d7cbb9aa.png)
![snippet1wireshark](https://user-images.githubusercontent.com/81243896/144340135-7073dd4c-ba87-4b3e-aea2-d81d71449711.png)
![snippet3wireshark](https://user-images.githubusercontent.com/81243896/144340137-5cd08862-fad2-46ee-b5a1-7a348906aeb8.png)
![snippet4wireshark](https://user-images.githubusercontent.com/81243896/144340140-f5397fcc-769e-4e27-8ef2-2e3b002bc87f.png)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7638170/sdkconfig.txt)
}

omairakash, 2021-12-03:
Wifi is turned off when tcp packet is sent and turned on back again after a 1 second delay and again a packet is sent.

negativekelvin, 2021-12-03:
> Wifi is turned off when tcp packet is sent 

Maybe the packet does not always finish transmission before you turn off WiFi?

omairakash, 2021-12-03:
yeah, actually just finished further testing and found out that i was shutting down wifi straight after sending the packet. i think even though it is a blocking tcp transmission if you send the tcp packet and straight away shutdown wifi, sometimes it doesnt send packet. i am assuming data packet might still be getting processed on hardware level. Even if it fails to send the data packet, is not a issue because it can be handled but esp-32 doesnt give you a transmission error.

xueyunfei998, 2021-12-06:
hi @omairakash 

The mechanism of TCP sending data is as follows：

1 TCP has a corresponding sending window. When the application layer transmits data to LwIP, it will first fill in the sending window. Every time TCP successfully sends a data, it will receive an ACK. If an ACK is not received, the corresponding sending window will not move

2  When WiFi is disconnected, the data that has not been sent will be placed in the send window.

The solution to your problem is as follows：

1 When the WiFi is disconnected, a disconnect event will be received. In the event, close the socket.

omairakash, 2021-12-07:
Adding a delay of 50 ms for shutting down  WiFi after closing the TCP socket, fixed my problem. So, i guess if you are using TCP sockets and want to turn off WiFi immediately after sending, you have to wait for suitable time. That way data packets are not lost because wifi had shutdown before sending. & as suggested by @xueyunfei998,  it would be good practice. to close the socket after WiFi disconnect event.

closed 2022-03-08 by Alvin1Zhang