espressif/esp-idf #7998: ESP32-S3 USB and Bluetooth cannot work together

hhuysqt, 2021-12-02:
## Environment

- Development Kit:      ESP32-S3-DevKitC-1
- Kit version: v1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: v5.0-dev-553-g42cce06704
- Build System:         idf.py
- Compiler version: 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

ESP32-S3 USB goes off when Bluetooth is turned on. I found this topic **a week ago** in the forum: [https://www.esp32.com/viewtopic.php?f=25&t=24561](https://www.esp32.com/viewtopic.php?f=25&t=24561), which says that this issue will be solved in a new release **in just a few days**. However the work-around is still missed in the latest esp-idf. 

**Looking for an update ASAP!**

Dazza0, 2021-12-02:
@hhuysqt Does https://github.com/espressif/esp-idf/commit/65f3bf4a8a21f6ef51649c3202ca282e9b04d124 resolve the issue?

hhuysqt, 2021-12-02:
Oh yes, this resolves the problem. I had forgotten to update the git submodules... Thanks for your reply!

closed 2021-12-02 by hhuysqt

acebearxp, 2022-02-22:
> Oh yes, this resolves the problem. I had forgotten to update the git submodules... Thanks for your reply!

Confirmed! I had the same issue in v4.4 and updating the submodule fix this.
```shell
# switch to esp_phy\lib submodule
cd %IDF_PATH%
cd components\esp_phy\lib
# manually update the submodule to master
git checkout master
# then re-build & re-flash
...
```

NicoBocki, 2022-12-29:
The problem still exist in release 2.06