espressif/esp-idf #7999: quicklz decompress or bsdiff_patch will dead, when task_wdt got triggered.

Zmmfly, 2021-12-02:
## Environment

- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): v4.3.1
- Build System:  
    // idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt.
- Power Supply: 3.3V

## Problem Description
First, i upgrade firmware  by myself protocol,  it's has two process used much time.
If any process used more than 5 seconds, it's will triggered task_wdk, and will always not complete.
The all process run in CPU0, and CPU0 had many task.

So, is the task_wdt's problem or other?

0xjakob, 2021-12-03:
@Zmmfly It is a configuration problem. You can first try to increase the watchdog timeout by changing the option [CONFIG_ESP_TASK_WDT_TIMEOUT_S](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-task-wdt-timeout-s). Once you found an appropriate timeout, it's probably best to change the timeout at runtime using [`esp_task_wdt_init`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html?highlight=task%20wdt#_CPPv417esp_task_wdt_init8uint32_tb) and then change it back afterwards to have the proper watchdog protection before and after the long-running task.

I have this information from the [Task WDT documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html?highlight=task%20wdt#task-watchdog-timer).

Zmmfly, 2021-12-10:
@0xjakob Thanks for your response, i closed wdt.

closed 2021-12-10 by Zmmfly