espressif/esp-idf #8000: Distribute code in OTA

hdpklm, 2021-12-02:
Hello All

I have a lots of librarys and I want to distribute in different ota partition, so wen I want to update any library only update one partition, not the full code, and not need to repeat all library in all partition and consume a lot of internet for update the code.

example of partitions:
factory: boot code and initialize the sensors
OTA_0: has the library of bluetooth
OTA_1: has library of sensors
OTA_2: has some equation and diferent functions.
OTA_3: reserve for download and update specific OTA number

So I need a way to call function in other partition from the main OTA or Factory partition.

is this available and how can I do that

thank you a lot

mahavirj, 2021-12-10:
@hdpklm 

> So I need a way to call function in other partition from the main OTA or Factory partition.

There is no provision for this as now. IDF bootloader only deals with selected active OTA partition, copying its code/data section to IRAM/DRAM and setting up flash cache mappings etc.

> consume a lot of internet for update the code

For this you may consider "delta" OTA update, we had done preliminary analysis and you may find discussion at https://github.com/espressif/esp-idf/issues/7543 useful in this context.

Hope this helps.

hdpklm, 2021-12-11:
Hello @mahavirj 

Thanks for answering
Can I download a binary code to the storage partition and run it from the main function, or copy to memory and call it address

mahavirj, 2022-01-17:
@hdpklm

No, this is not supported feature. I recommend that you go through https://esp32.com forum thread, there are some relevant topics on supporting such feature (mostly exploratory work around ELF loading). I will go ahead and close this issue.

closed 2022-01-17 by mahavirj