espressif/esp-idf #8002: Bootloader is too large for 48KB staging area.

0xFEEDC0DE64, 2021-12-02:
We upgraded the esp-idf used in our firmware project, before updating the esp-idf we used an about a year old fork with some patches and as most of them are already upstreamed and in master, we tried flashing our firmware (and bootloader and partition_table) with the newest idf.

I'm directly greeted with a "Bootloader is too large for 48KB staging area." message, even before the bootloader can output any log messages.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
Bootloader is too large for 48KB staging area.
ets_secure_boot_v2.c 74 
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
Bootloader is too large for 48KB staging area.
ets_secure_boot_v2.c 74 
ets Jul 29 2019 12:21:46
```

Is there anything I can check now?

AxelLin, 2021-12-02:
Which esp-idf versions are you using (the old and new version)?

kayzen9, 2021-12-02:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/bootloader.html#bootloader-size

negativekelvin, 2021-12-02:
How big is the bootloader bin? What is your bootloader log level and compiler optimization level?

0xFEEDC0DE64, 2021-12-06:
Thanks for all your responses,
we switched from a 4.4-dev based idf to the newest master,

build/bootloader/bootloader.bin grew in size to 56KB

we have secure boot and flash encryption enabled, and logging is set to info

0xjakob, 2021-12-07:
@0xFEEDC0DE64 Have you tried to reduce the log level already, as described in the document @kayzen9 has suggested?

Alvin1Zhang, 2021-12-20:
@0xFEEDC0DE64 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

0xFEEDC0DE64, 2021-12-21:
Hi, I just tried with newest master again, still the bootloader was too big, but I was able to "shrink" the bootloader in size enough by  disabling the XMC flash support and by disabling the factory partition with gpio.

closed 2021-12-21 by 0xFEEDC0DE64