espressif/esp-idf #8003: Expose option to specify stop bits for ModBus Serial.

atanisoft, 2021-12-02:
**Is your feature request related to a problem? Please describe.**

Currently it is forced to one stop bit, some devices use two stop bits:
https://github.com/espressif/esp-idf/blob/ddc44956bf718540d5451e17e1becf6c7dffe5b8/components/freemodbus/port/portserial_m.c#L219-L227

**Describe the solution you'd like**

Similar to baud and parity it would be good to expose the stop bits setting.

**Describe alternatives you've considered**

As a workaround I'm adding code to reset the stop bits *AFTER* calling `mbc_master_start()` but I'd like to avoid that if possible.

**Additional context**

mattiaVDM, 2024-01-16:
> **Is your feature request related to a problem? Please describe.**
>
> Currently it is forced to one stop bit, some devices use two stop bits: [...]

Hello, how do you set the stop bits after initialization? i need to configure them at runtime

atanisoft, 2024-01-16:
> Hello, how do you set the stop bits after initialization?

Use `uart_set_stop_bits`

alisitsyn, 2024-01-17:
> Hello, how do you set the stop bits after initialization? i need to configure them at runtime

Here is duplicate issue against esp-modbus:
https://github.com/espressif/esp-modbus/issues/25

See also the options in v2.0.0_beta release: 
https://github.com/espressif/esp-modbus/discussions/45
https://github.com/espressif/esp-modbus/blob/release/v2.0/examples/serial/mb_serial_master/main/serial_master.c#L274

closed 2024-12-17 by Alvin1Zhang