espressif/esp-idf #8004: Cannot transmit data as SPI slave

j-haller, 2021-12-02:
## Environment

- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: release v4.4
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description

I am currently using an ESP32-S3 and trying to send and receive data as SPI slave, but as soon as I populate my TX buffer with any data, random bytes are transmitted and received. If the TX buffer is empty (only zeros), then receiving data works as expected. The SPI master is a ESP32-Devkit, but the same behavior can be seen if a Raspberry Pi is used as master. All lines (MOSI, MISO, CLK and CS) are about 13 cm (5.1 inch) long.
I have tried different clock speeds (100 kHz, 1 MHz and 10 MHz), modes (from 0 to 3) and pre-transmission delays (cs_ena_pretrans set to 0, 1, 2, 5 and 10) with no improvement. 

### Expected Behavior

Receiving and transmitting data as SPI slave.

### Actual Behavior

Random data is received and transmitted.

### Steps to reproduce

1. Connect a ESP32 (SPI master) to a ESP32-S3 (SPI slave) and transmit data over SPI. No handshake line is used, only MOSI, MISO, CLK and CS.
2. Run code examples below.

Note:
- If TX buffer on SPI slave is filled with zeros, receiving data works as expected.
- If TX buffer on SPI slave is filled with data, random bytes are transmitted and received.

### Code to reproduce this issue

SPI master: https://gist.github.com/j-haller/1a857e5251325c894d7f5b9e63d692f5
SPI slave: https://gist.github.com/j-haller/f80e72eee7e722118d1c74ff294bc008

Icarus113, 2021-12-06:
Hi @j-haller ,
It's possibly due to lack of handshake signal. SPI Master driver can start a transaction (by making CS signal active, generating CLK signal) at any time, even if SPI Slave is not ready (Slave hasn't load its data to its hardware fifo). Under this condition, SPI Master can still generate correct TX data, and receive random data from MISO line.

j-haller, 2021-12-07:
Hello @Icarus113 ,
thank you for your reply. I tried to use the spi_slave/sender and spi_slave/receiver examples, which are using an additional handshake signal, but without success. Something similar is happening: spi master and slave cannot receive any data from each other. But if I modify the receiver example, so that the TX buffer is empty, spi slave can receive the message from spi master.

j-haller, 2021-12-09:
I made a few tests with an oscilloscope. The blue signal is MOSI and red is MISO with a clock frequency of 100 kHz. Additionally, a handshake signal is used. Unfortunately my scope only has two channels, so I couldn't show clock and chip select at the same time, but both signals looked ok.

Here, spi master send "Hello World" and spi slave send twelve times 0x00. As expected spi master received zeros and spi slave received "Hello World".
[image]

Here, spi master still send out "Hello World" but spi slave should send twelve 0x55.
Spi master received: {0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x55, 0x50, 0x00, 0x00, 0x00}
Spi slave received: {0x21 ('!'), 0x65 ('e'), 0x1B (ESC), 0x20 (SP), 0x6F ('°'), 0xF7, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}
The number of "0x55" that are actually send out by spi slave varies for each transactions.
[image]

Again, spi master send "Hello World" and spi slave send "dlroW olleH".
Spi master received: {0x5B ('['), 0x2D ('-'), 0x6D ('m'), 0xAD, 0xC9, 0x6D ('m'), 0xD5, 0x00, 0x00, 0x00, 0x00, 0x00}
Spi slave received: {0x1B (ESC), 0x57 ('W'), 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}
[image]

I repeated the test, but this time spi slave send out "Hello World" and the message from spi master varies. The blue signal is still MOSI and red is MISO. The clock frequency is 100 kHz.

Spi master send out twelve 0x00 and spi slave "Hello World". Spi master received "Hello World" and spi slave received twelve zeros.
[image]

Here, spi master send twelve 0x55 and spi slave should be sending "Hello World" since I didn't change anything on that side. Instead spi slave send varying amounts of 0x55 and sometimes a 0x40 at the end.
Spi master received: {0x55, 0x55, 0x55, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}
Spi slave received: {0x55, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}
[image]

Here, spi master send "dlroW olleH" and spi slave "Hello World".
Spi master received: {0x48 ('H'), 0x6D ('m'), 0x96, 0xD8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}
Spi slave received: {0x64 ('d'), 0x37 ('7'), 0xF5, 0x7B, 0x63 ('c'), 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}
[image]

closed 2022-12-13 by j-haller

Gartosz, 2025-04-01:
Hi, have you managed to resolve your issue. I'm having the same problem on ESP32-S3 and I've seen other people reporting it. It works as a master without any issue with another ESP32, but not as a slave unfortunately :/ Is there something I can do to make it work? @hallkyon