espressif/esp-idf #8005: Use websocket to crash during regular execution of http tasks

jiaxieya, 2021-12-03:
Operating System: Windows
Using an IDE?: VS Code
Power Supply: USB
I (48) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (49) boot: compile time 16:03:22
I (49) boot: chip revision: 3
I (51) boot.esp32c3: SPI Speed : 80MHz
I (56) boot.esp32c3: SPI Mode : DIO
I (60) boot.esp32c3: SPI Flash Size : 4MB

## Problem Description
The regular execution of the http task is used to obtain the necessary parameters to change the link used by the websocket, but the execution of the websocket connection after the second run of the http task causes a crash. Part of the parameter part is in privacy. I have deleted it in the code. The websocket link can be replaced with any link for testing.

### Code to reproduce this issue

static void http_test_task(void *pvParameters)
{
    char local_response_buffer[MAX_HTTP_OUTPUT_BUFFER] = {0};
    /**
     * NOTE: All the configuration parameters for http_client must be spefied either in URL or as host and path parameters.
     * If host and path parameters are not set, query parameter will be ignored. In such cases,
     * query parameter should be specified in URL.
     *
     * If URL as well as host and path parameters are specified, values of host and path will be considered.
     */
    esp_http_client_config_t config = {
        .host = "api.m.taobao.com",
        .path = "/rest/api3.do?api=mtop.common.getTimestamp",
        .query = "esp",
        .event_handler = _http_event_handler,
        .user_data = local_response_buffer, // Pass address of local buffer to get response
        .disable_auto_redirect = true,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    cJSON *root = NULL;
    cJSON *item = NULL;
    cJSON *number = NULL;

   
    char timestamp[] = "";
    char *out;

    while (1)
    {

        // GET
        client = esp_http_client_init(&config);

        esp_err_t err = esp_http_client_perform(client);
        if (err == ESP_OK)
        {
            // ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d",
            //          esp_http_client_get_status_code(client),
            //          esp_http_client_get_content_length(client));
        }
        else
        {
            ESP_LOGE(TAG, "HTTP GET request failed: %s", esp_err_to_name(err));
        }
        // ESP_LOGI(TAG, "local_response_buffer: %s", local_response_buffer);
        /**************************************************************************************/
        root = cJSON_Parse(local_response_buffer);
        if (!root)
        {
            printf("Error before: [%s]\n", cJSON_GetErrorPtr());
        }
        else
        {
            item = cJSON_GetObjectItem(root, "data");
[... 180 lines cut ...]
3fca9df0: 0x2c223463 0x6e6f6322 0x746e6574 0x61223a22 0x63305248 0x4c364d48 0x61683979 0x4c705658
3fca9e10: 0x5a77396d 0x637a3557 0x596c5647 0x59756732 0x64763432 0x4c7a5248 0x64765232 0x62733532
3fca9e30: 0x4c6b4632 0x51596c31 0x5668646e 0x54354531 0x4d5a5230 0x54453130 0x564e706d 0x597a4645
3fca9e50: 0x61734a7a 0x52584231 0x565a4e6a 0x57334645 0x65524a54 0x57363145 0x52526c6d 0x54304931
3fca9e70: 0x64425255 0x52553130 0x524e4654 0x57704a46 0x4d56646b 0x52556c56 0x52504654 0x54334645
3fca9e90: 0x634b6845 0x4d457046 0x62756748 0x227a4158 0x6572222c 0x746c7573 0x2264695f 0x222c313a
3fca9eb0: 0x6c5f7369 0x22747361 0x7572743a 0x69222c65 0x69665f73 0x6873696e 0x72743a22 0x222c6575
3fca9ed0: 0x6e6f736a 0x6772615f 0x7b3a2273 0x6e616322 0x226c6563 0x2230223a 0x7464222c 0x223a2265
3fca9ef0: 0x22776172 0x7464222c 0x323a2273 0x7266222c 0x5f656d61 0x3a226469 0x75222c31 0x3a226c72
3fca9f10: 0x7d223122 0x73222c7d 0x3a226469 0x61776122 0x64633030 0x62363331 0x30786440 0x31313030
3fca9f30: 0x62343035 0x61356564 0x30383531 0x222c2230 0x65646f63 0x30223a22 0x64222c22 0x22637365
3fca9f50: 0x7573223a 0x73656363 0x767d2273 0x222c2265 0x22646973 0x7761223a 0x63303061 0x36333164
3fca9f70: 0x78644062 0x31303030 0x34303531 0x35656462 0x38353161 0x2c223030 0x78657422 0x223a2274
3fca9f90: 0x6c6c6568 0x222c226f 0x64697575 0x61223a22 0x30306177 0x33316463 0x64406236 0x30303078
3fca9fb0: 0x30353131 0x65646234 0x35316135 0x22303038 0x6f76222c 0x5f656369 0x77736e61 0x3a227265
3fca9fd0: 0x63227b5b 0x65746e6f 0x3a22746e 0xa0bde422 0xefbda5e5 0xa7e88cbc 0xb088e581 0xe6a0bde4
3fca9ff0: 0xbee59188 0xa38de888 0xefb8b9e5 0x2c2281bc 0x70797422 0x223a2265 0x22535454 0x2c7d5d7d
3fcaa010: 0x73657222 0x5f746c75 0x3a226469 0x69222c31 0x616c5f73 0x3a227473 0x65757274 0x7369222c
3fcaa030: 0x6e69665f 0x22687369 0x6c61663a 0x2c7d6573 0x64697322 0x61223a22 0x30306177 0x33316463

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

ESP-YJM, 2021-12-03:
@jiaxieya It seems that the pointer used to access the data was either not initialized or was corrupted. Could you provide a concise code to reproduce the problem so that I can reproduce it locally. I can’t combine the code you pasted to reproduce the problem.

Alvin1Zhang, 2021-12-13:
@jiaxieya Thanks for reporting, would you please help share if any updates for the issue? Thanks.

jiaxieya, 2021-12-27:
> @jiaxieya Thanks for reporting, would you please help share if any updates for the issue? Thanks.
I have not figured out the specific problem. The method I currently take is to adjust the priority of freertos tasks so that they are the same, and I set up two websocket clients to switch between them in turn (to ensure that there is sufficient time for each release) , This problem seems to be solved

closed 2021-12-27 by jiaxieya