espressif/esp-idf #8007: Sleep mode doesn't work

boris92git, 2021-12-03:
## Environment

- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it):
v4.4-dev-3565-g697f829d60
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
build it via docker(espressif/idf@sha256:bdcb958b9fc25a4109f2c18b9569adaae0d81abac03dd4dcf02983fd857fabc8)
- Operating System: Linux
- Using an IDE?: No
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

ESP32 doesn't wake up from sleep mode if RTC clock source is set like an Internal 8.5MHz oscillator, divided by 256

### Expected Behavior

I expect the ESP32 to wake up 5 seconds after running this test code.

### Actual Behavior

ESP32 doesn't wake up(((

### Steps to reproduce

Set RTC clock source:
menuconfig->Component config->ESP32-specific->RTC clock source: Internal 8.5MHz oscillator, divided by 256
and flash the code from the section below.

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "sdkconfig.h"
#include "esp_log.h"
#include <esp_sleep.h>

extern "C" {
void app_main();
}

void app_main(void)
{
    esp_deep_sleep(5000000);
}
```

I think the problem occurs due to this commit: 
https://github.com/espressif/esp-idf/commit/73384adfbe3a4365fb8fefdae9b1f9df4e509635
If I remove lines or set RTC source like internal 160kHz RC oscillator.
The test code is correctly working.

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

AxelLin, 2021-12-27:
Hit the same issue with v4.3.2-102-gda6c5be6c1, esp32 does not wakeup from deepsleep if ESP32_RTC_CLK_SRC_INT_8MD256 is used.

AxelLin, 2022-01-14:
@Alvin1Zhang 
Anyone from espressif can take a look at the issue? (The mentioned PR fixed the issue).

MLStoltzenburg, 2022-01-28:
Hi,

The same is happening with me.
If I enable CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256, the deep sleep mode doesn't work.

I am using release/v4.3 -- commit: 5fc51252e94170f54c0afeef625cd417451fdf13

bugadani, 2022-02-17:
To clarify, deep sleep works, wakeup doesn't :) I seem to be experiencing this issue, with EXT0 as my wakeup source. Using `CONFIG_ESP32_RTC_CLK_SRC_INT_RC` works.

AxelLin, 2022-03-02:
The same issue also happens on ESP32C3 if CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y, tested on v4.3.2-355-g047adc5ebc.

AxelLin, 2022-03-03:
ESP32S2 also has the same issue if CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y.

AxelLin, 2022-03-24:
Hi support team,
This is a regression in stable branches (it was working fine in older esp-idf version), and
it has been reported for more than 3 months.
Any one can help to fix it?

esp-lis, 2022-03-25:
> Hi support team, This is a regression in stable branches (it was working fine in older esp-idf version), and it has been reported for more than 3 months. Any one can help to fix it?

You can use the following code to power on the Internal 8MHz OSC power domain before deep sleep.
`ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC8M, ESP_PD_OPTION_ON));`

ginkgm, 2022-03-27:
Hi @boris92git , @AxelLin , @MLStoltzenburg ,

Thanks for reporting this issue.

I see all the commits you reporting that have such issue, have a common commit: e44ead535640525969c7e85892f38ca349d5ddf4. 

I guess it's not related to this commit: 73384adfbe3a4365fb8fefdae9b1f9df4e509635

Could you help to comment these lines and have a look? cc68fbf1679ce4376c2b26cf73c7b69498a0564e5973c2f4fdf125ba69b064ceR192

In this commit, a new control bit is introduced - int_8m_pd_en, which is true by default. This makes the code run into the second path (disable 8M, which doesn't exist before), which will bypass the protection in above lines. In previous code, if using lightsleep, the code will run into the first path (power up 8M). 

We are working on this. But if I can get your confirmation, it will be great.

AxelLin, 2022-03-28:
> Hi @boris92git , @AxelLin , @MLStoltzenburg ,
> 
> Thanks for reporting this issue.
> 
> I see all the commits you reporting that have such issue, have a common commit: [e44ead5](https://github.com/espressif/esp-idf/commit/e44ead535640525969c7e85892f38ca349d5ddf4).
> 
> I guess it's not related to this commit: [73384ad](https://github.com/espressif/esp-idf/commit/73384adfbe3a4365fb8fefdae9b1f9df4e509635)
> 
> Could you help to comment these lines and have a look? [cc68fbf1679ce4376c2b26cf73c7b69498a0564e5973c2f4fdf125ba69b064ceR192](https://github.com/espressif/esp-idf/commit/e44ead535640525969c7e85892f38ca349d5ddf4#diff-cc68fbf1679ce4376c2b26cf73c7b69498a0564e5973c2f4fdf125ba69b064ceR192-R194)
> 
@ginkgm 
If you mean below changes, it does **not** work. (tested on esp32-wroom-32e).
Tested by calling esp_deep_sleep(5000000); in app_main, I think you can also verify it.
```
diff --git a/components/esp_hw_support/port/esp32/rtc_sleep.c b/components/esp_hw_support/port/esp32/rtc_sleep.c
index dff1b49acd..8d7f24c593 100644
--- a/components/esp_hw_support/port/esp32/rtc_sleep.c
+++ b/components/esp_hw_support/port/esp32/rtc_sleep.c
@@ -186,11 +186,11 @@ void rtc_sleep_init(rtc_sleep_config_t cfg)
         REG_CLR_BIT(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_CK8M_FORCE_PU);
     }
     //Keep the RTC8M_CLK on in light_sleep mode if the ledc low-speed channel is clocked by RTC8M_CLK.
-    if (!cfg.int_8m_pd_en && GET_PERI_REG_MASK(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_DIG_CLK8M_EN_M)) {
+    if (!cfg.deep_slp && GET_PERI_REG_MASK(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_DIG_CLK8M_EN_M)) {
         REG_CLR_BIT(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_CK8M_FORCE_PD);
         REG_SET_BIT(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_CK8M_FORCE_PU);
-    } else {
-        REG_CLR_BIT(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_CK8M_FORCE_PU);
+    //} else {
+    //    REG_CLR_BIT(RTC_CNTL_CLK_CONF_REG, RTC_CNTL_CK8M_FORCE_PU);
     }
 
     /* enable VDDSDIO control by state machine */

```

ginkgm, 2022-03-28:
@AxelLin 
wow..... thanks a lot.. Will look into this issue again..

AxelLin, 2022-04-25:
@ginkgm 
Do you have any update?

igrr, 2022-05-09:
@AxelLin the fix for this issue is finished and is under review now, it's a bit different than the linked PR. We'll try to merge it soon.

ginkgm, 2022-05-13:
There are two fixes together fix this issue. The first one has been merged to master: cc37f63e14af5a04ce016244cef496c860987a0c (waiting to be synced).

The backport and the second part is on the way. Thanks for reporting and patience.

closed 2022-05-15 by espressif-bot

ginkgm, 2022-06-04:
The other fix is here: 1f6fad6f486140d79e6835df45f96104b185a04b

Backport to 4.4 and earlier version is on the way

AxelLin, 2022-07-13:
@ginkgm 
v4.3 branch still does not include this fix.

AxelLin, 2022-08-14:
> The other part of fix is here: [1f6fad6](https://github.com/espressif/esp-idf/commit/1f6fad6f486140d79e6835df45f96104b185a04b)
> 
> Backport to 4.4 and earlier version is on the way

@ginkgm 
I wait yet another month for the fix but the v4.3 branch still does not include this fix.

AxelLin, 2022-09-29:
@ginkgm @igrr 
There is another wrong wake up time issue with CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y, https://github.com/espressif/esp-idf/issues/9455 .
So CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y still does not work well with deepsleep.
Sigh, I didn't expect such long time to fix the wakeup from sleep mode issue.

AxelLin, 2022-10-14:
> @AxelLin the fix for this issue is finished and is under review now, it's a bit different than the linked PR. We'll try to merge it soon.

This still does not work well because of wrong wake up time, https://github.com/espressif/esp-idf/issues/9455 .
I'm not sure if anyone is working on this because there is no response for months.

KonssnoK, 2023-05-10:
Hi guys, this patch does not work for ESP32S3
see
https://github.com/espressif/esp-idf/issues/11362