espressif/esp-idf #8008: WiFi softAP client connect issues on ESP32s3

jasenharris, 2021-12-03:
## Environment

- Development Kit:      ESP32-S3-DevKitC-1-N8
- Kit version (for WroverKit/PicoKit/DevKitC): v1?
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-3675-g35b20cadce
- Build System: eclipse frontend for idf
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): TBD - default that installed with latest IDF tools for eclipse
- Operating System:  macOS
- Using an IDE?: Yes Eclipse
- Power Supply: USB

## Problem Description

Using the softAP example program. WiFI AP advertises and client attempts to connect, but repeatedly joins and then fails. Tried from multiple clients. 

### Expected Behavior

Wifi client to join softAP successfully.

### Actual Behavior

Wifi client joins and then immediately leaves, attempts multiple times and then ultimately leaves and is unable to connect to the softAP.

### Steps to reproduce

1. build and flash onto esp32s3 dev kit the softAP example (examples/wifi/getting_started/softAP/)
2. attempt to connect wifi client.

### Code to reproduce this issue

default code.

## Debug Logs

I (28392) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<255,255>, prof:11
I (28392) wifi:station: ee:40:a8:91:09:a6 join, AID=1, bgn, 20
I (32402) wifi:station: ee:40:a8:91:09:a6 leave, AID = 1, bss_flags is 658530, bss:0x3fcf3e34
I (32402) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<255,255>, prof:11
I (41842) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<255,255>, prof:11
I (41842) wifi:station: ee:40:a8:91:09:a6 join, AID=1, bgn, 20
I (45842) wifi:station: ee:40:a8:91:09:a6 leave, AID = 1, bss_flags is 658530, bss:0x3fcf41b0
I (45842) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<255,255>, prof:11
I (89542) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<255,255>, prof:11
I (89542) wifi:station: ee:40:a8:91:09:a6 join, AID=1, bgn, 20
I (93542) wifi:station: ee:40:a8:91:09:a6 leave, AID = 1, bss_flags is 658530, bss:0x3fcf41b0
I (93542) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<255,255>, prof:11

## Other items if possible

If you disable the use NVS flash option in the sdkconfig then the wifi will all clients to connect. This then leads to another issue that when the TinyUSB is enabled, as soon as the wifi initializes, the TinyUSB driver stops working. I was able to trace this issue back to wifi not really working correctly in the example project softAP and I suspect there is an issue here that needs to be resolved.

xueyunfei998, 2021-12-16:
hi @jasenharris 

Please upgrade to the latest master for testing. This problem has not reappeared in my local test and should have been solved.

kayleejacob1, 2022-02-08:
Iam using ESP32-s3- devkit C esp-idf version 4.4,  I have same issue if use nvs flash wifi is checked it wont connect.  This worked fine yesterday but today it surfaced.  

this is code example 
`  ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_ap();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &wifi_event_handler,
                                                        NULL,
                                                        NULL));
    wifi_config_t wifi_config={};

   
    strcpy((char *) wifi_config.ap.ssid, "test");
    strcpy((char *) wifi_config.ap.password, ESP_WIFI_PASS);
    wifi_config.ap.max_connection = MAX_STA_CONN;
	wifi_config.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_softap finished. SSID:%s password:%s channel:%d", apssid, ESP_WIFI_PASS, ESP_WIFI_CHANNEL); `

log

I (431) wifi:wifi driver task: 3fce5a34, prio:23, stack:6656, core=0
I (431) system_api: Base MAC address is not set
I (431) system_api: read default base MAC address from EFUSE
I (451) wifi:wifi firmware version: 7679c42
I (451) wifi:wifi certification version: v7.0
I (451) wifi:config NVS flash: enabled
I (451) wifi:config nano formating: disabled
I (451) wifi:Init data frame dynamic rx buffer num: 32
I (461) wifi:Init management frame dynamic rx buffer num: 32
I (461) wifi:Init management short buffer num: 32
I (471) wifi:Init dynamic tx buffer num: 32
I (471) wifi:Init static tx FG buffer num: 2
I (481) wifi:Init static rx buffer size: 1600
I (481) wifi:Init static rx buffer num: 10
I (481) wifi:Init dynamic rx buffer num: 32
I (491) wifi_init: rx ba win: 6
I (491) wifi_init: tcpip mbox: 32
I (501) wifi_init: udp mbox: 6
I (501) wifi_init: tcp mbox: 6
I (501) wifi_init: tcp tx win: 5744
I (511) wifi_init: tcp rx win: 5744
I (511) wifi_init: tcp mss: 1440
I (521) wifi_init: WiFi IRAM OP enabled
I (521) wifi_init: WiFi RX IRAM OP enabled
I (531) phy_init: phy_version 302,3ddfab3,Dec  3 2021,14:39:12
I (581) wifi:mode : softAP (7c:df:a1:e0:76:d1)
I (581) wifi:Total power save buffer number: 16
I (581) wifi:Init max length of beacon: 752/752

error This just keep repeating below

I (16425) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (16425) wifi:station: b8:76:3f:3f:99:ec join, AID=1, bgn, 20
W (16435) wifi:<ba-add>idx:2 (ifx:1, b8:76:3f:3f:99:ec), tid:0, ssn:0, winSize:64
I (20435) wifi:station: b8:76:3f:3f:99:ec leave, AID = 1, bss_flags is 658530, bss:0x3fcf53a0
I (20435) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
W (20435) wifi:<ba-del>idx
I (21035) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
I (21035) wifi:station: b8:76:3f:3f:99:ec join, AID=1, bgn, 20
W (21045) wifi:<ba-add>idx:2 (ifx:1, b8:76:3f:3f:99:ec), tid:0, ssn:4, winSize:64
I (25035) wifi:station: b8:76:3f:3f:99:ec leave, AID = 1, bss_flags is 658530, bss:0x3fcf571c
I (25035) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<0,0>, prof:1
W (25045) wifi:<ba-del>idx

AxelLin, 2022-02-12:
@xueyunfei998 

https://github.com/espressif/esp-idf/issues/8008#issuecomment-1033043740 reported same issue if use nvs flash wifi is checked it wont connect.
You should test with the reported esp-idf version (or at least the v4.4 branch) rather than asking people to use master tree.

xueyunfei998, 2022-02-24:
[image]

xueyunfei998, 2022-02-24:
hi @jasenharris 

I tested it with the commit you provided. Two ESP32s3 boards were used to write ap and STA respectively. STA can normally connect ap, and no problem was found.

In addition, the commit you provided was last October, and you can upgrade to the latest 4.4 for testing.

If there are any questions, please provide the sniffer.

jack0c, 2022-05-20:
@jasenharris @kayleejacob1 I suggest update to the latest IDFv4.4, since esp32s3 is first supported in IDFv4.4, lots of bug fixed in the duration.