espressif/esp-idf #8009: DOC: Deprecated functions have no non-deprecated alternative

bullestock, 2021-12-03:
The functions esp_vfs_dev_uart_set_tx_line_endings() and esp_vfs_dev_uart_set_rx_line_endings() have been marked as deprecated for some time now, but I have been unable to find out what their replacements are.

igrr, 2021-12-03:
Hi @bullestock, these would be 
```
int esp_vfs_dev_uart_port_set_tx_line_endings(int uart_num, esp_line_endings_t mode);
int esp_vfs_dev_uart_port_set_rx_line_endings(int uart_num, esp_line_endings_t mode);
```
(additional `port` argument to identify the UART port you want to change the line endings for)

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html#_CPPv441esp_vfs_dev_uart_port_set_tx_line_endingsi18esp_line_endings_t

bullestock, 2021-12-05:
@igrr thanks. I have created a pull request to add this to the docs.

bullestock, 2021-12-05:
Pull request #8013 addresses this.

closed 2021-12-05 by bullestock