espressif/esp-idf #8016: A problem with the serial port command calling websocket api

jiaxieya, 2021-12-06:
## Environment

Operating System: Windows
Using an IDE?: VS Code
Power Supply: USB
I (48) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (49) boot: compile time 16:03:22
I (49) boot: chip revision: 3
I (51) boot.esp32c3: SPI Speed : 80MHz
I (56) boot.esp32c3: SPI Mode : DIO
I (60) boot.esp32c3: SPI Flash Size : 4MB

## Problem Description

Try to parse the serial port to send data through the state machine during the serial port interruption and do the corresponding processing, such as connecting to the server, sending information to the server, ending the connection and clearing the used resources for the next connection. The current known problem is that the recursive mutex is used in the websocket api. After shielding the recursive mutex in esp_websocket_client.c, the problem is still unresolved. How to modify the use of api or other usage, while maintaining the use of serial port interrupts.

igrr, 2021-12-06:
@jiaxieya if you mean that the problem is that websocket functions cannot be called from an interrupt context, then this is expected. I recommend using a FreeRTOS queue or a ringbuffer in this case. In ISR context, you can push commands into a queue, then in task context, receive them from the queue and call websocket functions.

jiaxieya, 2021-12-06:
I used a ring buffer to store data. During interrupt processing, I was only responsible for storing the received character string. At the same time, there was a serial port detection task that continued to run to process the data in the buffer. At the same time of processing, The serial port interrupt part is received and stored in the buffer at a small interval, which makes me trouble using websocket api

AxelLin, 2021-12-06:
> I used a ring buffer to store data. 

FYI, there are 2 ringbuf bug fixes not yet available in v4.3 at the moment.
https://github.com/espressif/esp-idf/issues/7716 
https://github.com/espressif/esp-idf/issues/7344

jiaxieya, 2021-12-06:
> > I used a ring buffer to store data.
> 
> FYI, there are 2 ringbuf bug fixes not yet available in v4.3 at the moment. #7716 #7344

The fifo I use is not in the component, but a third-party component. There is no bug in using fifo at present

igrr, 2021-12-06:
@jiaxieya could you please try to supply more information for us to help you troubleshoot this? Please try to narrow down the issue you are seeing to some simpler piece of code you can share with us, or explain in sufficient detail what you are trying to do in your code.

jiaxieya, 2021-12-07:
> @jiaxieya能否请您尝试为我们提供更多信息以帮助您解决此问题？请尝试将您看到的问题缩小到可以与我们分享的一些更简单的代码段，或者足够详细地解释您要在代码中执行的操作。

I'm sorry that my statement can't cause trouble to you. My main program is a task to detect the length of the specified fifo. If the length of the fifo changes (data is received from the serial port interrupt), the internal data is parsed and processed, so I set three kinds of data to trigger the websocket There are three states of api, 1.esp_websocket_client_start 2.esp_websocket_client_send_bin 3.esp_websocket_client_destroy, there is no problem during the first run. When I want to execute it the second time, there is a problem in esp_websocket_client_start, and the error message is as follows. I tried to close the recursive mutex, but did not get the desired result.

Stack error message：

abort() was called at PC 0x40381a4d on core 0
0x40381a4d: lock_acquire_generic at D:/esp/esp-idf-4.3.1/components/newlib/locks.c:139

Stack dump detected
Core  0 register dump:
MEPC    : 0x403867a0  RA      : 0x40386e48  SP      : 0x3fcaf280  GP      : 0x3fc8e400
0x403867a0: panic_abort at D:/esp/esp-idf-4.3.1/components/esp_system/panic.c:379

0x40386e48: rtos_int_enter at D:/esp/esp-idf-4.3.1/components/freertos/port/riscv/portasm.S:35

TP      : 0x3fc903e8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fcaf2bc  S1      : 0x3fcaf2bc  A0      : 0x3fcaf2bc  A1      : 0x3fcaf29e
A2      : 0x00000000  A3      : 0x3fcaf2e9  A4      : 0x00000001  A5      : 0x3fc96000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcaf2a0  S3      : 0x3c0b9cbc
S4      : 0x0000001a  S5      : 0x3fcaf9c0  S6      : 0x00060000  S7      : 0x3fc95000
S8      : 0x00000000  S9      : 0xc0380000  S10     : 0x00000000  S11     : 0x3fc96330
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

igrr, 2021-12-07:
It seems that you are calling a blocking function from interrupt context, which isn't supported.

Please make sure you do minimal amount of work from the interrupt context and only call blocking functions from task context.

If you use [IDF Monitor](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/tools/idf-monitor.html#automatic-address-decoding) then you should be able to get a better backtrace which may point to the location of the issue.

jiaxieya, 2021-12-07:
> 似乎您正在从中断上下文调用阻塞函数，这是不受支持的。
> 
> 请确保您从中断上下文做最少的工作，并且只从任务上下文调用阻塞函数。
> 
> 如果您使用[IDF Monitor，](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/tools/idf-monitor.html#automatic-address-decoding)那么您应该能够获得更好的回溯，它可能指向问题的位置。

事实上我想通过串口转发数据到websocket的服务器，因为需要提升效率我开启了串口中断来将数据存入fifo，但是目前是模拟测试中第一遍流程不会出现任何问题，第二次连接之后，在发送指定指令给服务器，服务器方断开连接时造成此崩溃，对于 idf-monitor返回的信息我得不到有效操作或指示（似乎是对于esp32-c3支持不够友好），完整返回信息如下，如果使用gdb调试是否会有更好的调试效果。
abort() was called at PC 0x40381a4d on core 0
0x40381a4d: lock_acquire_generic at D:/esp/esp-idf-4.3.1/components/newlib/locks.c:139

Stack dump detected
Core  0 register dump:
MEPC    : 0x40386770  RA      : 0x40386e18  SP      : 0x3fcaf180  GP      : 0x3fc8e400
0x40386770: panic_abort at D:/esp/esp-idf-4.3.1/components/esp_system/panic.c:379

0x40386e18: rtos_int_enter at D:/esp/esp-idf-4.3.1/components/freertos/port/riscv/portasm.S:35

TP      : 0x3fc924f0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fcaf1bc  S1      : 0x3fcaf1bc  A0      : 0x3fcaf1bc  A1      : 0x3fcaf19e
A2      : 0x00000000  A3      : 0x3fcaf1e9  A4      : 0x00000001  A5      : 0x3fc96000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcaf1a0  S3      : 0x3c0b7a54
S4      : 0x3fc95000  S5      : 0x3fcaf86c  S6      : 0x3fc96000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'd:\\download\\esp32-esp32-test\\build\\esp-http-client-example.elf', '-ex', 'target remote | "C:\\Users\\user\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe" "d:\\esp\\esp-idf-4.3.1\\tools\\..\\tools\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\user\\AppData\\Local\\Temp\\tmpfa6gavwq"', '-ex', 'bt']' returned non-zero exit status 3221225781.
b''

Core  0 register dump:
MEPC    : 0x40386770  RA      : 0x40386e18  SP      : 0x3fcaf180  GP      : 0x3fc8e400
TP      : 0x3fc924f0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fcaf1bc  S1      : 0x3fcaf1bc  A0      : 0x3fcaf1bc  A1      : 0x3fcaf19e
A2      : 0x00000000  A3      : 0x3fcaf1e9  A4      : 0x00000001  A5      : 0x3fc96000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcaf1a0  S3      : 0x3c0b7a54
S4      : 0x3fc95000  S5      : 0x3fcaf86c  S6      : 0x3fc96000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fcaf180: 0x00000000 0xfffffffc 0x3fcaf1bc 0x4038c2dc 0x3c0a678d 0x3fcaf310 0x3fc96c90 0x40380030
3fcaf1a0: 0x38333034 0x64346131 0x3fc96a00 0x3fc90b6c 0x3fcaf1a0 0x3fc90b88 0x3fcaf19c 0x726f6261
3fcaf1c0: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x31383330 0x20643461
3fcaf1e0: 0x63206e6f 0x2065726f 0x00000030 0x00000000 0x00000004 0xffffffff 0x3fc96cf0 0x40381a50
3fcaf200: 0x00000000 0x0000000a 0x3fcaf86c 0x3fc96af4 0x3fcaf804 0x00000001 0x3fc96c90 0x40387992
3fcaf220: 0x3c0a678c 0x00000003 0x00000006 0x3c0b7a54 0x3fc96b5c 0x3fc9aed0 0x3fc96b5c 0x40381c4e
3fcaf240: 0x00000004 0x00000000 0x00000006 0x3fc96cf0 0x3c0a5e93 0x00000064 0x00000004 0x42086fe0
3fcaf260: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000032 0x00000001 0x00000004
3fcaf280: 0x00000000 0x3c0b6244 0x00000000 0x3fcaf658 0x00000000 0x00000000 0xffffffff 0x7fefffff
3fcaf2a0: 0x00000000 0x3fe00000 0x00000001 0x00000001 0x00000000 0x3fc00000 0x00000000 0x40300000
3fcaf2c0: 0x00000000 0x00000000 0xffffffff 0x7fefffff 0x00000000 0x3fe00000 0x00000000 0x3fcaf6b8
3fcaf2e0: 0x00000000 0x00c00000 0x00000000 0x00000003 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0x353234d4
3fcaf300: 0x3fcaf30c 0x00000000 0x00000000 0x343032d8 0x3fcaf31c 0x00000000 0x00000000 0x3c0a678c
3fcaf320: 0x00000006 0x3fcaf3bc 0x00000004 0x363237a5 0x00000008 0x00000000 0x00000000 0x3c0a6ef0
3fcaf340: 0x3fcaf34c 0x00000000 0x00000000 0x3c0a5e93 0x00000007 0x3fcaf3ef 0x00000001 0xa5a5a5a5
3fcaf360: 0x3fcb29cc 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcaf380: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcaf3a0: 0x00000009 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000023 0xa5a5a5a5 0x3232a5a5 0x36383032
3fcaf3c0: 0xffffffff 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000015 0xa5a5a5a5 0xa5a5a5a5 0x403891d2
3fcaf3e0: 0xffffffff 0xa5a5a5a5 0xa5a5a5a5 0x31000000 0xffffffff 0xa5a5a5a5 0x3fcafdd4 0x4038728e
3fcaf400: 0x00000000 0x00000000 0x3fcafdd4 0x403877d6 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x403891d2
3fcaf420: 0x00000000 0x00000000 0x3fca3e94 0x00000000 0x3fc91658 0x3fcaf50c 0x3fcafdd4 0x4038728e
3fcaf440: 0x3fc91658 0x00000000 0x3fcafdd4 0x403877d6 0x3c0a6768 0x3fcaf50c 0x00000000 0x00000000
3fcaf460: 0x00000826 0x3fcaf50c 0x3fcaf514 0x00000000 0x00000001 0x00000018 0x00000001 0x0000003f
3fcaf480: 0x3fc925b8 0x00000000 0x00000000 0x42025f64 0x00000000 0x00000001 0x00000000 0x42013a12
3fcaf4a0: 0x00000001 0x0000003f 0x00000000 0x00000000 0x40389a16 0x403877f6 0x00000000 0x40389392
[... cut ...]
3fcaf4e0: 0x00000001 0x00000001 0x00000000 0x40387e0a 0x3fcaf624 0x00002710 0x00000000 0xffffffff
3fcaf500: 0x00000000 0x00000002 0x3fca40d0 0x403891d2 0x00000004 0x00000000 0x3fc92168 0x00000000
3fcaf520: 0x00000004 0x00000000 0x3fc92168 0x4038728e 0x00000000 0x00000000 0x3fc92168 0x403877d6
3fcaf540: 0x00000000 0xffffffff 0x3fc92168 0x00000000 0x3fcaf5dc 0x3fcaf86c 0x3c0b8a78 0x3fc96000
3fcaf560: 0x3fcaf5ec 0x3fcaf86c 0x3c0b8a78 0x3fc96000 0x3fcaf804 0x00000001 0x3fc92168 0x40387992

ELF file SHA256: 916b2ba780e68a58
ELF file SHA256: 916b2ba780e68a58

closed 2021-12-27 by jiaxieya