espressif/esp-idf #8017: tinyusb serial not work with nimble gatt server

ShowTimeCreator, 2021-12-06:
## Environment

- Development Kit:     ESP32-S3-DevKitC-1
- Kit version : v1
- Module or chip used:  ESP32-S3-WROOM-1（ESP32-S3R2）
- IDF version: release/v4.4
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Power Supply: USB
## Problem Description

/dev/ttyACM0 will lost if create nimble gatt server, sometime can found but can't input and output, tinyusb and nimble work well independently.

```cpp
...
ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));
...
ESP_ERROR_CHECK(tusb_cdc_acm_init(&amc_cfg));
...
ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());
...

```

igrr, 2021-12-06:
This might be fixed by commit https://github.com/espressif/esp-idf/commit/65f3bf4a8a21f6ef51649c3202ca282e9b04d124 on master branch, but this commit is not in release/v4.4 yet. Could you give this commit a try, please?

ShowTimeCreator, 2021-12-07:
Thank you for your reply. It work perfectly on master branch.

closed 2021-12-07 by ShowTimeCreator

Alvin1Zhang, 2022-09-23:
Thanks for reporting, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/81ce15f0e088fd28578d340287fdaf7f28879efd