espressif/esp-idf #8018: How do I create a flash memory partition in esp32 without using csv file ?

enthunilu, 2021-12-06:
Development Kit: ESP32-Ethernet kit_A_V1.1
Kit version
Module or chip used: ESP32
IDF version :v4.2.1
Build System: Make|CMake|idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool NCG esp-2020r3)8.4.0
Operating System: Windows
(Windows only) environment type: Command Prompt
Using an IDE?: Yes (Eclipse)

Power Supply: USB

## Problem Description

//Detailed problem description goes here.

I am currently checking for an option to create memory partition in my esp32 board.
I am able to create it by using my partition.csv file.

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000,  1M,
ota_0,    app,  ota_0,   0x110000, 1M,
ota_1,    app,  ota_1,   0x210000, 1M,
log_1,    data,  log_1,  0x310000, 0x4000

like this. However, I wanted to create partition run time. Is there a way or API's available with which we can create partition run time ?

igrr, 2021-12-06:
You can do this by calling `esp_partition_register_external`  function (see e.g. [this example](https://github.com/espressif/esp-idf/blob/98ad01e5fc0779d11209350d8ec31fe3d2137ed5/examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c#L142)), passing `esp_flash_default_chip` as the first argument. The new partition can be added only if it doesn't overlap any of the already defined partitions.
However adding partitions in the main flash at run time is **not** recommend, since there is nothing preventing a subsequent OTA update from unintentionally configuring the partitions differently.

enthunilu, 2021-12-17:
Thank you @igrr. It worked for me. And yes you are right about overwriting of memory part, that we need to take care in code.

enthunilu, 2021-12-17:
I am closing this issue.

closed 2021-12-17 by enthunilu