espressif/esp-idf #8019: sig mesh出现assert rwble.c 178, param 00000000 00000000后卡死

liangzc1, 2021-12-06:
使用bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server例程，多个设备放在一起配同一个mesh网络，入网后频繁打印
[19:21:14.581]收←◆[0;31mE (1327897) BLE_MESH: Out of relay buffers[0m

[19:21:14.629]收←◆[0;31mE (1327947) BLE_MESH: Out of relay buffers[0m

[19:21:15.459]收←◆[0;31mE (1328777) BLE_MESH: Out of relay buffers[0m

[19:21:15.717]收←◆[0;31mE (1329037) BLE_MESH: Out of relay buffers[0m

[19:21:15.772]收←◆[0;31mE (1329087) BLE_MESH: Out of relay buffers[0m

[19:21:15.798]收←◆[0;31mE (1329117) BLE_MESH: Out of relay buffers[0m

一段时间后有设备打印assert rwble.c 178, param 00000000 00000000，看门狗没有被触发，设备未重启，并且程序卡死

closed 2021-12-06 by liangzc1

InfiniteYuan, 2022-06-07:
Please use the latest esp-idf and test again.

InfiniteYuan, 2022-06-07:
@liangzc1  Is this problem solved?