espressif/esp-idf #8020: Proposal to change the memory allocation to support modules with or without PSRAM

cicciocb, 2021-12-06:
IN the actual implementation, if CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL is defined to 1 and the module does not have PSRAM, the program crashes because it try to allocate memory exclusively from the PSRAM.

I propose to slightly change the memory allocation routines in order to allocate the memory even if the PSRAM is not available.
This will enable to put the same code in modules with or without PSRAM.
This can be done in the file **esp-idf/components/bt/host/nimble/port/src/esp_nimble_mem.c**

The change is very simple,

IRAM_ATTR void *nimble_platform_mem_malloc(size_t size)
{
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    //return heap_caps_malloc(size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
	**return heap_caps_malloc_prefer(size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);**
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_malloc_prefer(size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#else
    return malloc(size);
#endif
}

IRAM_ATTR void *nimble_platform_mem_calloc(size_t n, size_t size)
{
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_calloc(n, size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    //return heap_caps_calloc(n, size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
	**return heap_caps_calloc_prefer(n, size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);**
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_calloc_prefer(n, size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#else
    return calloc(n, size);
#endif
}

Regards
cicciocb

rahult-github, 2021-12-08:
Hi @cicciocb ,

Via menuconfig, the option to enable  BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL is only possible if External SPIRAM is available: 

 As seen [here](https://github.com/espressif/esp-idf/blob/4143a3f572f6060258ffac91f110a095a8478966/components/bt/host/nimble/Kconfig.in#L18)  
   

>     config BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
>        bool "External SPIRAM"
>       depends on SPIRAM_USE_CAPS_ALLOC || SPIRAM_USE_MALLOC

Every chip has a option to enable / disable support for SPIRAM . E.g. [here](https://github.com/espressif/esp-idf/blob/4143a3f572f6060258ffac91f110a095a8478966/components/esp32/Kconfig#L68)

If user has NOT enabled External SPIRAM support , the above mentioned Nimble option will not be visible and hence cannot be enabled.

So, the code you mentioned, should not get hit, if the above options are not enabled.  

Also, one more point is, in case , we do add the changes as suggested, it may break some other use case ( i.e. no SPIRAM available, but we still allocate memory from internal memory and attempt to proceed). 
  
Thanks,
 Rahul

leeebo, 2021-12-08:
HELLO
IM LIBO
I HAVE GOT YOUR LETTER
PLEASE WAIT FOR REPLY

cicciocb, 2021-12-08:
> Hi @cicciocb ,
>
> Via menuconfig, the option to enable BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL is only possible if External SPIRAM is available:
>
> As seen [...]

Thanks Rahul,
the idea was the integration of the same binary in modules with or without PSRAM, without the need to compile separate versions, obviously enabling the option in menuconfig.

However, I understand your point of view to have something "unbreakable" so that thanks for your answer and consideration.

The point can be closed

Regards
cicciocb

closed 2021-12-08 by cicciocb