espressif/esp-idf #8021: Passing variables when building

silardg, 2021-12-06:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
- v2.12-rc3-215-ga160b3a
- Build System:        idf.py]

## Problem Description

I would like to pass a variable when I compile my project that will determine if my project is release or debug. 

I'm not looking to use KConfig, because it would mess with code-versioning.

I was using the command `idf.py build -DCONFIG_SYSTEM_RELEASE_MODE=0`
And in the source code: `#define SYSTEM_RELEASE_MODE @CONFIG_SYSTEM_RELEASE_MODE@`

igrr, 2021-12-06:
You can pass additional flags to C compiler, C++ compilers, or the preprocessor using `EXTRA_CFLAGS`, `EXTRA_CXXFLAGS`, `EXTRA_CPPFLAGS` environment variables, respectively. So in this case you can add `-DCONFIG_SYSTEM_RELEASE_MODE=0` to `EXTRA_CPPFLAGS`.

Alternatively, you can add arbitrary flags in your project CMakeLists.txt:

```
if(CONFIG_SYSTEM_RELEASE_MODE)
  # define preprocessor macro with the same name
  add_compile_definitions(CONFIG_SYSTEM_RELEASE_MODE=1)
else()
  add_compile_definitions(CONFIG_SYSTEM_RELEASE_MODE=0)
endif()
```
with that, defining `CONFIG_SYSTEM_RELEASE_MODE` CMake variable (`idf.py -DCONFIG_SYSTEM_RELEASE_MODE=1`) will result in `-DCONFIG_SYSTEM_RELEASE_MODE=1` to be passed to preprocessor.

(Note, I would _recommend against_ using `CONFIG_` prefix in the names of these variables or macros, as normally in ESP-IDF project things starting with `CONFIG_` come from / are defined in Kconfig. It wouldn't be the case here, hence your hypothetical colleague might initially be confused looking at this macro and think that it relates to a Kconfig setting.)

ESP-Marius, 2022-08-31:
Sorry for the late reply @silardg 

The example provided above should also work fine with C++.

E.g. adding:
```
if(CONFIG_SYSTEM_RELEASE_MODE)
    # define preprocessor macro with the same name
    add_compile_definitions(CONFIG_SYSTEM_RELEASE_MODE=1)
else()
    add_compile_definitions(CONFIG_SYSTEM_RELEASE_MODE=0)
endif()
```
In [examples/cxx/exceptions/main/CMakeLists.txt](https://github.com/espressif/esp-idf/blob/master/examples/cxx/exceptions/main/CMakeLists.txt)

and

```
#if CONFIG_SYSTEM_RELEASE_MODE
#error "CONFIG_SYSTEM_RELEASE_MODE was defined"
#endif
```
In [examples/cxx/exceptions/main/exception_example_main.cpp](https://github.com/espressif/esp-idf/blob/master/examples/cxx/exceptions/main/exception_example_main.cpp)

gives the expected result, i.e. a build error with the message `"CONFIG_SYSTEM_RELEASE_MODE was defined`