espressif/esp-idf #8022: 'wifi_promiscuous_pkt_t' 'noise_floor' unit [esp-now]

MacWyznawca, 2021-12-06:
## Environment

- Development Kit:      ESP32-Wrover-IE
- IDF version: v5.0-dev-489-gef98a363e3-dirty
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- Using an IDE?: YES
- Power Supply:        custom 3,3 V

## Problem Description

Probably wrong unit in description wifi_promiscuous_pkt_t noise_floor unit
Not 0,25 dBm but 1 dBm

### Actual Behavior

ex. -90 or similar value without calculation

[image]

### Code to reproduce this issue

    wifi_promiscuous_pkt_t *promiscuous_pkt = (wifi_promiscuous_pkt_t *)(data - sizeof(wifi_pkt_rx_ctrl_t) - 39/*sizeof(espnow_frame_format_t)*/);
    wifi_pkt_rx_ctrl_t *rx_ctrl = &promiscuous_pkt->rx_ctrl;

    ESP_LOGE(LUAIR_TAG, "RSSI: %d, Noise %d Channel: %d", rx_ctrl->rssi, rx_ctrl->noise_floor, rx_ctrl->channel);

(data - data from ESP-NOW callback)

MaxwellAlan, 2022-03-24:
hi @MacWyznawca 
Thanks for your reporting, you are right. We will update these comments and docs.

jack0c, 2022-04-25:
The code have been merged to master, you can find the fix in 68d4c47b7ef509ce1ae66714d2fe0d741e49d188

closed 2022-05-06 by Alvin1Zhang