espressif/esp-idf #8023: LAN8720A stop working

SERIDJ, 2021-12-06:
I'm trying to run the ethernet example with my esp32, using a custom breakout board based on the schematic for the Waveshare LAN8720 ETH board : https://www.waveshare.com/LAN8720-ETH-Board.htm.

so if i use Waveshare LAN8720 ETH board with esp32 kit its works fine all time but when i use my custom board its work for around 3 month without reboot.
when i reboot board sometimes PHY LAN cant connect to the network and get the ip adress 

output log when its work fine : 
``
I (1165) esp_eth.netif.glue: ethernet attached to netif
I (1177) eth_example: Ethernet Started
I (5177) eth_example: Ethernet Link Up
I (5177) eth_example: Ethernet HW Addr 7c:9e:bd:e1:39:23
I (6104) esp_netif_handlers: eth ip: 192.168.0.168, mask: 255.255.255.0, gw: 192.168.0.254
I (6105) eth_example: Ethernet Got IP Address
I (6108) eth_example: ~~~~~~~~~~~
I (6112) eth_example: ETHIP:192.168.0.168
I (6116) eth_example: ETHMASK:255.255.255.0
I (6121) eth_example: ETHGW:192.168.0.254
I (6126) eth_example: ~~~~~~~~~~~
I (6130) eth_example: exturn derived_ip_addr_string from Ethernet  = 192.168.0.168 
```

output log when LAN cant connect correctly : 
``
I (1129) system_api: Base MAC address is not set
I (1129) system_api: read default base MAC address from EFUSE
I (1151) esp_eth.netif.glue: 24:62:ab:f3:20:73
I (1151) esp_eth.netif.glue: ethernet attached to netif
I (1163) eth_example: Ethernet Started
``

david-cermak, 2021-12-07:
Hi @SERIDJ 

Could you please tell us which IDF version are you using?

>  if i use Waveshare LAN8720 ETH board with esp32 kit its works fine all time but when i use my custom board

We cannot help much with this issue, even if the schematic is the same, there could be some problems with routing.

Could you please check whether the problem is not related to https://github.com/espressif/esp-idf/issues/7227 for example or if https://github.com/espressif/esp-idf/commit/8e069dd7555ae06465e2fa7798ee7a9d1adbe689 would possibly the initialisation issue?

SERIDJ, 2021-12-07:
hi @david-cermak thanks for ur reply .

-So for the version iam using  v4.3-dirty.

- Iam usign external Oscillator 50 Mhz connect to the LAN (Pin 5 XTAL1/CLKIN ) and ESP32 G0.

-to avoid problem of initialisation  i use GPIO5 to control oscillator : https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

So i use     rest phy LAN8720A to enbale Oscillator ``` phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;```
and nRST pin for LAN is connected to vcc with 4.7k resistor. ( i dont use hard reset for my LAN in this case)

Iam not sure if this is good good idea knowing that we use this board for industrial ...

sauttefk, 2021-12-07:
I also ran in such a problem, that the LAN8720 was not responding sometimes after a warm start (reboot), but I'm clocking the LAN8720 with GPIO17.
Cold starts always worked. So I connect the nRST pin of the LAN8720 (which costed me another valuable GPIO-pin).
But unfortunately this also did not solve the problem completely and the nRST low phase had to be increased #7811, as otherwise the LAN8720 was in some zombie state.

SERIDJ, 2021-12-07:
Hi @sauttefk  ur using a custom or Waveshare board ? bcs me with Waveshare board and without nRST control from soft its work all time cold and warm start ....

SERIDJ, 2021-12-07:
i try to work in board with LAN cant connect to network and cant detect cable link up/down.

so what i did in first its use hardawre rest (nRST) with GPIO13 and enbale oscillator with GPIO5.

before i start Ethernet i enable first my oscillator.

result its same my log output stop with  I (6786)  ``` eth_example: Ethernet Started ``` 

something strange if i touch nRST pin with physical cable , i have link up and there is some trafic with emac and green led start to blink 
 ```

I (6786) eth_example: Ethernet Started
D (28732) emac_esp32: working in 10Mbps
D (28732) emac_esp32: working in half duplex
D (28732) emac_esp32: Flow control not enabled for the link
D (28735) event: running post ETH_EVENT:2 with handler 0x400f7e6c and context 0x3ffdce44 on loop 0x3ffdbc64
0x400f7e6c: esp_netif_action_connected at C:/Users/Adminprogiris/esp/esp-idf/components/esp_netif/esp_netif_handlers.c:44

D (28745) esp_netif_handlers: esp_netif action connected with netif0x3ffdcb90 from event_id=2
D (28753) esp_netif_lwip: check: remote, if=0x3ffdcb90 fn=0x400f8ea8
0x400f8ea8: esp_netif_up_api at C:/Users/Adminprogiris/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1203

D (28760) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffdcb90
D (28766) esp_netif_lwip: check: local, if=0x3ffdcb90 fn=0x400f977c
0x400f977c: esp_netif_update_default_netif_lwip at C:/Users/Adminprogiris/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:174

D (28772) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffdcb90
D (28779) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (28785) esp_netif_lwip: check: remote, if=0x3ffdcb90 fn=0x400f893c
0x400f893c: esp_netif_dhcpc_start_api at C:/Users/Adminprogiris/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1000

D (28792) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffdcb90
D (28798) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffdcb90
D (28805) esp_netif_lwip: if0x3ffdcb90 start ip lost tmr: no need start because netif=0x3ffdcc00 interval=120 ip=0
D (28816) esp_netif_lwip: starting dhcp client
D (28821) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (28826) event: running post ETH_EVENT:2 with handler 0x400db468 and context 0x3ffdceec on loop 0x3ffdbc64
0x400db468: eth_event_handler at C:\Users\Adminprogiris\Desktop\vscode_workspace\softhyperboxv1v0\build/../main/connectionethernet.c:29

I (28836) eth_example: Ethernet Link Up
I (28841) eth_example: Ethernet HW Addr 24:62:ab:f3:20:73
D (29568) emac_esp32: receive len= 778
D (30035) emac_esp32: receive len= 60
D (30054) emac_esp32: receive len= 86
D (30055) emac_esp32: receive len= 86
D (30069) emac_esp32: receive len= 86
D (30088) emac_esp32: receive len= 86
D (30088) emac_esp32: receive len= 86
D (30135) emac_esp32: receive len= 86
D (30148) emac_esp32: receive len= 86
D (30150) emac_esp32: receive len= 86
D (30302) emac_esp32: receive len= 60
D (30302) emac_esp32: receive len= 86
D (30476) emac_esp32: receive len= 86
D (31969) emac_esp32: receive len= 60
D (32280) emac_esp32: receive len= 60
D (32789) emac_esp32: receive len= 

 ```

SERIDJ, 2021-12-08:
i found this forum : https://www.eevblog.com/forum/microcontrollers/no-ethernet-phy-connectivity-with-esp-32-wroom/25/

Did the wESP32 developer already tell you that you need to drive NRST properly? Simply pulling it high permanently will not work and the Olimex hack with a cap does not work for a warm reset. I've been running into this issue myself in the last few days, to where the LAN8710A resets continuously.

 ```
I assume you don't care about using an extra pin, so you can make sure to drive NRST properly:

1. Pull NRST low
2. Wait for the 50mhz clock to come up (it might be already up)
3. Wait for a small period (100us worked in my case)
4. Pull NRST high

For me this makes the LAN8710a run rock solid and more importantly it can handle warm resets/debugging sessions. My code: https://github.com/tinic/lightguy/blob/master/ethernetif.cpp#L73 Code is for an STM32F107 clone, not the ESP32, so not really useful for you :-)

 ```

SERIDJ, 2021-12-08:
@sauttefk , 

in my case cold start not necessarily always worked

  i test now my board , with the hardware reset the problem is not fixed ... 

 ```
*****************************************
*****************************************
*****************************************
****************normal Eth***************
*****************************************
*****************************************
*****************************************
*****************************************
enbale oscillator
enbale done

*****************************************
*****************************************
*****************************************
****************eth dhcp***************
*****************************************
*****************************************
*****************************************
*****************************************
esp_rom_gpio_pad_select_gpio(lan8720->reset_gpio_num) is : 13
nrst LOW 100us
nrst HIGH
I (1087) system_api: Base MAC address is not set
I (1091) system_api: read default base MAC address from EFUSE
I (1118) esp_eth.netif.glue: 24:0a:c4:5f:82:57
I (1118) esp_eth.netif.glue: ethernet attached to netif
I (1130) eth_example: Ethernet Started
 ```

SERIDJ, 2021-12-09:
@david-cermak hi, please u can tell me how i can check registre after reset in soft ?

[image]

kostaond, 2021-12-10:
Hi @SERIDJ,

> one think its about order , bcs with esp idf we cant pull nRst low and wait for clk ,

You should be able to achieve that order by following:
- Pull down the nRST pin to low by external 10k pull-down resistor. When you reset ESP32, then its GPIO is in default state and keeps the nRST low (for more information about GPIO reset state, please see TRM Section 4.10). Note that this is proven to work with GPIO5 initialized as input with internal PU (see [here](https://dl.espressif.com/dl/schematics/SCH_ESP32-Ethernet-Kit_A_V1.2_20200528.pdf), just different PHY is used). However, GPIO13 should be even better since it is reset to input with PD.
- Enable your CLK.
- Just start the Ethernet driver. It will initialize the reset GPIO to output mode, drives it to low (e.g. still no state change at nRST), waits 150us and drives the output to high which will take the LAN8720 out of reset (see `lan8720_reset_hw` for the sequence).

> in my case cold start not necessarily always worked

Did you tried it with 150us reset time as suggested by @sauttefk?

Please give it a try and let us know if it helped.

kostaond, 2022-01-03:
@SERIDJ, any update on the issue?

closed 2022-01-29 by Alvin1Zhang