espressif/esp-idf #8028: esp_hid_host demo: printing in callback causes stackoverflow or freeze

ChrisIdema, 2021-12-07:
https://github.com/espressif/esp-idf/blob/2ee9140e941fb102830bb023a939a8b4c9b5eace/examples/bluetooth/esp_hid_host/main/esp_hid_host_main.c#L49

When connecting a mouse and moving it around very fast for a few seconds causes a stack overflow or a freeze. 
Message when a stack overflow is detected:

>  ***ERROR*** A stack overflow in task esp_bt_hidh_eve has been detected.

Increasing the stack size of "esp_bt_hidh_events" task from 2048 to 4096 prevents a stack overflow when using a BT keyboard, but a BT mouse generates more events and still triggers a stack overflow or freeze.

A simple workaround I used is to check the system time and to only print after 200 milliseconds have passed since the last printing in this callback.

igrr, 2021-12-07:
@ChrisIdema I think this should be addressed by https://github.com/espressif/esp-idf/commit/d1c8c268d47570cbc89b0c4568b4125b2bf3b935 and https://github.com/espressif/esp-idf/commit/5266d816022bd1c3636cb4759601bacaca94c37d.

Sorry to ask the question from the issue template, but which IDF version are you using?

ChrisIdema, 2021-12-07:
@igrr 
I use the same version of IDF as from the example. In this case commit 2ee9140e941fb102830bb023a939a8b4c9b5eace .

Like I wrote d1c8c268d47570cbc89b0c4568b4125b2bf3b935 doesn't fix the stackoverflow for the mouse.
I will check out 5266d816022bd1c3636cb4759601bacaca94c37d .

jack0c, 2022-01-09:
@ChrisIdema Can you try ade22c907264d180a3d7c67a6c0a77dcd7e0b105  which fix a heap issue?

ChrisIdema, 2022-01-19:
I did not try it. I'm not working on this project anymore.

closed 2022-01-24 by Alvin1Zhang