espressif/esp-idf #8031: Crash (LoadProhibited) in BLE disconnect in "Wifi provisioning" example

QuentinFarizon, 2021-12-07:
## Environment

- Development Kit: M5Stack Atom
- Module or chip used: ESP32-PICO-D4
- IDF version : v4.3.1-354-g98d34e5f6d
- Build System:  idf.py
- Compiler version :  xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I am flashing the unmodified wifi provisoning example (provisioning/wifi_prov_mgr) and encountering a crash at the end of a successful provisioning.
I reproduce it around 80% of times (reproduced it several dozen times, and a few times it didn't crash).

The only config I modified is the log level.

### Expected Behavior

No crash

### Actual Behavior

Here is the stack :
```
I (58878) app: Hello World!
I (59878) app: Hello World!
D (60608) protocomm_nimble: disconnect; reason=531 
D (60608) protocomm_nimble: Inside disconnect w/ session - 0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400dbf11  PS      : 0x00060330  A0      : 0x800dc8d3  A1      : 0x3ffd8350  
0x400dbf11: transport_simple_ble_disconnect at /home/quentin/esp/esp-idf/components/protocomm/src/transports/protocomm_nimble.c:540

A2      : 0x3ffd83fc  A3      : 0x00000000  A4      : 0x3f40c664  A5      : 0x0000ecc0  
A6      : 0x3f40bbcc  A7      : 0x00000213  A8      : 0x00000000  A9      : 0x3ffd8300  
A10     : 0x00000004  A11     : 0x3f40bbcc  A12     : 0x3f40bbe0  A13     : 0x0000ecc0  
A14     : 0x3f40bbcc  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace:0x400dbf0e:0x3ffd8350 0x400dc8d0:0x3ffd8370 0x400e11e5:0x3ffd83d0 0x400e2381:0x3ffd83f0 0x400e23ae:0x3ffd8450 0x400de243:0x3ffd84b0 0x400de5f1:0x3ffd84d0 0x400ddc89:0x3ffd84f0 0x400e6cfe:0x3ffd8510 0x400dc122:0x3ffd8530 0x40093ee9:0x3ffd8550
0x400dbf0e: transport_simple_ble_disconnect at /home/quentin/esp/esp-idf/components/protocomm/src/transports/protocomm_nimble.c:540
0x400dc8d0: simple_ble_gap_event at /home/quentin/esp/esp-idf/components/protocomm/src/transports/protocomm_nimble.c:236 (discriminator 9)
0x400e11e5: ble_gap_call_event_cb at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:720
0x400e2381: ble_gap_conn_broken at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1257
0x400e23ae: ble_gap_rx_disconn_complete at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1284 (discriminator 4)
0x400de243: ble_hs_hci_evt_disconn_complete at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:196
0x400de5f1: ble_hs_hci_evt_process at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:857
0x400ddc89: ble_hs_event_rx_hci_ev at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:520
0x400e6cfe: ble_npl_event_run at /home/quentin/esp/esp-idf/examples/provisioning/wifi_prov_mgr/build/../../../../components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at /home/quentin/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78
0x400dc122: nimble_host_task at /home/quentin/esp/esp-idf/components/protocomm/src/transports/protocomm_nimble.c:471
0x40093ee9: vPortTaskWrapper at /home/quentin/esp/esp-idf/components/freertos/port/xtensa/port.c:168
```

Please find full debug log :
[result.txt](https://github.com/espressif/esp-idf/files/7671624/result.txt)

### Steps to reproduce
```
cd examples/provisioning/wifi_prov_mgr
idf.py flash monitor
```
And successfully connect to a Wifi AP

laukik-hase, 2021-12-08:
Hello, @QuentinFarizon

Thanks for the issue. But this has already been fixed in the commit [`2166619`](https://github.com/espressif/esp-idf/commit/21666192d5ab805af05563b805b315c90e7dc4a6) on branch `release/v4.3`. This commit updated the `nimble` submodule pointer which included the fix for this bug.

QuentinFarizon, 2021-12-08:
Nice ! Thank you very much for yout work !

closed 2021-12-08 by QuentinFarizon