espressif/esp-idf #8032: In packages\framework-espidf\components\bt\common\osi\fixed_queue.c not checking for NULL pointers

norventa, 2021-12-07:
When the error label in fixed_queue_new is jumped to, fixed_queue_free gets called pasing ret that might contain NULL pointers: ret->enqueue_sem or ret->dequeue_sem.
In fixed_queue_free, NULLs are not tested for: osi_sem_free(&queue->enqueue_sem) and osi_sem_free(&queue->dequeue_sem).
A dereference null pointer exception will result.

I have not checked but there could be other such issues.

`
fixed_queue_t *fixed_queue_new(size_t capacity)
{
    fixed_queue_t *ret = osi_calloc(sizeof(fixed_queue_t));
    if (!ret) {
        goto error;
    }

    osi_mutex_new(&ret->lock);
    ret->capacity = capacity;

    ret->list = list_new(NULL);
    if (!ret->list) {
        goto error;
    }

    osi_sem_new(&ret->enqueue_sem, capacity, capacity);
    if (!ret->enqueue_sem) {
        goto error;
    }

    osi_sem_new(&ret->dequeue_sem, capacity, 0);
    if (!ret->dequeue_sem) {
        goto error;
    }

    return ret;

error:;
    fixed_queue_free(ret, NULL);
    return NULL;
}

void fixed_queue_free(fixed_queue_t *queue, fixed_queue_free_cb free_cb)
{
    const list_node_t *node;

    if (queue == NULL) {
	    return;
	}

    fixed_queue_unregister_dequeue(queue);

    if (free_cb) {
        for (node = list_begin(queue->list); node != list_end(queue->list); node = list_next(node)) {
            free_cb(list_node(node));
        }
    }

    list_free(queue->list);
    osi_sem_free(&queue->enqueue_sem);
    osi_sem_free(&queue->dequeue_sem);
    osi_mutex_free(&queue->lock);
    osi_free(queue);
}
`

JustCteator, 2021-12-09:
I have encountered a similar problem. I use Bluedroid to transmit a data stream. On some smartphones, the Bluetooth channel is disconnecting from the esp32 in 5 second after connected. This causes the problem described here on the esp 32 side - an exception and a reboot of the chip.

xiewenxiang, 2021-12-31:
Thanks for your advice. We will add check for it.

Alvin1Zhang, 2024-12-17:
Thanks for reporting, sorry for the very slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/800f360cc3e76faa3875b4076c013f2656af8d74, feel free to reopen.

closed 2024-12-17 by Alvin1Zhang