espressif/esp-idf #8033: Processor hang during crash

cjstott94, 2021-12-08:
## Environment
- Development Kit: Custom PCB
- Module or chip used:  ESP32-WROOM-32E (8MB)
- IDF version: 4.3.0
- Build System: CMake
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt/PlatformIO
- Using an IDE?: Yes, VSCode + PlatformIO(5.2.3) + espressif32(3.3.2+sha.e170b13)
- Power Supply: external 3.3V

## Problem Description

Processor hang during crash

### Expected Behavior

Any crashes should result in CPU restarting

### Actual Behavior

Processor Halts

### Steps to reproduce

1. Firmware built with CONFIG_ESP32S2_PANIC_PRINT_REBOOT and CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH enabled
2. Download and apply new firmware update over a ppp connection 
3. Initiate restart via esp_restart()
4. After esp_restart is called a uart task was erroneously still active and calls esp_netif_receive [here](https://github.com/espressif/esp-idf/blob/233dc30fb1a376d7ca0c5d74bdd410ca368f6bf7/components/esp_netif/lwip/esp_netif_lwip.c#L885)
5. InstrFetchProhibited Exception occurs probably due to a esp_netif handle being freed (PC is 0xfefefefe)
6. Processor halts after printing ELF sha256 (somewhere between [here](https://github.com/espressif/esp-idf/blob/39cbf2f7ed4810913031055065da08227020e8b4/components/esp_system/panic.c#L282-L342))

[image]

I'm Guessing that most likely something happens when saving a coredump in elf format to flash similar to this issue #6519
LWIP stack has 4K of memory allocated
Though I don't get an indication of a double exception over uart
(We don't have an easy way of hooking up jtag ATM)

I can probably fix this specific crash by ensuring the modem/ppp is cleaned up properly before a restart 
and/or 
by increasing the stack size to give extra overhead for the coredump as suggested here @gerekon in https://github.com/espressif/esp-idf/issues/6519#issuecomment-779968116_

Though my main concern is if a problem is occurring during the coredump, why is it not doing some sort of hard reset?
**It's critical that the application reboot on any sort of exception.** 

Is there a way we could adjust the [_DoubleExceptionVector](https://github.com/espressif/esp-idf/blob/7bd93e3362e212b1e27fbd19a7a45cbe8e0b46ff/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S#L458) so that it instead reboots straight away instead of invoking the panic handler?

Or is there any other known workaround? Maybe skipping core dump if there isn't enough stack space?

## Other items if possible
- sdkconfig file [sdkconfig.txt](https://github.com/espressif/esp-idf/files/7672942/sdkconfig.txt)

KaeLL, 2021-12-08:
Reminds of [this](https://github.com/espressif/esp-idf/issues/3810#issuecomment-817994026).
@igrr

cjstott94, 2021-12-09:
I've so far narrowed where it's halting down to 
[esp_panic_handler()](https://github.com/espressif/esp-idf/blob/d0dd9ce1b9825c29ae2b462556c4bf3965ab774b/components/esp_system/panic.c#L347)
=>[esp_core_dump_to_flash()](https://github.com/espressif/esp-idf/blob/166c30e7b2ed1dcaae56179329540a862915208a/components/espcoredump/src/core_dump_flash.c#L342)
==>[esp_core_dump_write()](https://github.com/espressif/esp-idf/blob/f65c8249af109de349650b9cf79ae28399261750/components/espcoredump/src/core_dump_common.c#L146)
===>[esp_core_dump_write_elf()](https://github.com/espressif/esp-idf/blob/f65c8249af109de349650b9cf79ae28399261750/components/espcoredump/src/core_dump_elf.c#L569)
====>[esp_core_dump_do_write_elf_pass()](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/espcoredump/src/core_dump_elf.c#L539)
=====>[elf_write_core_dump_user_data()](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/espcoredump/src/core_dump_elf.c#L476)
======>elf_add_segment()? maybe even [esp_core_dump_flash_write_data?](https://github.com/espressif/esp-idf/blob/166c30e7b2ed1dcaae56179329540a862915208a/components/espcoredump/src/core_dump_flash.c#L101)

Though curiously, if I force a crash by inserting code like this just before the call to esp_core_dump_do_write_elf_pass
`*(int *)0 = 0; `

A StoreProhibited exception is triggered and then the second panic handler correctly displays
`Re-entered core dump! Exception happened during core dump!`
(Though isn't a crash during the panic handler meant to trigger a [DoubleException](https://github.com/espressif/esp-idf/issues/3810#issuecomment-514016584)?)

[image]

The app Still halts if that crashing code is inserted after the call to esp_core_dump_do_write_elf_pass

I also tried updating to ESP-IDF 4.3.1 and this fixed the problem
Though I cant tell if it fixed the underlying problem during a core-dump as I can no longer trigger the initial exception

I couldn't find any changes to the coredump panic handler in 4.3.1 [here](https://github.com/espressif/esp-idf/compare/v4.3...v4.3.1#files_bucket)

If it helps I'm also using Encrypted flash and making a release build

david-cermak, 2021-12-15:
Hi @cjstott94 

Just a note about the former exception. Have you correctly deinitialized the PPP component as indicated here:

https://github.com/espressif/esp-idf/blob/233dc30fb1a376d7ca0c5d74bdd410ca368f6bf7/examples/protocols/pppos_client/main/pppos_client_main.c#L324-L326

after the OTA update?

Just assume that the receive callback could be called normally (maybe after entering `esp_restart()` function) before performing a restart.

cjstott94, 2021-12-20:
Yes that was the problem, not de-initializing properly
Fixed that issue already, just concerned whether a similar error could cause the processor to lock up again
If it locks up then there's no chance we'll be able to do any OTA bug fixes

brunohpg, 2022-12-06:
Check also (https://github.com/espressif/esp-idf/issues/10110).