espressif/esp-idf #8038: ESP32S3: build errors when including rtc_wdt.h

boarchuz, 2021-12-08:
## Environment

- Module or chip used:  ESP32S3
- IDF version: v5.0-dev-621-g4143a3f572

## Problem Description

RTC_WDT_STG_SEL_x are defined for all targets except ESP32S3 in their respective rtc_cntl_reg.h.
ESP32S3 instead defines analogous RWDT_LL_STG_SEL_x in rwdt_ll.h

Including "soc/rtc_wdt.h" (which needs RTC_WDT_STG_SEL_x) results in an error when compiling for ESP32S3.

### Code to reproduce this issue

```cpp
#include "soc/rtc_wdt.h"

void app_main(void){}
```

## Debug Logs

```
In file included from ../main/main.c:1:
/home/matt/espressif/esp-idf/components/esp_hw_support/include/soc/rtc_wdt.h:63:43: error: 'RTC_WDT_STG_SEL_OFF' undeclared here (not in a function); did you mean 'RTC_WDT_STAGE3'?
     RTC_WDT_STAGE_ACTION_OFF            = RTC_WDT_STG_SEL_OFF,          /*!< Disabled. This stage will have no effects on the system. */
                                           ^~~~~~~~~~~~~~~~~~~
                                           RTC_WDT_STAGE3
In file included from ../main/main.c:1:
/home/matt/espressif/esp-idf/components/esp_hw_support/include/soc/rtc_wdt.h:64:43: error: 'RTC_WDT_STG_SEL_INT' undeclared here (not in a function); did you mean 'RTC_WDT_STAGE3'?
     RTC_WDT_STAGE_ACTION_INTERRUPT      = RTC_WDT_STG_SEL_INT,          /*!< Trigger an interrupt. When the stage expires an interrupt is triggered. */
                                           ^~~~~~~~~~~~~~~~~~~
                                           RTC_WDT_STAGE3
/home/matt/espressif/esp-idf/components/esp_hw_support/include/soc/rtc_wdt.h:65:43: error: 'RTC_WDT_STG_SEL_RESET_CPU' undeclared here (not in a function); did you mean 'RTC_CNTL_SW_SYS_RST_M'?
     RTC_WDT_STAGE_ACTION_RESET_CPU      = RTC_WDT_STG_SEL_RESET_CPU,    /*!< Reset a CPU core. */
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
                                           RTC_CNTL_SW_SYS_RST_M
```

braveheart-cy, 2022-01-20:
Hi @zikalino, 

We seem having a similar issue while trying to run the OTA with the AWS IoT referring to this link.
https://github.com/AdityaHPatwardhan/esp-aws-iot/tree/feature/test_pre_provisioning/examples/ota/ota_http

Environment
・Module used: ESP32S3
・IDF version: v4.4
 
Probrem:
When compiling with set-target on esp32s3, we got the error.

The problem seems to be caused by RTC_WDT_STG_SEL_xx not being defined in esp_hw_support\include\soc\rtc_wdt.h
Refer to the attached error details.  
[error.txt](https://github.com/espressif/esp-idf/files/7902007/error.txt)

Could you please help to take a look into this?

martinberlin, 2022-04-07:
Hello @zikalino
Is there any update on this one?  For now I will simply try to avoid the wdt reset watchdog timers since I'm using this only sporadically when sending large chunks of SPI data but it will be nice to have it repaired.

igrr, 2022-04-07:
@martinberlin If you don't need the `rtc_wdt.h` header specifically, but you need to use RTC watchdog, you can try to use components/hal/include/hal/wdt_hal.h instead (`hal/wdt_hal.h`). Just like `rtc_wdt.h`, this is a "private" header, which can change in the future in incompatible ways. But it is supported on all chip targets.

You can find the usage example in `esp_restart_noos` function in components/esp_system/port/soc/esp32s3/system_internal.c.

Unfortunately for now we don't have a stable public API for the RTC watchdog.

ESP-Marius, 2024-08-15:
As already mentioned, this API was never supported on chips later than S2, the code base on master now correctly reflects this, so you wont get this kind of errors anymore at least.

closed 2024-08-15 by ESP-Marius