espressif/esp-idf #8039: 【ESP32-C3/ADV_DIRECT_IND】BT_BTM: state requested not supported: 5 error generation when using the ADV_TYPE_DIRECT_IND_LOW advertising type

xiaolongba, 2021-12-08:
Hi Espressif Experts,

Recently, I ran into one issue when using the `ADV_TYPE_DIRECT_IND_LOW `advertising type, the error info is as below:

`E (502) BT_BTM: state requested not supported: 5` and `error code is 13`

as we all know, the ADV_TYPE_DIRECT_IND PDU has only the advertiser's address and target's address, so I don't need to configure the adv_data and set the adv parameters only as below and then called the `esp_ble_gap_start_advertising(&gs_m_ble_adv_params)`:

```c
static esp_ble_adv_params_t gs_m_ble_adv_params = {
    .adv_int_min = ADV_INTERVAL,         //100ms
    .adv_int_max = ADV_INTERVAL,        //100ms
    .adv_type = ADV_TYPE_DIRECT_IND_LOW,
    .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .peer_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .peer_addr = {0xE4,0x05,0xF8,0x0D,0xB3,0x9C},
    .channel_map = ADV_CHNL_ALL,
    .adv_filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};
```

however, the event call function shows me the error is 13, as below:
```c
  case ESP_GAP_BLE_ADV_START_COMPLETE_EVT:
    if ((err = param->adv_start_cmpl.status) != ESP_BT_STATUS_SUCCESS)
    {
      ESP_LOGE(DEMO_TAG, "Adv start failed: %s %d", esp_err_to_name(err),err);
    }
    break;
```
[image]

also, I found a similar issue #2469 here, as @Weijian-Espressif said, it works fine with the latest master branch, but it still cannot work on my side even if using the latest mater branch. for this issue, could you give me a hand?

there is also a bug for `ADV_TYPE_DIRECT_IND_HIGH` advertising type, it is always advertising when using the  `ADV_TYPE_DIRECT_IND_HIGH` advertising type, but based on the spec described, as below:

> In high duty cycle connectable directed advertising mode, the time between the start of two consecutive ADV_DIRECT_IND PDUs sent on the same advertising channel index shall be less than or equal to 3.75 ms. The Link Layer shall exit the Advertising state no later than 1.28 s after the Advertising state was entered

so my questions are as below:
1. could you help me double-check, if it is possible to implement the ADV_TYPE_DIRECT_IND_LOW advertising at this moment?
2. regarding the ADV_TYPE_DIRECT_IND_HIGH advertising type, could you double-check also. it shall exit the Advertising state no later than 1.28 s, but the actual situation, it is always running when using ADV_TYPE_DIRECT_IND_HIGH advertising type at this moment.

thanks in advance. I am waiting for your news.

xiaolongba, 2021-12-10:
Hello,
any updates about it?

xiewenxiang, 2021-12-13:
@xiaolongba 

The problem has reappeared, we will fix in as soon as possible, and give you a reply.

xiaolongba, 2021-12-13:
@xiewenxiang 

big thanks bro. what you mean is these two issues I mentioned above, or one of them? I reported two bugs about ADV_DIRECT_IND here.

xiewenxiang, 2021-12-13:
@xiaolongba 

The fix I mentioned is for problem 1:
```
could you help me double-check, if it is possible to implement the ADV_TYPE_DIRECT_IND_LOW advertising at this moment?
```

Problem 2 is that every 1.28 s, the host starts broadcasting again, you can add debug log in  `btm_ble_gap.c`  => `void BTM_recovery_Pre_State(void)`.

xiaolongba, 2021-12-13:
@xiewenxiang 
clear for item1.

> Problem 2 is that every 1.28 s, the host starts broadcasting again, you can add debug log in `btm_ble_gap.c` => `void BTM_recovery_Pre_State(void)`.

got it. if it is possible to disable it?

xiewenxiang, 2021-12-13:
@xiaolongba 

Question 2, we need to discuss whether we need to remove the logic of automatically starting the high duty cycle connectable directed adv.

We'll get back to you as soon as possible.

xiaolongba, 2021-12-13:
@xiewenxiang 

Perfect, I will wait for your good news. In my past experience, other competitors have not this behavior, and it will be stopped after 1.28s

For your information, of course u can check with other platforms

xiaolongba, 2021-12-23:
@xiewenxiang 

Hello, Bro. any updates about it?

thanks.

xiewenxiang, 2021-12-31:
@xiaolongba 

https://github.com/espressif/esp-idf/tree/0030fba89c81436f4f88a2bfe3efc903985da06f

xiaolongba, 2021-12-31:
@xiewenxiang 

Hello, Bro.

Big thanks. let me update the SDK and try again. you are a good man, pls close this ticket and Happy New Year!

closed 2021-12-31 by xiaolongba