espressif/esp-idf #8040: 一种无法连接到WIFI的情况（A condition that ESP32 can't connect to WIFI）

Augtons, 2021-12-08:
## Environment

- Module or chip used:  `ESP32-WROOM-32`
- IDF version :     `v4.3.1`
- Build System:         `idf.py`
- Compiler version:  `8.4.0`

- Operating System:     `Linux (Manjaro, to be exact)`
- Using an IDE?: `Yes, 'Clion' had been used, but it just be used as an editor.`
- Power Supply:         `external 3.3V`

## Problem Description

I can't connect to wifi (get the IP address) if I call function `esp_event_loop_create_default();` after `esp_netif_create_default_wifi_sta();`

But after I switch their order, everything will be fine.

> 如果函数`esp_event_loop_create_default();`被放在`esp_netif_create_default_wifi_sta();`之后，那么Wifi不能连接成功（无法获取IP地址）
>
> 但是如果我交换他们的顺序，则一切正常

For example
> 例如

Can't connect to Wifi:
```cpp
    esp_netif_create_default_wifi_sta();
    esp_event_loop_create_default();
```
-----
Everything will be fine:
```cpp
    esp_event_loop_create_default();
    esp_netif_create_default_wifi_sta();
```

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "driver/uart.h"
#include "freertos/queue.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "string.h"
#include "driver/gpio.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_event.h"
#include "esp_netif_ip_addr.h"
#include "esp_http_client.h"
#include "freertos/event_groups.h"
#include "esp_log.h"

#define CONNECTED_BIT   BIT(0)
#define DEBUG       printf("断点：line:%d\nfile:%s\n\n", __LINE__, __FILE__)

//QueueHandle_t uart_queue;
esp_event_handler_instance_t eventHandlerInstance;
esp_event_handler_instance_t ipeventHandlerInstance;
TaskHandle_t http_task_handle;
esp_http_client_config_t httpconfig = {0};
esp_http_client_handle_t httpHandle;
EventGroupHandle_t eventGroupHandle;

esp_netif_t *netif;

void wifievent(void* event_handler_arg, esp_event_base_t event_base, int32_t event_id, void* event_data){
    if(event_base == WIFI_EVENT){
        if(event_id == WIFI_EVENT_STA_START){
            esp_wifi_connect();
        }else if(event_id == WIFI_EVENT_STA_DISCONNECTED){
            printf("wifi连接失败或连接已断开\n");
            //if (http_task_handle != NULL){
            xEventGroupClearBits(eventGroupHandle, CONNECTED_BIT);
            //}
            esp_wifi_connect();
        }else if(event_id == WIFI_EVENT_STA_CONNECTED){

        }
    }else if(event_base == IP_EVENT){
        if(event_id == IP_EVENT_STA_GOT_IP){
            ip_event_got_ip_t *ip_event_data = event_data;
            printf("Got IP:" IPSTR, IP2STR(&ip_event_data->ip_info.ip));
            xEventGroupSetBits(eventGroupHandle, CONNECTED_BIT);
            if (http_task_handle == NULL){
                //xTaskCreate(http_task, "http", 2048, NULL, 5, &http_task_handle);
            }
        }
    }
}

void app_main(void) {
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND){
        nvs_flash_erase();
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    eventGroupHandle = xEventGroupCreate();
    esp_netif_init();

/*** Where the problem arises 问题出现的地方***/
    esp_netif_create_default_wifi_sta();
    esp_event_loop_create_default();

/* the third parameter 'wifievent' is an callback function name, Their is no errors*/
   esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, wifievent, NULL, &eventHandlerInstance);
/* the third parameter 'wifievent' is an callback function name*/
   esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, wifievent, NULL, &ipeventHandlerInstance);

    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&wifi_init_config));

/*"HelloWorld"is my wifi ssid*/
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "HelloWorld",
            .password = "password",
        }
    };
    esp_wifi_set_config(WIFI_IF_STA, &wifi_config);
    esp_wifi_set_mode(WIFI_MODE_STA);
    esp_wifi_start();
}
```

## Debug Logs

```
I (880) wifi:state: assoc -> run (10)
I (980) wifi:connected with HelloWorld, aid = 1, channel 1, 40U, bssid = 8c:de:f9:a9:e7:ab
I (980) wifi:security: WPA2-PSK, phy: bgn, rssi: -42
I (990) wifi:pm start, type: 1

I (1060) wifi:AP's beacon interval = 102400 us, DTIM period = 1

```

I didn't get the IP. In fact, my router administration page shows that I'm not connected to Wifi at all
> 我并没有获取到IP地址。实际上，我的路由器管理页面显示根本没有连接到Wifi

But after I switch their order, everything will be fine.

> 但是如果我交换他们的顺序，则一切正常

This is the expected output （in the second case）【Got IP address: 192.168.31.179】
> 这是预期输出（第二种情况）【获取到IP地址: 192.168.31.179】

```
I (869) wifi:state: assoc -> run (10)
I (979) wifi:connected with HelloWorld, aid = 1, channel 1, 40U, bssid = 8c:de:f9:a9:e7:ab
I (979) wifi:security: WPA2-PSK, phy: bgn, rssi: -40
I (979) wifi:pm start, type: 1

W (999) wifi:<ba-add>idx:0 (ifx:0, 8c:de:f9:a9:e7:ab), tid:6, ssn:0, winSize:64
I (1079) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2059) esp_netif_handlers: sta ip: 192.168.31.179, mask: 255.255.255.0, gw: 192.168.31.1

```

david-cermak, 2021-12-09:
Hi @Augtons 

What is the reason to initialise the system loop *after* creating the wifi network interface? This wouldn't work as we have to register some actions to the default (system) loop using `esp_event_handler_register()` when creating the wifi interface, so we rely on having the system loop already up and running.

Augtons, 2021-12-10:
> 

Thank you very much for your reply. I did make a mistake.

closed 2021-12-10 by Augtons