espressif/esp-idf #8043: CMake error when doing idf.py set-target esp32

RenyV, 2021-12-08:
I downloaded the  Master IDF from github and installed in Ubuntu.
And then downloaded ESP32 AT project from github.

When I tried to run "idf.py set-target esp32"   ... I get the following error...

Vostro-3581:~/esp/esp-at$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/philo/esp/esp-at/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/philo/esp/esp-at/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/philo/esp/esp-at"...
CMake Error at CMakeLists.txt:9 (string):
  string no output variable specified

CMake Error at CMakeLists.txt:13 (string):
  string begin index: 9 is out of range 0 - 0

CMake Error at CMakeLists.txt:14 (string):
  string sub-command STRIP requires two arguments.

CMake Error at CMakeLists.txt:15 (string):
  string no output variable specified

silence:0 sdkconfig:/home/philo/esp/esp-at/module_config/module__default/sdkconfig.defaults
CMake Error at CMakeLists.txt:53 (include):
  include could not find load file:

    /home/philo/esp/esp-at/esp-idf/tools/cmake/project.cmake

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:80 (add_dependencies):
  Cannot add target-level dependencies to non-existent target "flash".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.

-- Configuring incomplete, errors occurred!
See also "/home/philo/esp/esp-at/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

xcguang, 2021-12-09:
Please use `build.py` in the AT project to compile the firmware, the detailed steps: https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html

RenyV, 2021-12-09:
Ok, Thanks. But when I run 
"./build.py menuconfig"
A new window opens and closes immediately.

https://user-images.githubusercontent.com/41112579/145361632-fa88754f-4b44-4b46-a35d-903a23bd19f2.mp4

IDF. v- 4.3ESP-IDF v4.3.1
Esp-at v -2.2.0.0

xcguang, 2021-12-09:
Could you try `python build.py menuconfig`?

RenyV, 2021-12-09:
PS C:\Users\aaere\esp\esp-at>  python build.py menuconfig
Traceback (most recent call last):
  File "build.py", line 25, in <module>
    import xlrd
ModuleNotFoundError: No module named 'xlrd'
PS C:\Users\aaere\esp\esp-at>

xcguang, 2021-12-09:
`python -m pip install pyyaml xlrd`

BTW, you can remove esp-idf and  build directory, and it will clone the correct esp-idf.

RenyV, 2021-12-09:
Yes I got this.
Is this ok?

[image]

RenyV, 2021-12-09:
Installed: python -m pip install pyyaml xlrd

I get this.
[image]

RenyV, 2021-12-09:
Then I run python ./build.py menuconfig

And it fails.. Please help.

PS C:\Users\aaere\esp\esp-at>  python build.py menuconfig
Platform name:
1. PLATFORM_ESP32
2. PLATFORM_ESP32C3
choose(range[1,2]):1

Module name:
1. WROOM-32
2. WROVER-32
3. PICO-D4
4. SOLO-1
5. MINI-1 (description: ESP32-U4WDH chip inside)
6. ESP32-D2WD (description: 2MB flash, No OTA)
7. ESP32_QCLOUD (description: QCLOUD TX:17 RX:16)
choose(range[1,7]):4

Enable silence mode to remove some logs and reduce the firmware size?
0. No
1. Yes
choose(range[0,1]):0
platform_name=ESP32,module_name=SOLO-1
Please wait for the SDK download to finish...
Cloning into 'esp-idf'...
remote: Enumerating objects: 289933, done.
remote: Total 289933 (delta 0), reused 0 (delta 0), pack-reused 289933
Receiving objects: 100% (289933/289933), 165.13 MiB | 7.59 MiB/s, done.
Resolving deltas: 100% (214286/214286), done.
Updating files: 100% (6722/6722), done.
old commit:37634f09b2e46901c46ec3dff663e7d076c71b69
checkout commit:5595042c16d989c8e44f999eb30ce0e6377fd86e
Please wait for the update to complete, which will take some time
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
Note: switching to '5595042c16d989c8e44f999eb30ce0e6377fd86e'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 5595042c16 Merge branch 'feature/windows_installer_moved_to_github_backport_4.2' into 'release/v4.2'
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib'
Submodule 'components/bt/host/nimble/nimble' (https://github.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
[... 65 lines cut ...]
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'c3d3a69021cfec3236ca2c0b63be4048ec6643a4'
Update completed
WARNING: IDF_PATH environment variable is set to c:\users\aaere\esp\esp-idf but esp-idf\tools\idf.py path indicates IDF directory c:\users\aaere\esp\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...
Traceback (most recent call last):
  File "esp-idf\tools\idf.py", line 794, in <module>
    main()
  File "esp-idf\tools\idf.py", line 711, in main
    cli = init_cli(verbose_output=checks_output)
  File "esp-idf\tools\idf.py", line 669, in init_cli
    extensions.append((name, import_module(name)))
  File "importlib\__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "c:\users\aaere\esp\esp-idf\tools\idf_py_actions\core_ext.py", line 8, in <module>
    from idf_py_actions.constants import GENERATORS, PREVIEW_TARGETS, SUPPORTED_TARGETS
ImportError: cannot import name 'PREVIEW_TARGETS' from 'idf_py_actions.constants' (C:\Users\aaere\esp\esp-at\esp-idf\tools\idf_py_actions\constants.py)
idf.py build ret: 1
idf.py build failed
PS C:\Users\aaere\esp\esp-at>

xcguang, 2021-12-09:
> WARNING: IDF_PATH environment variable is set to c:\users\aaere\esp\esp-idf but esp-idf\tools\idf.py path indicates IDF directory c:\users\aaere\esp\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...

[image]

You can set IDF_PATH to c:\users\aaere\esp\esp-at\esp-idf, and try again

xcguang, 2021-12-09:
Can you compile hello-world in the esp-idf/example successfully?

RenyV, 2021-12-09:
ESP-IDF v4.2.2-331-g5595042c16
PS C:\Users\aaere\esp\esp-idf\examples\get-started\hello_world> $IDF_PATH
C:\Users\aaere\esp\esp-idf
PS C:\Users\aaere\esp\esp-idf\examples\get-started\hello_world> idf.py set-target esp32
WARNING: IDF_PATH environment variable is set to c:\users\aaere\esp\esp-at\esp-idf but idf.py path indicates IDF directory c:\users\aaere\esp\esp-idf. Using the environment variable directory, but results may be unexpected...
The following Python requirements are not satisfied:
cryptography>=2.1.4,<35
To install the missing packages, please run "C:\Users\aaere\esp\esp-at\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\aaere\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: C:\Users\aaere\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe

ESP-IDF v4.2.2-331-g5595042c16
PS C:\Users\aaere\esp\esp-idf\examples\get-started\hello_world> $IDF_PATH
C:\Users\aaere\esp\esp-idf
PS C:\Users\aaere\esp\esp-idf\examples\get-started\hello_world> ${IDF_PATH} = "C:\Users\aaere\esp\esp_at\esp-idf"
PS C:\Users\aaere\esp\esp-idf\examples\get-started\hello_world> $IDF_PATH
C:\Users\aaere\esp\esp_at\esp-idf
PS C:\Users\aaere\esp\esp-idf\examples\get-started\hello_world> cd ..
PS C:\Users\aaere\esp\esp-idf\examples\get-started> cd ..
PS C:\Users\aaere\esp\esp-idf\examples> cd..
PS C:\Users\aaere\esp\esp-idf> cd ..
PS C:\Users\aaere\esp> cd .\esp-at\
PS C:\Users\aaere\esp\esp-at> cd .\esp-idf\
PS C:\Users\aaere\esp\esp-at\esp-idf> .\install.bat
Installing ESP-IDF tools
Selected targets are: esp32s2, esp32, esp32s3
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20210902 (already installed)
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
ERROR: This script was called from a virtual environment, can not create a virtual environment again
PS C:\Users\aaere\esp\esp-at\esp-idf>

RenyV, 2021-12-09:
PS C:\Users\aaere\esp\esp-at> $IDF_PATH
C:\Users\aaere\esp\esp-at\esp-idf
PS C:\Users\aaere\esp\esp-at> ${IDF_PATH} = "C:\Users\aaere\esp\esp-at\esp-idf"
PS C:\Users\aaere\esp\esp-at> $IDF_PATH
C:\Users\aaere\esp\esp-at\esp-idf
PS C:\Users\aaere\esp\esp-at> python build.py menuconfig
module_name SOLO-1
platform_name=ESP32,module_name=SOLO-1
WARNING: IDF_PATH environment variable is set to c:\users\aaere\esp\esp-idf but esp-idf\tools\idf.py path indicates IDF directory c:\users\aaere\esp\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...
Traceback (most recent call last):
  File "esp-idf\tools\idf.py", line 794, in <module>
    main()
  File "esp-idf\tools\idf.py", line 711, in main
    cli = init_cli(verbose_output=checks_output)
  File "esp-idf\tools\idf.py", line 669, in init_cli
    extensions.append((name, import_module(name)))
  File "importlib\__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "c:\users\aaere\esp\esp-idf\tools\idf_py_actions\core_ext.py", line 8, in <module>
    from idf_py_actions.constants import GENERATORS, PREVIEW_TARGETS, SUPPORTED_TARGETS
ImportError: cannot import name 'PREVIEW_TARGETS' from 'idf_py_actions.constants' (C:\Users\aaere\esp\esp-at\esp-idf\tools\idf_py_actions\constants.py)
idf.py build ret: 1
idf.py build failed

RenyV, 2021-12-09:
> Can you compile hello-world in the esp-idf/example successfully?

I was working fine  before I changed the path. to ${IDF_PATH} = "C:\Users\aaere\esp\esp-at\esp-idf"

Now I get this....

PS C:\Users\aaere\esp\esp-at\esp-idf\examples\get-started\hello_world> idf.py
WARNING: IDF_PATH environment variable is set to c:\users\aaere\esp\esp-idf but C:\Users\aaere\esp\esp-at\esp-idf\tools\idf.py path indicates IDF directory c:\users\aaere\esp\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...
Traceback (most recent call last):
  File "C:\Users\aaere\esp\esp-at\esp-idf\tools\idf.py", line 794, in <module>
    main()
  File "C:\Users\aaere\esp\esp-at\esp-idf\tools\idf.py", line 711, in main
    cli = init_cli(verbose_output=checks_output)
  File "C:\Users\aaere\esp\esp-at\esp-idf\tools\idf.py", line 669, in init_cli
    extensions.append((name, import_module(name)))
  File "importlib\__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "c:\users\aaere\esp\esp-idf\tools\idf_py_actions\core_ext.py", line 8, in <module>
    from idf_py_actions.constants import GENERATORS, PREVIEW_TARGETS, SUPPORTED_TARGETS
ImportError: cannot import name 'PREVIEW_TARGETS' from 'idf_py_actions.constants' (C:\Users\aaere\esp\esp-at\esp-idf\tools\idf_py_actions\constants.py)
PS C:\Users\aaere\esp\esp-at\esp-idf\examples\get-started\hello_world>

Alson-tang, 2021-12-09:
1. Run **ESP-IDF 4.3 CMD**, then switch to the root directory of AT
2. Delete the build directory and sdkconfig file
3. Execute build.py menuconfig
4. Enter the esp-idf directory after executing the command
5. Execute install.bat
6. Execute export.bat
7. Return AT root directory and then build.py build

RenyV, 2021-12-09:
ok i try now.

RenyV, 2021-12-09:
> 1. Run **ESP-IDF 4.3 CMD**, then switch to the root directory of AT
-Deleted / uninstalled all IDF installation. 
-Installed a new ESP-IDF 4.3 to path "C:\Users\aaere\esp\"
- Then I did 
         cd %userprofile%\esp
         git clone --recursive https://github.com/espressif/esp-at.git
- switch to esp-at directory  " cd esp-at"

> 2. Delete the build directory and sdkconfig file
since this a fresh installation, no build directory and sdkconfig file

> 3. Execute build.py menuconfig
- in esp/esp-at folder I run "build.py menuconfig"
- a new dialog opens, I configured for esp32, solo-1. and after some time the new dialog closed automatically.

> 4. Enter the esp-idf directory after executing the command
- entered the esp-idf directory using the command "cd c://users/aaere/esp/esp-at/esp-idf"
> 5. Execute install.bat
- excuted "install.bat" inside /esp/esp-at/esp-idf folder.
C:\Users\aaere\esp\esp-at\esp-idf>install.bat
Installing ESP-IDF tools
WARNING: File C:\Users\aaere\.espressif\idf-env.json was not found.
Creating C:\Users\aaere\.espressif\idf-env.json
Selected targets are: esp32, esp32s3, esp32s2
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Installing xtensa-esp32-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\aaere\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
Done
Extracting C:\Users\aaere\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\aaere\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0
Installing xtensa-esp32s2-elf@esp-2020r3-8.4.0
Downloading xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\aaere\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
Done
Extracting C:\Users\aaere\.espressif\dist\xtensa-esp32s2-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\Users\aaere\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Installing openocd-esp32@v0.10.0-esp32-20210902
Downloading openocd-esp32-win32-0.10.0-esp32-20210902.zip to C:\Users\aaere\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20210902.zip.tmp
Done
Extracting C:\Users\aaere\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20210902.zip to C:\Users\aaere\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902
Installing ninja@1.10.0
Downloading ninja-1.10.0-win64.zip to C:\Users\aaere\.espressif\dist\ninja-1.10.0-win64.zip.tmp
Done
Extracting C:\Users\aaere\.espressif\dist\ninja-1.10.0-win64.zip to C:\Users\aaere\.espressif\tools\ninja\1.10.0
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
ERROR: This script was called from a virtual environment, can not create a virtual environment again
> 6. Execute export.bat
> 7. Return AT root directory and then build.py build

RenyV, 2021-12-09:
PS C:\Users\aaere\esp\esp-at\esp-idf> .\export.bat
Setting IDF_PATH: C:\Users\aaere\esp\esp-at\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2021r1-8.4.0.
Not using an unsupported version of tool xtensa-esp32s2-elf found in PATH: esp-2021r1-8.4.0.
Not using an unsupported version of tool openocd-esp32 found in PATH: v0.10.0-esp32-20210401.
Not using an unsupported version of tool ninja found in PATH: 1.10.2.
    C:\Users\aaere\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin
    C:\Users\aaere\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\aaere\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin
    C:\Users\aaere\.espressif\tools\ninja\1.10.0\
    C:\Users\aaere\esp\esp-at\esp-idf\tools

Checking if Python packages are up to date...
The following Python requirements are not satisfied:
cryptography>=2.1.4,<35
To install the missing packages, please run "C:\Users\aaere\esp\esp-at\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\aaere\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: C:\Users\aaere\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
PS C:\Users\aaere\esp\esp-at\esp-idf>

RenyV, 2021-12-09:
C:\Users\aaere\esp\esp-at\esp-idf>export.bat
Setting IDF_PATH: C:\Users\aaere\esp\esp-at\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2021r1-8.4.0.
Not using an unsupported version of tool xtensa-esp32s2-elf found in PATH: esp-2021r1-8.4.0.
Not using an unsupported version of tool openocd-esp32 found in PATH: v0.10.0-esp32-20210401.
Not using an unsupported version of tool ninja found in PATH: 1.10.2.
    C:\Users\aaere\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin
    C:\Users\aaere\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\aaere\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin
    C:\Users\aaere\.espressif\tools\ninja\1.10.0\
    C:\Users\aaere\esp\esp-at\esp-idf\tools

Checking if Python packages are up to date...
The following Python requirements are not satisfied:
cryptography>=2.1.4,<35
To install the missing packages, please run "C:\Users\aaere\esp\esp-at\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\aaere\.espressif\python_env\idf4.3_py3.8_env
    Python interpreter used: C:\Users\aaere\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe

C:\Users\aaere\esp\esp-at\esp-idf>

Alson-tang, 2021-12-09:
I have a question ? Do you want to compile ESP32 AT module ? Which version do you want to compile ？

RenyV, 2021-12-09:
esp-at ver 2.2.0.0

RenyV, 2021-12-09:
> I have a question ? Do you want to compile ESP32 AT module ? Which version do you want to compile ？

ESP-AT v2.2.0.0

Alson-tang, 2021-12-09:
ESP32 v2.2.0.0 is built on IDF v4.2 version. So I suggest you run ESP-IDF 4.2 CMD

RenyV, 2021-12-09:
ok
when I tried the install ESP-IDF release/4.2 cmd, installation fails.
Shall I install IDF version 4.2.2

Alson-tang, 2021-12-09:
You can install 4.2 release version

RenyV, 2021-12-09:
> You can install 4.2 release version

Windows installer fails

Alson-tang, 2021-12-09:
Could you show me the error picture ?

RenyV, 2021-12-09:
Steps: what i did...
run "esp-idf-tools-setup-2.3.exe" to install 
- IDF in path "C:\Users\aaere\esp\esp-idf"
- tools in path "C:\Users\aaere\.espressif"

[image]
[image]
[image]
[image]

RenyV, 2021-12-09:
An then Installation went fine and I got CMD and Windows Shell for IDF

So opened the CMD prompt and typed this...

C:\Users\aaere\esp\esp-idf>idf.py --version
fatal: bad object HEAD
fatal: 'git status --porcelain=2' failed in submodule third-party/mruby
fatal: 'git status --porcelain=2' failed in submodule components/nghttp/nghttp2
WARNING: Git version unavailable, reading from source
ESP-IDF v4.2.2

C:\Users\aaere\esp\esp-idf>

I was expecting to get 4.2, but i got 4.2.2.

Alson-tang, 2021-12-10:
It should be no big problem, you can walk again in the way I gave you

RenyV, 2021-12-10:
Still Its not working for me...

Let me narrate the steps what I did...

1. Installed IDF version 4.2
please see the screen shot
[image]

2. cloned the esp-at project version 2.2.0.0 using the command...
"C:\Users\aaere\esp>git clone --recursive --branch v2.2.0.0_esp32 https://github.com/espressif/esp-at.git"
please see the screen shot
[image]

3. I entered to esp/esp-at folder and did "build.py menuconfig"..... So then a new cmd prompt window opens...
please see the screen shot
[image]

4. After entered the following details in new cmd prompt... the dialog clones "esp_idf" into esp/esp_at folder
  The details... "ESP32, SOLO1, No Silence"

5. As I get no warnings or error..... I am going to do build.py menuconfig again in esp/eap_at
and I am expecting the following dialog..
[image]

6 But What i get in reality is .... the following video.. please see..

https://user-images.githubusercontent.com/41112579/145526071-fe85f5e5-3110-4a8a-a541-5cda622beb31.mp4

RenyV, 2021-12-10:
@xcguang @Alson-tang  Any help?

Alson-tang, 2021-12-10:
After executing the fourth step, there will be an esp-idf directory in the root directory of AT.

After entering the esp-idf directory, execute install.sh bat and thus execute export.bat

Go back to the AT root directory and continue to execute build.py menuconfig.

RenyV, 2021-12-10:
C:\Users\aaere\esp\esp-at>cd esp-idf

C:\Users\aaere\esp\esp-at\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util
Skipping xtensa-esp32-elf@esp-2020r3-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r3-8.4.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping cmake@3.16.4 (already installed)
Installing openocd-esp32@v0.10.0-esp32-20200709
Downloading openocd-esp32-win32-0.10.0-esp32-20200709.zip to C:\Users\aaere\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20200709.zip.tmp
Done
Extracting C:\Users\aaere\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20200709.zip to C:\Users\aaere\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709
Skipping ninja@1.10.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Skipping dfu-util@0.9 (already installed)
Setting up Python environment
Installing Python packages from C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt
Processing c:\users\aaere\esp\esp-at\esp-idf\tools\kconfig_new\esp-windows-curses
  Preparing metadata (setup.py) ... done
Requirement already satisfied: setuptools>=21 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 4)) (51.3.3)
Requirement already satisfied: click>=5.0 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 8)) (8.0.3)
Requirement already satisfied: pyserial>=3.0 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 9)) (3.5)
Requirement already satisfied: future>=0.15.2 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 11)) (3.4.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 13)) (0.27)
Requirement already satisfied: gdbgui==0.13.2.0 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 15)) (0.13.2.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 18)) (0.9.0.2)
Requirement already satisfied: python-socketio<5 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 21)) (4.6.1)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 24)) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 25)) (3.1.9)
Requirement already satisfied: ecdsa>=0.16.0 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from -r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 26)) (0.17.0)
Requirement already satisfied: Pygments<3.0,>=2.2.0 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 15)) (2.10.0)
Requirement already satisfied: Flask<1.0,>=0.12.2 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 15)) (0.12.5)
Requirement already satisfied: Flask-Compress<2.0,>=1.4.0 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 15)) (1.10.1)
Requirement already satisfied: gevent<2.0,>=1.2.2 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 15)) (1.5.0)
Requirement already satisfied: Flask-SocketIO<3.0,>=2.9 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from gdbgui==0.13.2.0->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 15)) (2.9.6)
Requirement already satisfied: colorama in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from click>=5.0->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 8)) (0.4.4)
Requirement already satisfied: cffi>=1.12 in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt (line 11)) (1.15.0)
[... cut ...]
Python requirements from C:\Users\aaere\esp\esp-at\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\aaere\esp\esp-at\esp-idf>
C:\Users\aaere\esp\esp-at\esp-idf>cd ..

C:\Users\aaere\esp\esp-at>build.py menuconfig

C:\Users\aaere\esp\esp-at>build.py menuconfig

C:\Users\aaere\esp\esp-at>

Now I am getting the same result.  

Please see the video..

https://user-images.githubusercontent.com/41112579/145586436-29696f8e-52ab-4c68-8d7b-5482f366372c.mp4

xcguang, 2021-12-10:
In current cmd shell, run `python -m pip install pyyaml xlrd` and try again please.

RenyV, 2021-12-10:
C:\Users\aaere\esp\esp-at>python -m pip install pyyaml xlrd
Requirement already satisfied: pyyaml in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (6.0)
Requirement already satisfied: xlrd in c:\users\aaere\.espressif\python_env\idf4.2_py3.8_env\lib\site-packages (2.0.1)

C:\Users\aaere\esp\esp-at>build.py menuconfig

And I am getting same result. Please see the video.

https://user-images.githubusercontent.com/41112579/145589484-8f0e5caf-5b42-44e2-abda-132c50b555cd.mp4

xcguang, 2021-12-10:
`python build.py build` to check the error log please.

RenyV, 2021-12-10:
[python buil.py build.txt](https://github.com/espressif/esp-idf/files/7693510/python.buil.py.build.txt)

Build seems success. can you please check sir.?

RenyV, 2021-12-10:
Now I run 

cmd prompt: esp/esp-at/python build.py menuconfig

I get this.

https://user-images.githubusercontent.com/41112579/145594587-ede4f99c-23c0-4a98-b270-997dd33bbb91.mp4

[image]

RenyV, 2021-12-10:
Thanks @Alson-tang @xcguang

xcguang, 2021-12-10:
Good news

Alvin1Zhang, 2021-12-13:
Good to hear it is working, feel free to reopen.

closed 2021-12-13 by Alvin1Zhang

micronik-ir, 2022-01-18:
hello guys
I'm using Windows for development

I installed the ESP-IDF Tools and choose release/v4.2(release branch) .

I'm attempting to compile esp-at project for the ESP8266.

Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\Hossein\esp\.espressif\python_env\idf4.2_py3.8_env\Scripts
Python 3.8.7
Using Git in C:\Users\Hossein\esp\.espressif\tools\idf-git\2.30.1\cmd\
git version 2.30.1.windows.1
Setting IDF_PATH: C:\Users\Hossein\esp\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\Hossein\esp\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin
    C:\Users\Hossein\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\Hossein\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\Hossein\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\Hossein\esp\.espressif\tools\cmake\3.16.4\bin
    C:\Users\Hossein\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin
    C:\Users\Hossein\esp\.espressif\tools\ninja\1.10.0\
    C:\Users\Hossein\esp\.espressif\tools\idf-exe\1.0.1\
    C:\Users\Hossein\esp\.espressif\tools\ccache\3.7\
    C:\Users\Hossein\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\Hossein\esp\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\Hossein\esp\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

**C:\Users\Hossein\esp\esp-idf>cd ..**

**C:\Users\Hossein\esp>cd esp-at**

**C:\Users\Hossein\esp\esp-at>python build.py menuconfig**
module_name ESP8266_1MB
platform_name=ESP8266,module_name=ESP8266_1MB
WARNING: IDF_PATH environment variable is set to c:\users\hossein\esp\esp-idf but esp-idf\tools\idf.py path indicates IDF directory c:\users\hossein\esp\esp-at\esp-idf. Using the environment variable directory, but results may be unexpected...
Checking Python dependencies...
Python requirements from C:\Users\Hossein\esp\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\hossein\esp\esp-at\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp8266 -DCCACHE_ENABLE=0 c:\users\hossein\esp\esp-at"...
silence:1 sdkconfig:C:/Users/Hossein/esp/esp-at/module_config/module_esp8266_1mb/sdkconfig_silence.defaults
CMake Warning at esp-idf/tools/cmake/idf.cmake:14 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

Check if your project's top-level CMakeLists.txt includes the right
CMake files. Environment IDF_PATH will be used for the build.

Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:49 (include)

-- Found Git: C:/Users/Hossein/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
CMake Error at C:/Users/Hossein/esp/esp-idf/tools/cmake/targets.cmake:57 (message):
  Unable to resolve 'esp8266' for setting toolchain file.
Call Stack (most recent call first):
  esp-idf/tools/cmake/project.cmake:245 (__target_set_toolchain)
  CMakeLists.txt:51 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
idf.py build ret: 2
idf.py build failed

I went this under path , but the install.sh and export.sh files do not run in Windows .
There are no install.bat and export.bat files for esp-idf esp8266 .

**C:\Users\Hossein\esp\esp-at>cd esp-idf**

**C:\Users\Hossein\esp\esp-at\esp-idf>install.bat**
'install.bat' is not recognized as an internal or external command,
operable program or batch file.

**C:\Users\Hossein\esp\esp-at\esp-idf>install.sh**

How do you recommend the use of both the ESP32 IDF and ESP8266 IDF since both use the same ENV variable but the associated paths are different?