espressif/esp-idf #8047: 【ESP32-C3/IO mapping for radio TX and RX】Is it possible to implement the IO mapping for radio tx/rx monitoring?

xiaolongba, 2021-12-09:
**Is your feature request related to a problem? Please describe.**
> actually, It is not really related to a problem, instead, it is a new feature for radio Tx or Rx debugging when doing radio-related cases, such as the multi-connection, etc.

**Describe the solution you'd like**

> I would like to monitor the Radio Tx or Rx with GPIOs. In my last experience, there are some BLE suppliers can support this feature, such as Nordic, ST, Telink, NXP, etc.

**Describe alternatives you've considered**

> could you provide the corresponding APIs for GPIO configuration for Radio Tx or Rx monitoring, or disclose the radio Tx or Rx interrupt and users can implement the GPIO toggling there.

**Additional context**

as heard from others online, Espressif may support this feature, but it is not open for users.

so the requirements can you open for users? 

thanks in advance.

xiewenxiang, 2021-12-14:
@xiaolongba Sorry ，there is no such API for customer to use.

xiaolongba, 2021-12-14:
All right, does espressif have similar ways to monitor the radio status?

xiewenxiang, 2021-12-14:
There are ways to monitor it, but it's not open to the public now.

xiaolongba, 2021-12-14:
@xiewenxiang 

> There are ways to monitor it, but it's not open to the public now.

got it. as you know, other competitors such as ST, Nordic, NXP, etc. can open for users for using. this feature is very useful for radio debugging especially the multi-connection cases.

so could you help me confirm if it can open for users now? if not, pls close this ticket,

thanks a lot.

closed 2021-12-14 by Alvin1Zhang