espressif/esp-idf #8048: ULP wakeup period doubled in deep sleep when ext0 wake source + timer wake source enabled

cdollar393, 2021-12-09:
## Environment

- Development Kit: none
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used: ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-517-g68cf4ef2be
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: macOS
- (Windows only) environment type:
- Using an IDE?: Yes (vscode with IDF extension, but also using idf.py on cli)
- Power Supply: USB + battery

## Problem Description

My scenario is an ESP32 that is running a ULP program every 10 seconds using `ulp_set_wakeup_period()`. Each time the ULP program runs it increments a run counter and saves the RTC ticks counter value for that run in an array. The run counter and array of ticks values are stored in RTC memory to survive deep sleep.

The main IDF program running on the ESP32 starts the ULP program the first time it boots (it does nothing with the ULP when waking from deep sleep). The ESP32 then enters a loop where it looks for new data coming in from the ULP program. When new data is found the entire contents of the stored ULP data array is printed. After 60 seconds in this loop the main ESP32 CPU enters deep sleep, and stays in deep sleep for 60 seconds using `esp_sleep_enable_timer_wakeup()`.

### Expected Behavior

Expect the ULP to 1) increment runs count and 2) save a ticks value every 10 seconds, regardless of whether the main ESP32 CPU is awake or in deep sleep, and regardless of whether an `ext0` wake source is enabled or not.

### Actual Behavior

When the sleep timer is the only deep sleep wakeup source configured in the program, then the program works correctly. However, if an `ext0` wakeup source is configured in addition to the timer source then I observe that while the ESP32 main CPU is awake everything works correctly, but when in deep sleep the ULP wakeup period is doubled. Instead of seeing a ticks value saved every 10 seconds I see one every 20 seconds.

This doubling of the ULP wakeup period is only seen when the `ext0` wake source is used, and when the main CPU is in deep sleep.

I should note that in the above scenario, the `ext0` wake source does work as expected to wake the ESP32. The only strange behavior seems to be the ULP wake delay being doubled.

I also experimented with different wake delays, and the doubling of the delay is consistent.

### Steps to reproduce

1. Checkout, build and run the sample code as-is, and observe the doubling behavior of the ULP delay.
2. Comment-out the line with the `ext0` wakeup source in the code, rebuild and reflash/run, and observe correct ULP delays.

Note that the example does use the (awesome) [HULP library from boarchuz](https://github.com/boarchuz/HULP) for convenience. The project includes it as a git submodule, so be sure to fetch submodules before building.

And as you can see in the code, I'm using GPIO2 as the `ext0` wake pin.

### Code to reproduce this issue

See the code in this repo :)

## Debug Logs

I'll add logs for both when an `ext0` wake source is enabled and when not enabled. In each case I've included the logs of the first boot and 60s awake period, first 60s deep-sleep period, and the second boot (from deep sleep) for 60s until it goes back into deep-sleep.

This first log is when the `ext0` wake source *is enabled*. As you can see, the ESP32 boots, and then every 10 seconds it detects new data from the ULP and prints all saved data. After 60 seconds it enters deep sleep. Once the ESP32 wakes up and prints out the ULP data you can see that instead of saving 6 values while it was in deep sleep (one every 10s), it only saved 3 (one every 20s). But once the ESP32 is awake again the new data comes in every 10 seconds. While I've stopped the log here, it repeats just like this over and over as the ESP32 enters and exits deep sleep.

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5888
load:0x40078000,len:14804
ho 0 tail 12 room 4
load:0x40080400,len:3792
entry 0x40080694
I (29) boot: ESP-IDF v5.0-dev-517-g68cf4ef2be 2nd stage bootloader
I (29) boot: compile time 11:39:43
I (29) boot: chip revision: 3
I (31) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
[... 526 lines cut ...]
I (60073) ulp-test: ULP Data Index:   1 || Time:    10325ms -  10.3s || Delta:     9979ms -  10.0s
I (60083) ulp-test: ULP Data Index:   2 || Time:    20304ms -  20.3s || Delta:     9979ms -  10.0s
I (60093) ulp-test: ULP Data Index:   3 || Time:    30284ms -  30.3s || Delta:     9980ms -  10.0s
I (60103) ulp-test: ULP Data Index:   4 || Time:    40263ms -  40.3s || Delta:     9979ms -  10.0s
I (60113) ulp-test: ULP Data Index:   5 || Time:    50242ms -  50.2s || Delta:     9979ms -  10.0s
I (60113) ulp-test: ULP Data Index:   6 || Time:    60221ms -  60.2s || Delta:     9979ms -  10.0s
I (60123) ulp-test: ULP Data Index:   7 || Time:    70201ms -  70.2s || Delta:     9980ms -  10.0s
I (60133) ulp-test: ULP Data Index:   8 || Time:    80180ms -  80.2s || Delta:     9979ms -  10.0s
I (60143) ulp-test: ULP Data Index:   9 || Time:    90159ms -  90.2s || Delta:     9979ms -  10.0s
I (60153) ulp-test: ULP Data Index:  10 || Time:   100139ms - 100.1s || Delta:     9980ms -  10.0s
I (60163) ulp-test: ULP Data Index:  11 || Time:   110118ms - 110.1s || Delta:     9979ms -  10.0s
I (60173) ulp-test: ULP Data Index:  12 || Time:   120097ms - 120.1s || Delta:     9979ms -  10.0s
I (60183) ulp-test: ULP Data Index:  13 || Time:   130077ms - 130.1s || Delta:     9980ms -  10.0s
I (60183) ulp-test: ULP Data Index:  14 || Time:   140056ms - 140.1s || Delta:     9979ms -  10.0s
I (60193) ulp-test: ULP Data Index:  15 || Time:   150035ms - 150.0s || Delta:     9979ms -  10.0s
I (60203) ulp-test: ULP Data Index:  16 || Time:   160014ms - 160.0s || Delta:     9979ms -  10.0s
I (60213) ulp-test: ULP Data Index:  17 || Time:   169994ms - 170.0s || Delta:     9980ms -  10.0s
I (60223) ulp-test: ULP Data Index:  18 || Time:   179973ms - 180.0s || Delta:     9979ms -  10.0s
I (61233) ulp-test: Entering deep sleep for 60s
```

## Other items if possible

- The github project has an `sdkconfig.defaults` file - the only other change I made was to disable ANSI color in logging.
- As mentioned above, the ULP code does use the [HULP library](https://github.com/boarchuz/HULP), but only minimally. Based on the behavior I'm seeing I really don't think it is contributing to the strange ULP behavior.

boarchuz, 2021-12-09:
Hi @cdollar393,
Give this a try:
`REG_SET_FIELD(RTC_CNTL_TIMER2_REG, RTC_CNTL_ULPCP_TOUCH_START_WAIT, 0x20);`
https://github.com/espressif/esp-idf/issues/4690#issuecomment-981859554
The RTC peripherals domain will be powered on in deep sleep when ext0 wakeup is enabled. While the docs say this is not compatible with ULP _wakeup_, I think it means that the ULP is not supported at all in this case.

cdollar393, 2021-12-09:
@boarchuz, I owe you a beer (or 6)!! Adding that before entering deep sleep seems to fix the timing issue! Thank you!!

Alvin1Zhang, 2021-12-10:
@boarchuz Thanks for the notes, @cdollar393 thanks for reporting, feel free to reopen.

closed 2021-12-10 by Alvin1Zhang